avr-llvm/llvm #27: Update unit tests to current LLVM

dylanmckay, 2015-01-31:
Upon forking the original AVR-LLVM repository, a set of AVR tests were imported in the process.

They currently do not compile, and so they need to be fixed.

All of the imported tests exist in `test/CodeGen/AVR`.

A current table of tests which need to be fixed:

---------------|----------
File name      | Updated
---------------|----------
add.ll         |
alloca.ll      |
and.ll         |
brind.ll       |
call.ll        |
cmp.ll         |
com.ll         |
directmem.ll   |
dynalloca.ll   |
frame.ll       |
inline-asm.ll  |
interrupts.ll  |
io.ll          |
load.ll        |
mul.ll         |
neg.ll         |
or.ll          |
progmem.ll     |
return.ll      |
sext.ll        |
store.ll       |
sub.ll         |
trunc.ll       |
varargs.ll     |
xor.ll         |
zext.ll        |

RandomInsano, 2015-03-08:
Can we put this sort of thing in the Wiki or a standalone file?

dylanmckay, 2015-03-08:
> Can we put this sort of thing in the Wiki or a standalone file?

I believe this is a fairly small task, and that is overkill.

dylanmckay, 2015-05-04:
I had mistakenly closed the issue after I updated the tests to LLVM master, but the issues were still failing.

I have marked all failing tests as `XFAIL` so that they can be fixed.

commit b72e68d6 (2015-03-10): [AVR] Updated CodeGen tests to master

There is a set of tests located in `test/CodeGen/AVR` which came from
the original AVR-LLVM repository on SourceForge. They had not been
updated, so they used an old syntax of `load` and `getelementptr`.

Closes #27.
PR #189 (2016-05-01): [AVR] fix load/store tests
Partially fixes #27 by updating load/store tests to match with currently generated output.
closed 2016-05-01 by dylanmckay