avr-llvm/llvm #31: Create Getting Started Guide

RandomInsano, 2015-02-10:
How can I use your AVR fanciness? I'm doing some AVR work lately, so this seems like something worth playing with. I've got reasonable C experience, lots of Unix build fun experience, but about zero in the compiler development department.

Do you have some time to spend showing me how to start working on this? If I can get it working I'll MR a guide so others can pick this up faster.

Feel free to e-mail me at EdwinGuy@GMail.com

dylanmckay, 2015-02-10:
I'll writeup a wiki page describing the general structure of the backend, how it works, and what needs doing, thereby closing #5 in the process.

RandomInsano, 2015-02-10:
See, now if I'd seen that I wouldn't have opened this :)

Closing, adding my bits there.

closed 2015-02-10 by RandomInsano