avr-llvm/llvm #34: PassManager.h not found

RandomInsano, 2015-02-16:
Moving on from issue #33, I tried the Autotools instructions. I built first with `make -j9` but it failed on error. Running `make` again, I hit this guy:

```
make[1]: Nothing to be done for `all'.
make[1]: Nothing to be done for `all'.
...
make[3]: Nothing to be done for `all'.
llvm[3]: Compiling AVRTargetMachine.cpp for Debug+Asserts build
/Users/edwinamsler/Documents/Code/Personal/llvm-avr/avr-llvm/lib/Target/AVR/AVRTargetMachine.cpp:19:10: fatal error: 
      'llvm/PassManager.h' file not found
#include "llvm/PassManager.h"
         ^
1 error generated.
rm: /Users/edwinamsler/Documents/Code/Personal/llvm-avr/avr-llvm/build/lib/Target/AVR/Debug+Asserts/AVRTargetMachine.d.tmp: No such file or directory
make[3]: *** [/Users/edwinamsler/Documents/Code/Personal/llvm-avr/avr-llvm/build/lib/Target/AVR/Debug+Asserts/AVRTargetMachine.o] Error 1
make[2]: *** [AVR/.makeall] Error 2
make[1]: *** [Target/.makeall] Error 2
make: *** [all] Error 1
```

Looking into that a little bit more, I definitely have PassManager.h available, but it's in the wrong folder:

```
$ cd ..
$ find . | grep PassManager\.h
./include/llvm/Analysis/CGSCCPassManager.h
./include/llvm/IR/LegacyPassManager.h
./include/llvm/IR/PassManager.h
```

I saw code from the main LLVM project was folded in recently. Maybe it was moved upstream at some point?

RandomInsano, 2015-02-16:
More update while grep searches for other references to that file; looking back at the history of that file in Github, it's been there since at least this commit:
https://github.com/dylanmckay/avr-llvm/commit/ea9988447c5bd59827baf1b45dd5b62b4065d553

Does that sound right?

In any case, I'll post an inline patch file here with the changes needed. I can't do a PR without learning some fancy git-fu or removing my current fork of llvm.

RandomInsano, 2015-02-16:
Here's the patch (not even worth creating :P). It looks like @4ntoine's fork also has the same problem.

This patch file looks a bit stranger than I'm used to, so in case you've never used Git's patching either:
https://ariejan.net/2009/10/26/how-to-create-and-apply-a-patch-with-git/

```
From 665ab8f5a82782f0db4cf70b7efe37520fdeb5de Mon Sep 17 00:00:00 2001
From: Edwin Amsler <edwinguy@gmail.com>
Date: Mon, 16 Feb 2015 11:13:09 -0600
Subject: [PATCH] Fix include of PassManager.h

---
 lib/Target/AVR/AVRTargetMachine.cpp | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/lib/Target/AVR/AVRTargetMachine.cpp b/lib/Target/AVR/AVRTargetMachine.cpp
index fe7f6ac..211daf3 100644
--- a/lib/Target/AVR/AVRTargetMachine.cpp
+++ b/lib/Target/AVR/AVRTargetMachine.cpp
@@ -16,7 +16,7 @@
 #include "AVR.h"
 #include "llvm/CodeGen/Passes.h"
 #include "llvm/IR/Module.h"
-#include "llvm/PassManager.h"
+#include "llvm/IR/PassManager.h"
 #include "llvm/Support/TargetRegistry.h"
 
 using namespace llvm;
-- 
1.9.3 (Apple Git-50)
```

dylanmckay, 2015-02-16:
I believe this may be the problem:

http://reviews.llvm.org/rL229094

I will provide a fix soon. I believe your patch changes the include from `llvm/PassManager.h` to `llvm/IR/PassManager.h`, but we should be swapping it to `llvm/IR/LegacyPassManager.h`.

commit ee612290 (2015-02-17): [AVR] Updated AVR backend to LLVM master

In r229340, the `LLVM_DELETED_FUNCTION` macro was removed, in favor of
the `int MyFunction() = delete` syntax.

`include/llvm/PassManager.h` was moved to
`include/llvm/IR/LegacyPassManager.h`.

Closes #34.
closed 2015-02-17 by dylanmckay