avr-llvm/llvm #35: Make AVR triple valid

dylanmckay, 2015-02-17:
Currently, the target triple for AVR is `avr-none`.

This is not a valid triple (see [here](http://llvm.org/docs/LangRef.html#target-triple)).

I believe it should be `avr-unknown-none` or `avr-atmel-none`.

dylanmckay, 2015-02-17:
Check what AVR-GCC uses.

RandomInsano, 2015-02-18:
Well, by the docs, it should be avr-atmel-none right?

dylanmckay, 2015-02-18:
I thought so, but look at the output of `avr-gcc -v`:
```
Using built-in specs.
...
Target: avr
..
gcc version 4.9.2 20141224 (prerelease) (GCC) 
```

versus `gcc -v` on my machine:
```
Using built-in specs.
..
Target: x86_64-unknown-linux-gnu
...
gcc version 4.9.2 20141224 (prerelease) (GCC) 

```

"Target" is the target triple. For `x64`, it is in the normal triple format. For `avr`, it is not.

dylanmckay, 2015-02-18:
See [this](https://gcc.gnu.org/install/specific.html) page as well.

RandomInsano, 2015-02-18:
Weird. I guess ask the LLVM mailing list? The elders there may have an idea as to why GCC chose that route? It can't possibly be a flame/war bike shed topic right?

dylanmckay, 2015-02-18:
The [AVR-GCC mailing list](https://lists.nongnu.org/mailman/listinfo/avr-gcc-list) might be a better option.

dylanmckay, 2015-06-19:
I'm going to change the default triple to `avr-atmel-none`. It is more conventional. Compiler-rt also requires a three-part triple.

dylanmckay, 2015-06-19:
I don't believe there is much (if any) effort required for this. `avr-none` is not hardcoded anywhere, it is simply the argument passed to `configure`.

The architecture `avr` is already supported. The vendor `atmel` is already supported. And I just added support for the `none` OS in bd61c32037143f33b457db4e72eb7c42880266eb.

dylanmckay, 2015-06-21:
Closing as bd61c32 was all that needed to be done.

closed 2015-06-21 by dylanmckay