avr-llvm/llvm #37: Wrong MachineOperand accessor error

dylanmckay, 2015-02-17:
Code to reproduce:

C (`main.c`):
``` c

struct ss
{
    int a;
    int b;
    int c;
};

void loop(struct ss *x, struct ss **y, int z)
{
    int i;
    for (i=0; i<z; ++i)
    {
        x->c += y[i]->b;
    }
}
```

Command line:
``` bash
clang --target=avr -O1 main.c -o main.o
```

Or alternatively:

LLVM IR (`main.ll`):
``` asm
; ModuleID = '/home/dylan/avr/pointers/other.c'
target datalayout = "e-p:16:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8"
target triple = "avr"

%struct.ss = type { i16, i16, i16 }

; Function Attrs: nounwind
define void @loop(%struct.ss* nocapture %x, %struct.ss** nocapture readonly %y, i16 %z) #0 {
entry:
  %cmp3 = icmp sgt i16 %z, 0
  br i1 %cmp3, label %for.body.lr.ph, label %for.end

for.body.lr.ph:                                   ; preds = %entry
  %c = getelementptr inbounds %struct.ss* %x, i16 0, i32 2
  %c.promoted = load i16* %c, align 2, !tbaa !1
  br label %for.body

for.body:                                         ; preds = %for.body.lr.ph, %for.body
  %add5 = phi i16 [ %c.promoted, %for.body.lr.ph ], [ %add, %for.body ]
  %i.04 = phi i16 [ 0, %for.body.lr.ph ], [ %inc, %for.body ]
  %arrayidx = getelementptr inbounds %struct.ss** %y, i16 %i.04
  %0 = load %struct.ss** %arrayidx, align 2, !tbaa !6
  %b = getelementptr inbounds %struct.ss* %0, i16 0, i32 1
  %1 = load i16* %b, align 2, !tbaa !8
  %add = add nsw i16 %add5, %1
  %inc = add nuw nsw i16 %i.04, 1
  %cmp = icmp slt i16 %inc, %z
  br i1 %cmp, label %for.body, label %for.cond.for.end_crit_edge

for.cond.for.end_crit_edge:                       ; preds = %for.body
  store i16 %add, i16* %c, align 2, !tbaa !1
  br label %for.end

for.end:                                          ; preds = %for.cond.for.end_crit_edge, %entry
  ret void
}

attributes #0 = { nounwind "less-precise-fpmad"="false" "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "stack-protector-buffer-size"="8" "unsafe-fp-math"="false" "use-soft-float"="false" }

!llvm.ident = !{!0}

!0 = !{!"clang version 3.7.0 (https://github.com/llvm-mirror/clang.git a69a4576ade0bfa4644320b72a4cf425ea62e306) (https://github.com/llvm-mirror/llvm.git dedb06d9539f46e4aeed3a4dc53880147493c4e1)"}
!1 = !{!2, !3, i64 4}
!2 = !{!"ss", !3, i64 0, !3, i64 2, !3, i64 4}
!3 = !{!"int", !4, i64 0}
!4 = !{!"omnipotent char", !5, i64 0}
!5 = !{!"Simple C/C++ TBAA"}
!6 = !{!7, !7, i64 0}
!7 = !{!"any pointer", !4, i64 0}
!8 = !{!2, !3, i64 2}
```

Command line:
``` bash
llc main.ll -o main.o
```

Tested with dedb06d9539f46e4aeed3a4dc53880147493c4e1

4ntoine, 2015-02-22:
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/avr-libc/realloc.c -o /tmp/arduino_test1/realloc.c.o
Assertion failed: (isReg() && IsDef && "Wrong MachineOperand accessor"), function setIsEarlyClobber, file /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/include/llvm/CodeGen/MachineOperand.h, line 394.
```
0  clang-3.7                0x000000010be13b5e llvm::sys::PrintStackTrace(__sFILE*) + 46
1  clang-3.7                0x000000010be14f1b PrintStackTraceSignalHandler(void*) + 27
2  clang-3.7                0x000000010be15365 SignalHandler(int) + 565
3  libsystem_platform.dylib 0x00007fff9398a5aa _sigtramp + 26
4  libsystem_platform.dylib 0x00007fff54cf1830 _sigtramp + 3241570976
5  clang-3.7                0x000000010be14f4b raise + 27
6  clang-3.7                0x000000010be15002 abort + 18
7  clang-3.7                0x000000010be14fe1 __assert_rtn + 129
8  clang-3.7                0x000000010b361437 llvm::MachineOperand::setIsEarlyClobber(bool) + 151
9  clang-3.7                0x000000010b359a2a llvm::MachineInstr::addOperand(llvm::MachineFunction&, llvm::MachineOperand const&) + 1674
10 clang-3.7                0x000000010af3d37f llvm::MachineInstrBuilder::addReg(unsigned int, unsigned int, unsigned int) const + 479
11 clang-3.7                0x000000010c43c10b llvm::InstrEmitter::AddRegisterOperand(llvm::MachineInstrBuilder&, llvm::SDValue, unsigned int, llvm::MCInstrDesc const*, llvm::DenseMap<llvm::SDValue, unsigned int, llvm::DenseMapInfo<llvm::SDValue>, llvm::detail::DenseMapPair<llvm::SDValue, unsigned int> >&, bool, bool, bool) + 1611
12 clang-3.7                0x000000010c43cad9 llvm::InstrEmitter::AddOperand(llvm::MachineInstrBuilder&, llvm::SDValue, unsigned int, llvm::MCInstrDesc const*, llvm::DenseMap<llvm::SDValue, unsigned int, llvm::DenseMapInfo<llvm::SDValue>, llvm::detail::DenseMapPair<llvm::SDValue, unsigned int> >&, bool, bool, bool) + 2457
13 clang-3.7                0x000000010c43efab llvm::InstrEmitter::EmitMachineNode(llvm::SDNode*, bool, bool, llvm::DenseMap<llvm::SDValue, unsigned int, llvm::DenseMapInfo<llvm::SDValue>, llvm::detail::DenseMapPair<llvm::SDValue, unsigned int> >&) + 1787
14 clang-3.7                0x000000010c5253e6 llvm::InstrEmitter::EmitNode(llvm::SDNode*, bool, bool, llvm::DenseMap<llvm::SDValue, unsigned int, llvm::DenseMapInfo<llvm::SDValue>, llvm::detail::DenseMapPair<llvm::SDValue, unsigned int> >&) + 102
15 clang-3.7                0x000000010c5234dc llvm::ScheduleDAGSDNodes::EmitSchedule(llvm::MachineBasicBlock::bundle_iterator<llvm::MachineInstr, llvm::ilist_iterator<llvm::MachineInstr> >&) + 1340
16 clang-3.7                0x000000010c60a3dc llvm::SelectionDAGISel::CodeGenAndEmitDAG() + 8396
17 clang-3.7                0x000000010c6082fd llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator<llvm::Instruction const>, llvm::ilist_iterator<llvm::Instruction const>, bool&) + 253
18 clang-3.7                0x000000010c608019 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) + 5241
19 clang-3.7                0x000000010c605877 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 1143
20 clang-3.7                0x000000010b355c3e llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 110
21 clang-3.7                0x000000010b74d6db llvm::FPPassManager::runOnFunction(llvm::Function&) + 427
22 clang-3.7                0x000000010b74d9e8 llvm::FPPassManager::runOnModule(llvm::Module&) + 104
23 clang-3.7                0x000000010b74e3f4 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) + 1412
24 clang-3.7                0x000000010b74dc9e llvm::legacy::PassManagerImpl::run(llvm::Module&) + 302
25 clang-3.7                0x000000010b74eb71 llvm::legacy::PassManager::run(llvm::Module&) + 33
26 clang-3.7                0x000000010c666671 (anonymous namespace)::EmitAssemblyHelper::EmitAssembly(clang::BackendAction, llvm::raw_ostream*) + 1601
27 clang-3.7                0x000000010c665e32 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::raw_ostream*) + 114
28 clang-3.7                0x000000010c88bac5 clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) + 1397
29 clang-3.7                0x000000010d3bc74b clang::ParseAST(clang::Sema&, bool, bool) + 1291
30 clang-3.7                0x000000010c2764fa clang::ASTFrontendAction::ExecuteAction() + 522
31 clang-3.7                0x000000010c889b7f clang::CodeGenAction::ExecuteAction() + 3999
32 clang-3.7                0x000000010c275a78 clang::FrontendAction::Execute() + 120
33 clang-3.7                0x000000010c205dbc clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) + 1020
34 clang-3.7                0x000000010c2e7e41 clang::ExecuteCompilerInvocation(clang::CompilerInstance*) + 3201
35 clang-3.7                0x000000010af17570 cc1_main(llvm::ArrayRef<char const*>, char const*, void*) + 2496
[... cut ...]
clang-3.7: error: unable to execute command: Illegal instruction: 4
clang-3.7: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.7.0 (https://github.com/avr-llvm/clang.git 98cf874e96493bec6b96f3a8339f11150f4f198b) (llvm/llvm a411aba0af80de3090448dc3e56f39d8627a6647)
Target: avr
Thread model: posix
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.

4ntoine, 2015-03-03:
i'm able to reproduce error and debug until that point. Unfortunately i'm not so familiar to LLVM internals to understand what's wrong here.. Any suggestions?

dylanmckay, 2015-06-07:
Reopening because this same problem (although presumably with a different instruction with the wrong operand marked as `earlyclobber`).

See `test/CodeGen/AVR/mul.ll`

dylanmckay, 2015-06-07:
Fixed in 26896ec4092450267a03a53bda79919037ccebba.

closed 2015-06-07 by dylanmckay

4ntoine, 2015-06-13:
still happens:
```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp -o /tmp/arduino_test1/HardwareSerial.cpp.o
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:32:
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial_private.h:119:5: warning: 
      expression result unused; assign into a variable to force a volatile load [-Wunused-volatile-lvalue]
    *_udr;
    ^~~~~
fatal error: error in backend: ran out of registers during register allocation
clang-3.7: error: clang frontend command failed with exit code 70 (use -v to see invocation)
clang version 3.7.0 (https://github.com/avr-llvm/clang.git aebd4f6d15e44635ecc7338d4454389c4573bc2b) (llvm/llvm 9c7c5212863ed8b06323ad07fa84bad29ff051f8)
Target: avr
Thread model: posix
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.7: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/HardwareSerial-04d113.cpp
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/HardwareSerial-04d113.sh
clang-3.7: note: diagnostic msg: 

********************

```

dylanmckay, 2015-06-13:
The bug you describe is #1.

A hack existed that fixed it (in most cases I think) but I removed it to see if it changed anything in 7f4cc26311e08d90c9296897beff116e96066bfd.

You could try reverting this commit in your own private branch and see if the error exists, otherwise I will investigate when I have time.

commit 7afcd3cc (2015-04-24): [AVR] Fixed assertion error in backend.

The wrong operand in `LDWRdPtr` was marked as `earlyclobber`. This
caused an assertion error when LLVM tried to set the MachineOperand
`earlyclobber` flag.

Closes #37.