avr-llvm/llvm #42: Enable tests on travis-ci

dylanmckay, 2015-02-21:
Currently, most of our tests fail. this is because I wrote tests for every existing AVR instruction, including the ones that are not implemented. It is also because we have some encoding errors in several instructions.

Once we are in a passing state, get travis to run our tests in the build process.

RandomInsano, 2015-02-22:
Travis should run any tests that would be checked by running 'make test' after the build succeeds.

How are the tests done? We can likely define our own build + test commands by adding something like this:

```
script:
  - ./configure
  - make
  - ./test-app
```

dylanmckay, 2015-02-22:
Tests are executed using `make check`.

dylanmckay, 2015-03-08:
This issue depends on:
* Implement entire AVR instruction set (#14)
* Fix CodeGen unit tests (#27)

dylanmckay, 2015-05-04:
I have got the AVR MC test suite to the point where there are only two issues left:
* `MOVW Rd, Rr` (#84)
* `LDD`/`STD` (#71)

I will mark them as `XFAIL` so that they are expected to fail. Our test suite should then pass, and we can fix the aforementioned tests and remove the `XFAIL` annotation.

RandomInsano, 2015-05-04:
Noted! On my way home for dinner, will make the change tonight.

dylanmckay, 2015-05-04:
Note that `make check` will attempt to run all tests (including x86,ARM,etc).

We pass `-DLLVM_TARGETS_TO_BUILD="AVR"` to CMake in `.travis.yml`, and so the tests will fail with `make check` (because "x86 is an unknown target").

We probably have to run the AVR tests manually. Because all AVR tests are parented by folders named `AVR`, it would be sufficient to call:

``` bash
# run the LLVM testing tool (lit) on all files of type d (directory) with name "AVR"
llvm-lit $(find <llvm-root>/test -name AVR -type d)
```

RandomInsano, 2015-05-05:
Yup, looks good over here. Even if `make check` failed due to missing targets, it makes sense to only test the AVR components.

I'll fold the tests in now and see how she goes.

commit 9f1f4ea6 (2015-05-07): [AVR] Run tests on travis-ci

Closes #42.
closed 2015-05-07 by dylanmckay