avr-llvm/llvm #45: invalid operand for instruction

4ntoine, 2015-02-22:
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring.c -o /tmp/arduino_test1/wiring.c.o

```
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring.c:25:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_private.h:30:
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:724:12: warning: 
      declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:875:12: warning: 
      declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:888:12: warning: 
      declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring.c:185:3: error: invalid
      operand for instruction
                "1: sbiw %0,1" "\n\t" // 2 cycles
                ^
<inline asm>:1:10: note: instantiated into assembly here
        1: sbiw r24,1
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring.c:185:21: error: 
      unexpected token in argument list
                "1: sbiw %0,1" "\n\t" // 2 cycles
                                  ^
<inline asm>:2:8: note: instantiated into assembly here
        brne 1b
              ^
3 warnings and 2 errors generated.
```

4ntoine, 2015-02-22:
error code (asm):

```
// busy wait
	__asm__ __volatile__ (
		"1: sbiw %0,1" "\n\t" // 2 cycles
		"brne 1b" : "=w" (us) : "0" (us) // 2 cycles
	);
```

4ntoine, 2015-02-22:
full code (wiring.c):

```
/*
  wiring.c - Partial implementation of the Wiring API for the ATmega8.
  Part of Arduino - http://www.arduino.cc/

  Copyright (c) 2005-2006 David A. Mellis

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General
  Public License along with this library; if not, write to the
  Free Software Foundation, Inc., 59 Temple Place, Suite 330,
  Boston, MA  02111-1307  USA

  $Id$
*/

#include "wiring_private.h"

// the prescaler is set so that timer0 ticks every 64 clock cycles, and the
// the overflow handler is called every 256 ticks.
#define MICROSECONDS_PER_TIMER0_OVERFLOW (clockCyclesToMicroseconds(64 * 256))

// the whole number of milliseconds per timer0 overflow
#define MILLIS_INC (MICROSECONDS_PER_TIMER0_OVERFLOW / 1000)

// the fractional number of milliseconds per timer0 overflow. we shift right
// by three to fit these numbers into a byte. (for the clock speeds we care
// about - 8 and 16 MHz - this doesn't lose precision.)
#define FRACT_INC ((MICROSECONDS_PER_TIMER0_OVERFLOW % 1000) >> 3)
#define FRACT_MAX (1000 >> 3)

volatile unsigned long timer0_overflow_count = 0;
volatile unsigned long timer0_millis = 0;
static unsigned char timer0_fract = 0;

#if defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
ISR(TIM0_OVF_vect)
#else
ISR(TIMER0_OVF_vect)
#endif
{
	// copy these to local variables so they can be stored in registers
	// (volatile variables must be read from memory on every access)
	unsigned long m = timer0_millis;
	unsigned char f = timer0_fract;

	m += MILLIS_INC;
	f += FRACT_INC;
	if (f >= FRACT_MAX) {
		f -= FRACT_MAX;
		m += 1;
	}

	timer0_fract = f;
	timer0_millis = m;
	timer0_overflow_count++;
}

unsigned long millis()
[... 237 lines cut ...]

#if defined(ADCSRA)
	// set a2d prescale factor to 128
	// 16 MHz / 128 = 125 KHz, inside the desired 50-200 KHz range.
	// XXX: this will not work properly for other clock speeds, and
	// this code should use F_CPU to determine the prescale factor.
	sbi(ADCSRA, ADPS2);
	sbi(ADCSRA, ADPS1);
	sbi(ADCSRA, ADPS0);

	// enable a2d conversions
	sbi(ADCSRA, ADEN);
#endif

	// the bootloader connects pins 0 and 1 to the USART; disconnect them
	// here so they can be used as normal digital i/o; they will be
	// reconnected in Serial.begin()
#if defined(UCSRB)
	UCSRB = 0;
#elif defined(UCSR0B)
	UCSR0B = 0;
#endif
}

```

dylanmckay, 2015-03-07:
Error can be more easily reproduced with this assembly snippet:

`main.s`:
``` asm
1: sbiw r28, 24

; jump to label '1'
brne 1b
```

With the command line

``` bash
llvm-mc -arch=avr -filetype=obj main.s
```

Tested with revision a986ebcbbc3d3ae82d3f4a71e1d07f086bed6b82

dylanmckay, 2015-03-07:
There are two errors:

* Caused by issue #32, `r24, r25`, etc are not parsed as pointer registers.
* The suffix `b` on the `brne` operand is not supported by the asm parser. From the context it is used in, I believe it is used to explicitly state that a value is a symbol/label: `1

dylanmckay, 2015-03-08:
Both of these issues were caused by #53. The explanation is duplicated here:

The problem lies in [`AVRAsmParser.cpp:466`](https://github.com/avr-llvm/llvm/blob/26042bd81f8fc1a9aa003054201d182353644047/lib/Target/AVR/AsmParser/AVRAsmParser.cpp#L466).

Whenever we try and parse an operand, we first check if it is a valid register. If it is, then we treat it as such. For situations like `sbiw X, 24`, `X` is parsed into a `MCOperand` of type register, and then we get to `24` (which is a valid register, our code currently allows `31` instead of `r31` currently -- perhaps this is in error). `24` is parsed into an `MCOperand` of type register, even though it is not a register. We still get the correct output, however.

In the case of local labels, they are automatically taken to be registers. Because of this, the correct branch (in the else block) is not executed, and `MCParser.parseExpression()` doesn't handle it. `MCParser.parseExpression()` has code to handle a trailing `b` or `f`.

There is an ambiguity when parsing integer operands without context, because AVR-GCC supports specifiying register names (`r12`) and register numbers (`12`) interchangeably. Without context, we can't tell the difference.

There are two solutions:

* Drop support for numbered registers and only support the `r12` syntax. This would make us incompatible with AVR-GCC.
* Use context to disambiguate between the two cases. We could hook into the tablegen'd `AVRGenInstrInfo.inc`, check if the current operand is a register, then handle it appropriately.

We dropped support for numbered registers, and now the problem is solved.

closed 2015-03-08 by dylanmckay