avr-llvm/llvm #46: error: cannot select

4ntoine, 2015-02-22:
./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_analog.c -o /tmp/arduino_test1/wiring_analog.c.o

```
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_analog.c:27:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_private.h:30:
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:724:12: warning: 
      declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:875:12: warning: 
      declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:888:12: warning: 
      declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
fatal error: error in backend: Cannot select: 0x7fd9618e0860: i16 = JumpTable<0> [ID=2]
In function: analogWrite
clang-3.7: error: clang frontend command failed with exit code 70 (use -v to see invocation)
clang version 3.7.0 (https://github.com/avr-llvm/clang.git 98cf874e96493bec6b96f3a8339f11150f4f198b) (llvm/llvm a411aba0af80de3090448dc3e56f39d8627a6647)
Target: avr
Thread model: posix
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.7: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/wiring_analog-ef1cd4.c
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/wiring_analog-ef1cd4.sh
clang-3.7: note: diagnostic msg: 
```

4ntoine, 2015-02-22:
```
/*
  wiring_analog.c - analog input and output
  Part of Arduino - http://www.arduino.cc/

  Copyright (c) 2005-2006 David A. Mellis

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General
  Public License along with this library; if not, write to the
  Free Software Foundation, Inc., 59 Temple Place, Suite 330,
  Boston, MA  02111-1307  USA

  Modified 28 September 2010 by Mark Sproul

  $Id: wiring.c 248 2007-02-03 15:36:30Z mellis $
*/

#include "wiring_private.h"
#include "pins_arduino.h"

uint8_t analog_reference = DEFAULT;

void analogReference(uint8_t mode)
{
	// can't actually set the register here because the default setting
	// will connect AVCC and the AREF pin, which would cause a short if
	// there's something connected to AREF.
	analog_reference = mode;
}

int analogRead(uint8_t pin)
{
	uint8_t low, high;

#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
	if (pin >= 54) pin -= 54; // allow for channel or pin numbers
#elif defined(__AVR_ATmega32U4__)
	if (pin >= 18) pin -= 18; // allow for channel or pin numbers
#elif defined(__AVR_ATmega1284__) || defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644__) || defined(__AVR_ATmega644A__) || defined(__AVR_ATmega644P__) || defined(__AVR_ATmega644PA__)
	if (pin >= 24) pin -= 24; // allow for channel or pin numbers
#elif defined(analogPinToChannel) && (defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__))
	pin = analogPinToChannel(pin);
#else
	if (pin >= 14) pin -= 14; // allow for channel or pin numbers
#endif
	
#if defined(__AVR_ATmega32U4__)
	pin = analogPinToChannel(pin);
	ADCSRB = (ADCSRB & ~(1 << MUX5)) | (((pin >> 3) & 0x01) << MUX5);
#elif defined(ADCSRB) && defined(MUX5)
	// the MUX5 bit of ADCSRB selects whether we're reading from channels
	// 0 to 7 (MUX5 low) or 8 to 15 (MUX5 high).
	ADCSRB = (ADCSRB & ~(1 << MUX5)) | (((pin >> 3) & 0x01) << MUX5);
#endif
  
	// set the analog reference (high two bits of ADMUX) and select the
	// channel (low 4 bits).  this also sets ADLAR (left-adjust result)
	// to 0 (the default).
#if defined(ADMUX)
	ADMUX = (analog_reference << 6) | (pin & 0x07);
[... 190 lines cut ...]
				// connect pwm to pin on timer 5, channel B
				sbi(TCCR5A, COM5B1);
				OCR5B = val; // set pwm duty
				break;
			#endif

			#if defined(TCCR5A) && defined(COM5C1)
			case TIMER5C:
				// connect pwm to pin on timer 5, channel C
				sbi(TCCR5A, COM5C1);
				OCR5C = val; // set pwm duty
				break;
			#endif

			case NOT_ON_TIMER:
			default:
				if (val < 128) {
					digitalWrite(pin, LOW);
				} else {
					digitalWrite(pin, HIGH);
				}
		}
	}
}
```

4ntoine, 2015-02-22:
One more case:

```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp -o /tmp/arduino_test1/Stream.cpp.o
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:23:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Arduino.h:193:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/HardwareSerial.h:28:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.h:26:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Print.h:27:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Printable.h:23:
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/new.h:10:8: warning: 
      'operator new' is missing exception specification 'throw(std::bad_alloc)'
void * operator new(size_t size);
       ^
                                 throw(std::bad_alloc)
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/new.h:11:8: warning: 
      'operator new[]' is missing exception specification 'throw(std::bad_alloc)'
void * operator new[](size_t size);
       ^
                                   throw(std::bad_alloc)
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/new.h:12:6: warning: 
      'operator delete' is missing exception specification 'throw()'
void operator delete(void * ptr);
     ^
                                 throw()
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/new.h:13:6: warning: 
      'operator delete[]' is missing exception specification 'throw()'
void operator delete[](void * ptr);
     ^
                                   throw()
fatal error: error in backend: Cannot select: 0x7fdb83266260: glue = AVRISD::TST 0x7fdb832667b0 [ORD=9] [ID=32]
      dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
  0x7fdb832667b0: i8 = truncate 0x7fdb84031d50 [ORD=9] [ID=31] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
    0x7fdb84031d50: i16 = AVRISD::LSR 0x7fdb84031c40 [ORD=9] [ID=30] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
      0x7fdb84031c40: i16 = AVRISD::LSR 0x7fdb84031b30 [ORD=9] [ID=29] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
        0x7fdb84031b30: i16 = AVRISD::LSR 0x7fdb84031a20 [ORD=9] [ID=28] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
          0x7fdb84031a20: i16 = AVRISD::LSR 0x7fdb84031910 [ORD=9] [ID=27] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
            0x7fdb84031910: i16 = AVRISD::LSR 0x7fdb84031800 [ORD=9] [ID=26] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
              0x7fdb84031800: i16 = AVRISD::LSR 0x7fdb831db240 [ORD=9] [ID=25] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
                0x7fdb831db240: i16 = AVRISD::LSR 0x7fdb831db020 [ORD=9] [ID=23] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9
                  0x7fdb831db020: i16 = AVRISD::LSR 0x7fdb83266480 [ORD=9] [ID=21] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.cpp:36:9

In function: _ZN6Stream9timedReadEv
clang-3.7: error: clang frontend command failed with exit code 70 (use -v to see invocation)
clang version 3.7.0 (https://github.com/avr-llvm/clang.git 98cf874e96493bec6b96f3a8339f11150f4f198b) (llvm/llvm a411aba0af80de3090448dc3e56f39d8627a6647)
Target: avr
Thread model: posix
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.7: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Stream-81ac6d.cpp
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Stream-81ac6d.sh
clang-3.7: note: diagnostic msg: 

********************

```

4ntoine, 2015-02-22:
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Stream-81ac6d.cpp
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Stream-81ac6d.sh:

download here: http://dropmefiles.com/juTT4 within 14 days

4ntoine, 2015-02-22:
one more case:
```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Tone.cpp -o /tmp/arduino_test1/Tone.cpp.o
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Tone.cpp:37:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Arduino.h:193:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/HardwareSerial.h:28:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Stream.h:26:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Print.h:27:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Printable.h:23:
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/new.h:10:8: warning: 
      'operator new' is missing exception specification 'throw(std::bad_alloc)'
void * operator new(size_t size);
       ^
                                 throw(std::bad_alloc)
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/new.h:11:8: warning: 
      'operator new[]' is missing exception specification 'throw(std::bad_alloc)'
void * operator new[](size_t size);
       ^
                                   throw(std::bad_alloc)
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/new.h:12:6: warning: 
      'operator delete' is missing exception specification 'throw()'
void operator delete(void * ptr);
     ^
                                 throw()
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/new.h:13:6: warning: 
      'operator delete[]' is missing exception specification 'throw()'
void operator delete[](void * ptr);
     ^
                                   throw()
fatal error: error in backend: Cannot select: 0x7fd0db9d6b70: glue = AVRISD::TST 0x7fd0db9d94c0 [ORD=5] [ID=7]
      dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Tone.cpp:250:7
  0x7fd0db9d94c0: i8,ch = CopyFromReg 0x7fd0db4837b0, 0x7fd0db9de350 [ORD=3] [ID=6] dbg:/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Tone.cpp:250:7
    0x7fd0db9de350: i8 = Register %vreg3 [ID=1]
In function: _Z4tonehjm
clang-3.7: error: clang frontend command failed with exit code 70 (use -v to see invocation)
clang version 3.7.0 (https://github.com/avr-llvm/clang.git 98cf874e96493bec6b96f3a8339f11150f4f198b) (llvm/llvm a411aba0af80de3090448dc3e56f39d8627a6647)
Target: avr
Thread model: posix
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.7: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Tone-554ef3.cpp
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Tone-554ef3.sh
clang-3.7: note: diagnostic msg: 

********************

```

4ntoine, 2015-02-22:
files to test: http://dropmefiles.com/EkDYY

dylanmckay, 2015-04-23:
The AVR backend does not support jump tables yet. That is probably the root of this problem.

dylanmckay, 2015-04-24:
It looks like you have two different bugs from your three test cases. One jump table instruction selection error, and two `TST Rd` related instruction selection errors.

dylanmckay, 2015-06-10:
None of the bugs present in this issue exist anymore.

closed 2015-07-08 by dylanmckay