avr-llvm/llvm #47: Asm error

4ntoine, 2015-02-22:
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_digital.c -o /tmp/arduino_test1/wiring_digital.c.o

```
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_digital.c:28:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_private.h:30:
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:724:12: warning: 
      declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:875:12: warning: 
      declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:888:12: warning: 
      declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_digital.c:33:16: error: 
      too few operands for instruction
        uint8_t bit = digitalPinToBitMask(pin);
                      ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Arduino.h:142:34: note: expanded
      from macro 'digitalPinToBitMask'
#define digitalPinToBitMask(P) ( pgm_read_byte( digital_pin_to_bit_mask_PGM + (P) ) )
                                 ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:696:41: note: 
      expanded from macro 'pgm_read_byte'
#define pgm_read_byte(address_short)    pgm_read_byte_near(address_short)
                                        ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:410:43: note: 
      expanded from macro 'pgm_read_byte_near'
#define pgm_read_byte_near(address_short) __LPM((uint16_t)(address_short))
                                          ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:398:29: note: 
      expanded from macro '__LPM'
#define __LPM(addr)         __LPM_classic__(addr)
                            ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:258:9: note: 
      expanded from macro '__LPM_classic__'
        "lpm" "\n\t"            \
        ^
<inline asm>:1:2: note: instantiated into assembly here
        lpm
        ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/wiring_digital.c:34:17: error: 
      too few operands for instruction
        uint8_t port = digitalPinToPort(pin);
                       ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Arduino.h:141:31: note: expanded
      from macro 'digitalPinToPort'
#define digitalPinToPort(P) ( pgm_read_byte( digital_pin_to_port_PGM + (P) ) )
                              ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:696:41: note: 
      expanded from macro 'pgm_read_byte'
#define pgm_read_byte(address_short)    pgm_read_byte_near(address_short)
                                        ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:410:43: note: 
      expanded from macro 'pgm_read_byte_near'
#define pgm_read_byte_near(address_short) __LPM((uint16_t)(address_short))
                                          ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:398:29: note: 
      expanded from macro '__LPM'
#define __LPM(addr)         __LPM_classic__(addr)
                            ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:258:9: note: 
      expanded from macro '__LPM_classic__'
        "lpm" "\n\t"            \
[... 305 lines cut ...]
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino/Arduino.h:145:55: note: expanded
      from macro 'portOutputRegister'
#define portOutputRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_output_PGM + (P))) )
                                                      ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:705:41: note: 
      expanded from macro 'pgm_read_word'
#define pgm_read_word(address_short)    pgm_read_word_near(address_short)
                                        ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:418:43: note: 
      expanded from macro 'pgm_read_word_near'
#define pgm_read_word_near(address_short) __LPM_word((uint16_t)(address_short))
                                          ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:399:29: note: 
      expanded from macro '__LPM_word'
#define __LPM_word(addr)    __LPM_word_classic__(addr)
                            ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:288:28: note: 
      expanded from macro '__LPM_word_classic__'
        "adiw r30, 1"   "\n\t"              \
                           ^
<inline asm>:4:2: note: instantiated into assembly here
        lpm
        ^
fatal error: error in backend: Cannot select: 0x7fb933824ae0: i16 = JumpTable<0> [ID=2]
```

4ntoine, 2015-02-22:
Possible duplicate #46

dylanmckay, 2015-03-08:
There are several errors here which need to be split up into seperate issues.

* Support the `LPM` (no operands) instruction (#56)
* "invalid operand for instruction" error (#45)
* Instruction selection error (#46)

Closing as there are now issues for every error generated in the offending code.

closed 2015-03-08 by dylanmckay