avr-llvm/llvm #53: Support local labels

dylanmckay, 2015-03-08:
See `lib/MC/MCParser/AsmParser.cpp:870`.

For some reason, `AsmParser::parsePrimaryExpr` is not handling `b` and `f` suffixes on local labels.

dylanmckay, 2015-03-08:
The problem lies in [`AVRAsmParser.cpp:466`](https://github.com/avr-llvm/llvm/blob/26042bd81f8fc1a9aa003054201d182353644047/lib/Target/AVR/AsmParser/AVRAsmParser.cpp#L466).

Whenever we try and parse an operand, we first check if it is a valid register. If it is, then we treat it as such. For situations like `sbiw X, 24`, `X` is parsed into a MCOperand of type register, and then we get to `24` (which is a valid register, our code currently allows `31

dylanmckay, 2015-03-08:
There is an ambiguity when parsing integer operands without context, because AVR-GCC supports specifiying register names (`r12`) and register numbers (`12`) interchangeably. Without context, we can't tell the difference.

There are two solutions:

* Drop support for numbered registers and only support the `r12` syntax. This would make us incompatible with AVR-GCC.
* Use context to disambiguate between the two cases. We could hook into the tablegen'd `AVRGenInstrInfo.inc`, check if the current operand is a register, then handle it appropriately.

RandomInsano, 2015-03-08:
My opinion is to not support GCC numbered registers until someone comes along who needs it. It sounds like such a confusing idea.

dylanmckay, 2015-03-08:
> My opinion is to not support GCC numbered registers until someone comes along who needs it. It sounds like such a confusing idea.

I concur.

dylanmckay, 2015-03-08:
I have now removed support for numbered registers in d9c85f193f0bf40f776b7a6581c79419dd1bbe7a

closed 2015-03-08 by dylanmckay