avr-llvm/llvm #55: Another victim of issue #53?

RandomInsano, 2015-03-08:
This is the test I've fiddled with:

```
; RUN: llvm-mc -triple avr-none -show-encoding < %s | FileCheck %s

foo:

  sbr r16, 0x00
  sbr r31, 0xff
  
; CHECK: sbr r16, 0x00                 ; encoding: [0x00,0x60]
; CHECK: sbr r31, 0xff                 ; encoding: [0xff,0x6f]
```

I’ve confirmed this via AVR-GCC:

```
sbr r16, 0x00
sbr r31, 0xff
04000000 0060 FF6F 2E
```

But I’m hitting this during the test:

```
<stdin>:6:12: error: invalid operand for instruction
  sbr r16, 0x00
```

Does this have anything to do with issue #53?

dylanmckay, 2015-03-08:
There are no labels in this example, therefore it is not related to the local labels issue.

I will do some debugging to figure out the cause of this issue,

dylanmckay, 2015-03-08:
I believe this is a duplicate of #53 and is now fixed.

closed 2015-03-08 by dylanmckay