avr-llvm/llvm #58: Are we coverered for issues with SBR?

RandomInsano, 2015-03-08:
SBR can only operate on registers R16 to R31, if anything is lowered to use the first sixteen registers they're going to be in trouble since the results will be stored in the wrong location.

I need to find out how to do negative tests in the instruction testing framework.

RandomInsano, 2015-03-08:
I'm guessing "let Constraints" should include something to handle this. If no one is sure, I'll post to the LLVM mailing list.

RandomInsano, 2015-03-08:
It looks like we do fail properly. I just need to learn how to add the proper unit test.

dylanmckay, 2015-03-08:
If you look at `test/CodeGen/Mips/inlineasm-cnstrnt-bad-I-1.ll`, you can see how negative tests are implemented:

By prefixing the `RUN` comment with `not`:
``` asm
; RUN: not llc -march=mipsel < %s  2> %t
```

dylanmckay, 2015-03-08:
I believe we do handle this properly: `llvm-mc` will fail with `sbr r3, 5` with "invalid operand" and `sbr r17, 5` succeeds.

We could open an issue to improve error messages where only a subset of a register class is expected.

RandomInsano, 2015-03-11:
Honestly, that's probably a good idea. I'll go digging when I've got some time.

dylanmckay, 2015-03-12:
I decided to open up a new issue instead - see #67.

closed 2015-03-12 by dylanmckay