avr-llvm/llvm #63: fatal error: error in backend: expected relocatable expression

4ntoine, 2015-03-10:
./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR  -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c -o /tmp/arduino_test1/WInterrupts.c.o

In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c:31:
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:724:12: warning: 
      declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:875:12: warning: 
      declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:888:12: warning: 
      declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
fatal error: error in backend: expected relocatable expression

4ntoine, 2015-03-10:
WInterrupts.c

```
/* -*- mode: jde; c-basic-offset: 2; indent-tabs-mode: nil -*- */

/*
  Part of the Wiring project - http://wiring.uniandes.edu.co

  Copyright (c) 2004-05 Hernando Barragan

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General
  Public License along with this library; if not, write to the
  Free Software Foundation, Inc., 59 Temple Place, Suite 330,
  Boston, MA  02111-1307  USA
  
  Modified 24 November 2006 by David A. Mellis
  Modified 1 August 2010 by Mark Sproul
*/

#include <inttypes.h>
#include <avr/io.h>
#include <avr/interrupt.h>
#include <avr/pgmspace.h>
#include <stdio.h>

#include "wiring_private.h"

static volatile voidFuncPtr intFunc[EXTERNAL_NUM_INTERRUPTS];
// volatile static voidFuncPtr twiIntFunc;

void attachInterrupt(uint8_t interruptNum, void (*userFunc)(void), int mode) {
  if(interruptNum < EXTERNAL_NUM_INTERRUPTS) {
    intFunc[interruptNum] = userFunc;
    
    // Configure the interrupt mode (trigger on low input, any change, rising
    // edge, or falling edge).  The mode constants were chosen to correspond
    // to the configuration bits in the hardware register, so we simply shift
    // the mode into place.
      
    // Enable the interrupt.
      
    switch (interruptNum) {
#if defined(__AVR_ATmega32U4__)
	// I hate doing this, but the register assignment differs between the 1280/2560
	// and the 32U4.  Since avrlib defines registers PCMSK1 and PCMSK2 that aren't 
	// even present on the 32U4 this is the only way to distinguish between them.
    case 0:
	EICRA = (EICRA & ~((1<<ISC00) | (1<<ISC01))) | (mode << ISC00);
	EIMSK |= (1<<INT0);
	break;
    case 1:
	EICRA = (EICRA & ~((1<<ISC10) | (1<<ISC11))) | (mode << ISC10);
	EIMSK |= (1<<INT1);
	break;	
    case 2:
        EICRA = (EICRA & ~((1<<ISC20) | (1<<ISC21))) | (mode << ISC20);
        EIMSK |= (1<<INT2);
        break;
    case 3:
        EICRA = (EICRA & ~((1<<ISC30) | (1<<ISC31))) | (mode << ISC30);
[... 243 lines cut ...]
    intFunc[EXTERNAL_INT_0]();
}

ISR(INT1_vect) {
  if(intFunc[EXTERNAL_INT_1])
    intFunc[EXTERNAL_INT_1]();
}

#if defined(EICRA) && defined(ISC20)
ISR(INT2_vect) {
  if(intFunc[EXTERNAL_INT_2])
    intFunc[EXTERNAL_INT_2]();
}
#endif

#endif

/*
ISR(TWI_vect) {
  if(twiIntFunc)
    twiIntFunc();
}
*/

```

dylanmckay, 2015-06-10:
Reduced testcases:

`WInterrupts.ll`:
```
llc WInterrupts.ll -mcpu=atmega328p -filetype=obj
```
``` asm
; ModuleID = '/usr/share/arduino/hardware/arduino/avr/cores/arduino/WInterrupts.c'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr"

@intFunc = internal global [2 x void ()*] zeroinitializer, align 2
@llvm.used = appending global [2 x i8*] [i8* bitcast (void ()* @__vector_1 to i8*), i8* bitcast (void ()* @__vector_2 to i8*)], section "llvm.metadata"

; Function Attrs: nounwind optsize
define void @attachInterrupt(i8 zeroext %interruptNum, void ()* %userFunc, i16 %mode) #0 {
entry:
  tail call void @llvm.dbg.value(metadata i8 %interruptNum, i64 0, metadata !18, metadata !39), !dbg !40
  tail call void @llvm.dbg.value(metadata void ()* %userFunc, i64 0, metadata !19, metadata !39), !dbg !41
  tail call void @llvm.dbg.value(metadata i16 %mode, i64 0, metadata !20, metadata !39), !dbg !42
  %cmp = icmp ult i8 %interruptNum, 2, !dbg !43
  br i1 %cmp, label %if.then, label %if.end, !dbg !45

if.then:                                          ; preds = %entry
  %conv = zext i8 %interruptNum to i16, !dbg !46
  %arrayidx = getelementptr inbounds [2 x void ()*], [2 x void ()*]* @intFunc, i16 0, i16 %conv, !dbg !47
  store volatile void ()* %userFunc, void ()** %arrayidx, align 2, !dbg !49, !tbaa !50
  switch i8 %interruptNum, label %if.end [
    i8 0, label %sw.bb
    i8 1, label %sw.bb.8
  ], !dbg !54

sw.bb:                                            ; preds = %if.then
  %0 = load volatile i8, i8* inttoptr (i16 105 to i8*), align 1, !dbg !55, !tbaa !57
  %conv3 = zext i8 %0 to i16, !dbg !55
  %and = and i16 %conv3, 252, !dbg !58
  %or = or i16 %and, %mode, !dbg !59
  %conv4 = trunc i16 %or to i8, !dbg !60
  store volatile i8 %conv4, i8* inttoptr (i16 105 to i8*), align 1, !dbg !61, !tbaa !57
  %1 = load volatile i8, i8* inttoptr (i16 61 to i8*), align 1, !dbg !62, !tbaa !57
  %or6 = or i8 %1, 1, !dbg !62
  store volatile i8 %or6, i8* inttoptr (i16 61 to i8*), align 1, !dbg !62, !tbaa !57
  br label %if.end, !dbg !63

sw.bb.8:                                          ; preds = %if.then
  %2 = load volatile i8, i8* inttoptr (i16 105 to i8*), align 1, !dbg !64, !tbaa !57
  %conv9 = zext i8 %2 to i16, !dbg !64
  %and10 = and i16 %conv9, 243, !dbg !65
  %shl11 = shl i16 %mode, 2, !dbg !66
  %or12 = or i16 %and10, %shl11, !dbg !67
  %conv13 = trunc i16 %or12 to i8, !dbg !68
  store volatile i8 %conv13, i8* inttoptr (i16 105 to i8*), align 1, !dbg !69, !tbaa !57
  %3 = load volatile i8, i8* inttoptr (i16 61 to i8*), align 1, !dbg !70, !tbaa !57
  %or15 = or i8 %3, 2, !dbg !70
  store volatile i8 %or15, i8* inttoptr (i16 61 to i8*), align 1, !dbg !70, !tbaa !57
  br label %if.end, !dbg !71

if.end:                                           ; preds = %sw.bb, %sw.bb.8, %if.then, %entry
  ret void, !dbg !72
}

; Function Attrs: nounwind optsize
define void @detachInterrupt(i8 zeroext %interruptNum) #0 {
entry:
  tail call void @llvm.dbg.value(metadata i8 %interruptNum, i64 0, metadata !25, metadata !39), !dbg !73
  %conv = zext i8 %interruptNum to i16, !dbg !74
  %cmp = icmp ult i8 %interruptNum, 2, !dbg !76
  br i1 %cmp, label %if.then, label %if.end, !dbg !77

if.then:                                          ; preds = %entry
[... 372 lines cut ...]
	pop	r30
	pop	r27
	pop	r26
	pop	r25
	pop	r24
	pop	r23
	pop	r22
	pop	r21
	pop	r20
	pop	r19
	pop	r18
	pop	r1
	pop	r0
	pop	r0
	out	63, r0
	pop	r1
	pop	r0
	reti
.Lfunc_end3:
	.size	__vector_2, .Lfunc_end3-__vector_2

	.type	intFunc,@object         ; @intFunc
	.local	intFunc
	.comm	intFunc,4,2
```

dylanmckay, 2015-06-10:
This is caused because `AVRMCExpr::evaluateAsRelocatableImpl` doesn't handle the case where the `hi8(..)`, `lo8(..)` fixups have non-absolute immediates as targets (i.e. a label as a target).

agnat, 2015-06-11:
Hehe, I'm already on it. Here is _my_ test case:

````
foo: ldi r24, lo8(foo)
````

The whole fixup-relocation-expression-business seems a bit out of whack. Note, that this is probably (partly) my doing, since I beefed up AVRMCExpr a bit, left a dangling TODO... yadda yadda

dylanmckay, 2015-06-11:
*Your* testcase is nice, but it could use some work :)

I have strong doubts that you had anything to do with this bug due to how old it is.

closed 2015-06-12 by dylanmckay

4ntoine, 2015-06-13:
still having the issue:
```
MBA-Anton:bin asmirnov$ ./clang --version
clang version 3.7.0 (https://github.com/avr-llvm/clang.git aebd4f6d15e44635ecc7338d4454389c4573bc2b) (llvm/llvm 9c7c5212863ed8b06323ad07fa84bad29ff051f8)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
```

```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR  -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c -o /tmp/arduino_test1/WInterrupts.c.o
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c:31:
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:724:12: warning: 
      declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:875:12: warning: 
      declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:888:12: warning: 
      declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
fatal error: error in backend: expected relocatable expression
```

dylanmckay, 2015-06-13:
Can you try `llvm-mc --triple avr -mcpu=atmega328p -filetype=obj` and type `ldi r24, lo8(main)` and then `Enter Ctrl+D Enter`? does this succeed?

commit bccaf2a8 (2015-06-12): [AVR] Add relocation addend and handle symbols

Fix #63.