avr-llvm/llvm #64: error: couldn't allocate output register for constraint 'w'

4ntoine, 2015-03-10:
./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR  -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring.c -o /tmp/arduino_test1/wiring.c.o

...

Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring.c:185:3: error: 
      couldn't allocate output register for constraint 'w'
                "1: sbiw %0,1" "\n\t" // 2 cycles

```
...
// busy wait
	__asm__ __volatile__ (
		"1: sbiw %0,1" "\n\t" // 2 cycles
		"brne 1b" : "=w" (us) : "0" (us) // 2 cycles
	);
}
...
```

dylanmckay, 2015-04-23:
Test case (`main.c`):
``` c
int main() {
    volatile int us = 12;
    __asm__ __volatile__ (
        "1: sbiw %0,1" "\n\t" // 2 cycles
        "brne 1b" : "=w" (us) : "0" (us) // 2 cycles
    );
}
```

Command line:
``` bash
./clang --target avr-none main.c
```

dylanmckay, 2015-04-23:
When compiling this, currently issue #37 is encountered. When that issue is fixed, check if this problem still persists.

dylanmckay, 2015-04-23:
This exists in this repository as opposed to `avr-llvm/clang` because the assertion is triggered in `llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp`.

If it turns out the problem is caused by `clang`, move the issue there.

dylanmckay, 2015-04-24:
Issue #37 is now fixed. When compiling the snippet, issue #77 is hit.

I believe that this issue is currently fixed. Closing.

closed 2015-04-24 by dylanmckay

4ntoine, 2015-06-13:
still having this issue, but now crash instead of outputting error message:

```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR  -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring.c -o /tmp/arduino_test1/wiring.c.o
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring.c:25:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_private.h:30:
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:724:12: warning: 
      declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:875:12: warning: 
      declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:888:12: warning: 
      declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
Assertion failed: (Idx && Idx < getNumSubRegIndices() && "This is not a subregister index"), function getSubReg, file /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/lib/MC/MCRegisterInfo.cpp, line 28.
0  clang-3.7                0x0000000110f993ae llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 46
1  clang-3.7                0x0000000110f9ac79 PrintStackTraceSignalHandler(void*) + 25
2  clang-3.7                0x0000000110f9b0d8 SignalHandler(int) + 584
3  libsystem_platform.dylib 0x00007fff8f4ddf1a _sigtramp + 26
4  clang-3.7                0x0000000113ea3756 clang::Stmt::StatisticsEnabled + 91697
5  clang-3.7                0x0000000110f9ac9b raise + 27
6  clang-3.7                0x0000000110f9ad52 abort + 18
7  clang-3.7                0x0000000110f9ad31 __assert_rtn + 129
8  clang-3.7                0x0000000110bba871 llvm::MCRegisterInfo::getSubReg(unsigned int, unsigned int) const + 145
9  clang-3.7                0x000000010ffbec15 llvm::AVRInstPrinter::getPrettyRegisterName(unsigned int, llvm::MCRegisterInfo const&) + 101
10 clang-3.7                0x000000010ff83b01 (anonymous namespace)::AVRAsmPrinter::printOperand(llvm::MachineInstr const*, unsigned int, llvm::raw_ostream&, char const*) + 209
11 clang-3.7                0x000000010ff837e3 (anonymous namespace)::AVRAsmPrinter::PrintAsmOperand(llvm::MachineInstr const*, unsigned int, unsigned int, char const*, llvm::raw_ostream&) + 883
12 clang-3.7                0x00000001114bbf90 EmitGCCInlineAsmStr(char const*, llvm::MachineInstr const*, llvm::MachineModuleInfo*, int, int, llvm::AsmPrinter*, unsigned int, llvm::raw_ostream&) + 3824
13 clang-3.7                0x00000001114baec3 llvm::AsmPrinter::EmitInlineAsm(llvm::MachineInstr const*) const + 1347
14 clang-3.7                0x000000011149ae46 llvm::AsmPrinter::EmitFunctionBody() + 1206
15 clang-3.7                0x000000010ff83cd8 llvm::AsmPrinter::runOnMachineFunction(llvm::MachineFunction&) + 56
16 clang-3.7                0x00000001103e298e llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 110
17 clang-3.7                0x000000011082d86d llvm::FPPassManager::runOnFunction(llvm::Function&) + 413
18 clang-3.7                0x000000011082db78 llvm::FPPassManager::runOnModule(llvm::Module&) + 104
19 clang-3.7                0x000000011082e573 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) + 1395
20 clang-3.7                0x000000011082de2e llvm::legacy::PassManagerImpl::run(llvm::Module&) + 302
21 clang-3.7                0x000000011082ecf1 llvm::legacy::PassManager::run(llvm::Module&) + 33
22 clang-3.7                0x0000000111852358 (anonymous namespace)::EmitAssemblyHelper::EmitAssembly(clang::BackendAction, llvm::raw_pwrite_stream*) + 1592
23 clang-3.7                0x0000000111851b32 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::raw_pwrite_stream*) + 114
24 clang-3.7                0x0000000111aacf5f clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) + 1375
25 clang-3.7                0x0000000112608baf clang::ParseAST(clang::Sema&, bool, bool) + 1263
26 clang-3.7                0x000000011141caba clang::ASTFrontendAction::ExecuteAction() + 522
27 clang-3.7                0x0000000111aaafb3 clang::CodeGenAction::ExecuteAction() + 3987
28 clang-3.7                0x000000011141c030 clang::FrontendAction::Execute() + 112
[... cut ...]
clang-3.7: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.7.0 (https://github.com/avr-llvm/clang.git aebd4f6d15e44635ecc7338d4454389c4573bc2b) (llvm/llvm 9c7c5212863ed8b06323ad07fa84bad29ff051f8)
Target: avr
Thread model: posix
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.7: note: diagnostic msg: 
********************

```

4ntoine, 2015-06-13:
let me know if anything (files, dumps, assistance in reproducing) needed

dylanmckay, 2015-06-13:
@agnat The failure path includes your `getPrettyRegisterName` function - do you have any ideas?

dylanmckay, 2015-06-13:
@4ntoine Could you create an LLVM IR file (`clang -S -emit-llvm -o main.ll`) and post it? They are a fair amount easier to debug.

dylanmckay, 2015-06-13:
Could you also create a seperate issue for the problem you just mentioned because it is a different error and probably unrelated.

agnat, 2015-06-14:
> [...] do you have any ideas?

Hehe... no, not really. Not without actually tracing it. Which I will do at some point but probably not right now.

> let me know if anything (files, dumps, assistance in reproducing) needed

Thanks, @4ntoine. But what we'd really need is reduction. Like @dylanmckay said, providing LLVM IR files is a good first step. We do not usually build clang because it's large and slow and... in the way. If you really want to kick it try reducing your code to the smallest possible example that still shows _the same error_.  (Yes, that means ripping apart `/Applications/Arduino.app/Contents/Resources/Java/hardware/`).