avr-llvm/llvm #65: fatal error: error in backend: Cannot select:

4ntoine, 2015-03-10:
./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_analog.c -o /tmp/arduino_test1/wiring_analog.c.o

```
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_analog.c:120:10: error: 
      couldn't allocate output register for constraint 'r'
                switch(digitalPinToTimer(pin))
                       ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/Arduino.h:170:32: note: 
      expanded from macro 'digitalPinToTimer'
#define digitalPinToTimer(P) ( pgm_read_byte( digital_pin_to_timer_PGM + (P) ) )
                               ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:1046:41: note: 
      expanded from macro 'pgm_read_byte'
#define pgm_read_byte(address_short)    pgm_read_byte_near(address_short)
                                        ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:638:43: note: 
      expanded from macro 'pgm_read_byte_near'
#define pgm_read_byte_near(address_short) __LPM((uint16_t)(address_short))
                                          ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:626:29: note: 
      expanded from macro '__LPM'
#define __LPM(addr)         __LPM_classic__(addr)
                            ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:413:9: note: 
      expanded from macro '__LPM_classic__'
        "lpm" "\n\t"            \
        ^
```

4ntoine, 2015-03-10:
fatal error: error in backend: Cannot select: 0x7fe164117040: i16 = JumpTable<0> [ID=2]
In function: analogWrite
clang-3.7: error: clang frontend command failed with exit code 70 (use -v to see invocation)
clang version 3.7.0 (https://github.com/avr-llvm/clang.git 98cf874e96493bec6b96f3a8339f11150f4f198b) (llvm/llvm b72e68d6d2cd7b0b9d2aa0d654289b31d8007609)
Target: avr
Thread model: posix
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.7: note: diagnostic msg: 
********************

dylanmckay, 2015-03-10:
Duplicate of issue #46

closed 2015-03-10 by dylanmckay

4ntoine, 2015-06-13:
this can be duplicate, but it still happens:
```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_analog.c -o /tmp/arduino_test1/wiring_analog.c.o
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_analog.c:27:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_private.h:30:
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:724:12: warning: 
      declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:875:12: warning: 
      declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:888:12: warning: 
      declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
fatal error: error in backend: Cannot select: 0x7fc69b0cbd80: i16 = JumpTable<0> [ID=2]
In function: analogWrite
clang-3.7: error: clang frontend command failed with exit code 70 (use -v to see invocation)
clang version 3.7.0 (https://github.com/avr-llvm/clang.git aebd4f6d15e44635ecc7338d4454389c4573bc2b) (llvm/llvm 9c7c5212863ed8b06323ad07fa84bad29ff051f8)
Target: avr
Thread model: posix
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.7: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/wiring_analog-8260ee.c
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/wiring_analog-8260ee.sh
clang-3.7: note: diagnostic msg: 

********************

```

dylanmckay, 2015-06-13:
I reopened #46, and made the title more suitable.