avr-llvm/llvm #66: assertion failed: (R == 0 && "Already initialized this value register!")

4ntoine, 2015-03-10:
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c -o /tmp/arduino_test1/wiring_digital.c.o
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c:28:
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_private.h:30:
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:724:12: warning: 
      declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:875:12: warning: 
      declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/stdio.h:888:12: warning: 
      declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c:33:16: error: 
      couldn't allocate output register for constraint 'r'
        uint8_t bit = digitalPinToBitMask(pin);
                      ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/Arduino.h:169:34: note: 
      expanded from macro 'digitalPinToBitMask'
#define digitalPinToBitMask(P) ( pgm_read_byte( digital_pin_to_bit_mask_PGM + (P) ) )
                                 ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:1046:41: note: 
      expanded from macro 'pgm_read_byte'
#define pgm_read_byte(address_short)    pgm_read_byte_near(address_short)
                                        ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:638:43: note: 
      expanded from macro 'pgm_read_byte_near'
#define pgm_read_byte_near(address_short) __LPM((uint16_t)(address_short))
                                          ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:626:29: note: 
      expanded from macro '__LPM'
#define __LPM(addr)         __LPM_classic__(addr)
                            ^
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:413:9: note: 
      expanded from macro '__LPM_classic__'
        "lpm" "\n\t"            \
        ^
Assertion failed: (R == 0 && "Already initialized this value register!"), function InitializeRegForValue, file /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/include/llvm/CodeGen/FunctionLoweringInfo.h, line 158.
0  clang-3.7                0x000000010976356e llvm::sys::PrintStackTrace(__sFILE*) + 46
1  clang-3.7                0x000000010976492b PrintStackTraceSignalHandler(void*) + 27
2  clang-3.7                0x0000000109764d75 SignalHandler(int) + 565
3  libsystem_platform.dylib 0x00007fff9398a5aa _sigtramp + 26
4  libsystem_platform.dylib 0x00007f8969927460 _sigtramp + 3589918416
5  clang-3.7                0x000000010976495b raise + 27
6  clang-3.7                0x0000000109764a12 abort + 18
7  clang-3.7                0x00000001097649f1 __assert_rtn + 129
8  clang-3.7                0x0000000109f2f209 llvm::FunctionLoweringInfo::InitializeRegForValue(llvm::Value const*) + 137
9  clang-3.7                0x0000000109eed9f9 llvm::SelectionDAGBuilder::getValueImpl(llvm::Value const*) + 5257
10 clang-3.7                0x0000000109eedc0c llvm::SelectionDAGBuilder::getNonRegisterValue(llvm::Value const*) + 108
11 clang-3.7                0x0000000109eef87a llvm::SelectionDAGBuilder::CopyValueToVirtualRegister(llvm::Value const*, unsigned int) + 74
12 clang-3.7                0x0000000109edd120 llvm::SelectionDAGBuilder::CopyToExportRegsIfNeeded(llvm::Value const*) + 288
13 clang-3.7                0x0000000109edbfbd llvm::SelectionDAGBuilder::visit(llvm::Instruction const&) + 189
14 clang-3.7                0x0000000109f5ac4c llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator<llvm::Instruction const>, llvm::ilist_iterator<llvm::Instruction const>, bool&) + 140
15 clang-3.7                0x0000000109f5a9d3 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) + 5155
16 clang-3.7                0x0000000109f58285 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 1141
17 clang-3.7                0x0000000108c84c5e llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 110
18 clang-3.7                0x00000001090880ab llvm::FPPassManager::runOnFunction(llvm::Function&) + 427
19 clang-3.7                0x00000001090883b8 llvm::FPPassManager::runOnModule(llvm::Module&) + 104
20 clang-3.7                0x0000000109088dc4 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) + 1412
21 clang-3.7                0x000000010908866e llvm::legacy::PassManagerImpl::run(llvm::Module&) + 302
22 clang-3.7                0x0000000109089541 llvm::legacy::PassManager::run(llvm::Module&) + 33
23 clang-3.7                0x0000000109fb9121 (anonymous namespace)::EmitAssemblyHelper::EmitAssembly(clang::BackendAction, llvm::raw_ostream*) + 1601
24 clang-3.7                0x0000000109fb88e2 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::raw_ostream*) + 114
25 clang-3.7                0x000000010a1dec25 clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) + 1397
26 clang-3.7                0x000000010ad1101b clang::ParseAST(clang::Sema&, bool, bool) + 1291
[... 7 lines cut ...]
34 clang-3.7                0x00000001088430da main + 1274
35 libdyld.dylib            0x00007fff8fb515fd start + 1
36 libdyld.dylib            0x0000000000000041 start + 1883957829
Stack dump:
0.	Program arguments: /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/build/bin/clang-3.7 -cc1 -triple avr -emit-obj -disable-free -main-file-name wiring_digital.c -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -target-cpu atmega328p -target-linker-version 241.9 -g -dwarf-column-info -ffunction-sections -fdata-sections -coverage-file /tmp/arduino_test1/wiring_digital.c.o -resource-dir /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/build/bin/../lib/clang/3.7.0 -dependency-file /tmp/arduino_test1/wiring_digital.c.d -MT /tmp/arduino_test1/wiring_digital.c.o -D F_CPU=16000000L -D USB_VID=null -D USB_PID=null -D ARDUINO=105 -I /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include -Os -Wall -fdebug-compilation-dir /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/build/bin -ferror-limit 19 -fmessage-length 115 -mstackrealign -fobjc-runtime=gcc -fdiagnostics-show-option -fcolor-diagnostics -vectorize-loops -vectorize-slp -o /tmp/arduino_test1/wiring_digital.c.o -x c /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c 
1.	<eof> parser at end of file
2.	Code generation
3.	Running pass 'Function Pass Manager' on module '/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c'.
4.	Running pass 'AVR DAG->DAG Instruction Selection' on function '@pinMode'
clang-3.7: error: unable to execute command: Illegal instruction: 4
[... cut ...]
clang-3.7: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.7: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/wiring_digital-48fbde.c
clang-3.7: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/wiring_digital-48fbde.sh
clang-3.7: note: diagnostic msg: 

********************

4ntoine, 2015-03-10:
```
void pinMode(uint8_t pin, uint8_t mode)
{
	uint8_t bit = digitalPinToBitMask(pin);
	uint8_t port = digitalPinToPort(pin);
	volatile uint8_t *reg, *out;

	if (port == NOT_A_PIN) return;

	// JWS: can I let the optimizer do this?
	reg = portModeRegister(port);
	out = portOutputRegister(port);

	if (mode == INPUT) { 
		uint8_t oldSREG = SREG;
                cli();
		*reg &= ~bit;
		*out &= ~bit;
		SREG = oldSREG;
	} else if (mode == INPUT_PULLUP) {
		uint8_t oldSREG = SREG;
                cli();
		*reg &= ~bit;
		*out |= bit;
		SREG = oldSREG;
	} else {
		uint8_t oldSREG = SREG;
                cli();
		*reg |= bit;
		SREG = oldSREG;
	}
}
```

4ntoine, 2015-03-10:
i will send the files to test if necessary (too big to copy/paste here)

RandomInsano, 2015-03-11:
Pastebin? It has syntax highlighting too!

dylanmckay, 2015-07-17:
Arduino successfully compiles now.

closed 2015-07-17 by dylanmckay

4ntoine, 2015-08-05:
that's great news, i have to check it

dylanmckay, 2015-08-05:
I believe #149 is the only issue leading to a broken Arduino. See the comment on the issue for details.