avr-llvm/llvm #70: Arithmetic operations are being incorrectly lowered

dylanmckay, 2015-03-25:
We are not handling the lowering of various instructions correctly.

This problem came about in [this](http://sourceforge.net/p/avr-llvm-backend/mailman/avr-llvm-backend-mail/thread/550F76D3.8070803%40uwaterloo.ca/) mailing list exchange.

Division is being lowered into a call to the `libgcc` function `__udivqi3`. This function does not exist. GCC instead lowers this to `__divmodhi4`, which is what we should be doing.

dylanmckay, 2015-04-23:
We should be able to handle this by adding a case to the switch in `AVRISelLowering::LowerOperation()`. As @simonpcook said, custom lowering the divide may be the best solution.

See the `ARMISelLowering::LowerOperation()` for an example.

dylanmckay, 2015-05-07:
Note that a number of the compiler runtime library functions do not follow the normal AVR ABI.

See the [avr-gcc wiki](https://gcc.gnu.org/wiki/avr-gcc), heading "Exceptions to the calling convention" for details.

dylanmckay, 2015-05-07:
I believe the best way would be to create a custom calling convention which special cases all of the routines from the AVR-GCC wiki which do not follow the standard convention. This way libcalls would work as expected.

dylanmckay, 2015-07-19:
I have implemented 8 and 16-bit division using custom lowering.

We cannot custom lower a 32-bit (or greater) values because the type is illegal. I am not sure how to go about this.

closed 2015-07-22 by dylanmckay

agnat, 2015-07-22:
The follow-up is #149.

> We cannot custom lower a 32-bit (or greater) values because the type is illegal. I am not sure how to go about this.

Hehe... The whole concept of "legal types" is ... I wouldn't say flawed... but it just doesn't cut it for small targets. It is exactly what I'm trying to hack up to get rid of the pseudo instructions. If I succeed handling the division library calls shouldn't be a problem.