avr-llvm/llvm #72: Backend data layout does not match clang expected target description

seanmkauffman, 2015-04-08:
I'm compiling from a fresh checkout of avr-llvm with clang checked out in tools.  When I go and try to compile something to IR I get the following error:

```bash
~/workspace/avr-work/test $ ../bin/clang -O1 --target=avr-none-none -emit-llvm test.c -S -o test.ll
error: backend data layout 'e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8'
      does not match expected target description 'e-p:16:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8'
1 error generated.
```

This is new, as I have successfully built and used clang/opt/llc from this setup before.  I tried a clean rebuild and it did not fix the problem.  I also tried just manually changing the string in AVRTargetMachine.cpp to the one that clang is evidently expecting and that caused me to be able to generate IR again.  It doesn't seem that I can compile anything down to AVR at the moment with avr-llvm, though, with or without that workaround.

In case you're wondering, it doesn't matter what the source file contains.  In the case above it contained this:

```C
~/workspace/avr-work/test $ cat test.c 
int main(void) {
  return 0;
}
```

I have verified that running clang with a number of officially supported targets (x86_64-unknown-linux-gnu, for example) works fine.

RandomInsano, 2015-04-08:
Here's a silly question, but did you checkout vanilla Clang, or the one within this organization?

I think I need to implement CI for the rest of the projects now.

seanmkauffman, 2015-04-08:
The one in the wiki instructions.
```bash
~/workspace/llvm/tools/clang $ git remote -v
origin	https://github.com/avr-llvm/clang.git (fetch)
origin	https://github.com/avr-llvm/clang.git (push)
```

dylanmckay, 2015-04-09:
Confirmed; I am getting this error as well.

dylanmckay, 2015-04-09:
The problem is fixed in `avr-llvm/clang` commit [https://github.com/avr-llvm/clang/commit/00a0b3863ac817790d9b344da4fc499b71140a7a](00a0b3863ac817790d9b344da4fc499b71140a7a).

dylanmckay, 2015-04-09:
Clang had a different target layout than LLVM did for AVR. Now Clang has the same datalayout as LLVM.

dylanmckay, 2015-04-09:
@seanmkauffman

Note that it is no longer necessary to pass `-O1` to clang, because issue #15 has been fixed.

closed 2015-04-09 by dylanmckay

dylanmckay, 2015-04-09:
@RandomInsano 

> I think I need to implement CI for the rest of the projects now.

Currently there are about 10 failing machine code tests for AVR (out of about 85). The majority of these tests are failing because we aren't parsing immediates prefixed with `+` or `-` properly (for example, `rjmp 4` works, but `rjmp -4` and `rjmp +4` does not. I believe there is a way to disable tests, so perhaps we can disable the failing tests and then enable CI.