avr-llvm/llvm #75: Crash during compilation to assembly

seanmkauffman, 2015-04-22:
llc fourierf.ll -mtriple avr-none -o fourierf.s

```
llc: /home/seanmk/workspace/llvm/lib/Target/AVR/AVRISelLowering.cpp:1137: virtual llvm::SDValue llvm::AVRTargetLowering::LowerCall(TargetLowering::CallLoweringInfo &, SmallVectorImpl<llvm::SDValue> &) const: Assertion `Mask && "Missing call preserved mask for calling convention"' failed.
#0 0x1ffb4ae llvm::sys::PrintStackTrace(llvm::raw_ostream&) /home/seanmk/workspace/llvm/lib/Support/Unix/Signals.inc:424:15
#1 0x1ffc509 PrintStackTraceSignalHandler(void*) /home/seanmk/workspace/llvm/lib/Support/Unix/Signals.inc:483:1
#2 0x1fff5a4 SignalHandler(int) /home/seanmk/workspace/llvm/lib/Support/Unix/Signals.inc:199:60
#3 0x7fb68080a250 __restore_rt (/lib64/libpthread.so.0+0x11250)
#4 0x7fb67f80d795 gsignal (/lib64/libc.so.6+0x38795)
#5 0x7fb67f80ec18 abort (/lib64/libc.so.6+0x39c18)
#6 0x7fb67f806812 (/lib64/libc.so.6+0x31812)
#7 0x7fb67f8068c2 (/lib64/libc.so.6+0x318c2)
#8 0xbdf3cb llvm::AVRTargetLowering::LowerCall(llvm::TargetLowering::CallLoweringInfo&, llvm::SmallVectorImpl<llvm::SDValue>&) const /home/seanmk/workspace/llvm/lib/Target/AVR/AVRISelLowering.cpp:1137:3
#9 0x146f9d8 llvm::TargetLowering::LowerCallTo(llvm::TargetLowering::CallLoweringInfo&) const /home/seanmk/workspace/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:7309:15
#10 0x1471307 llvm::SelectionDAGBuilder::lowerInvokable(llvm::TargetLowering::CallLoweringInfo&, llvm::MachineBasicBlock*) /home/seanmk/workspace/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:5484:3
#11 0x1455a0f llvm::SelectionDAGBuilder::LowerCallTo(llvm::ImmutableCallSite, llvm::SDValue, bool, llvm::MachineBasicBlock*) /home/seanmk/workspace/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:5550:39
#12 0x1443740 llvm::SelectionDAGBuilder::visitCall(llvm::CallInst const&) /home/seanmk/workspace/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:6086:1
#13 0x143ad2e llvm::SelectionDAGBuilder::visit(unsigned int, llvm::User const&) /home/seanmk/workspace/llvm/include/llvm/IR/Instruction.def:165:1
#14 0x143a223 llvm::SelectionDAGBuilder::visit(llvm::Instruction const&) /home/seanmk/workspace/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:967:3
#15 0x149f95c llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator<llvm::Instruction const>, llvm::ilist_iterator<llvm::Instruction const>, bool&) /home/seanmk/workspace/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:601:5
#16 0x149f3e1 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) /home/seanmk/workspace/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:1328:5
#17 0x149d287 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) /home/seanmk/workspace/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:464:33
#18 0xbd041b (anonymous namespace)::AVRDAGToDAGISel::runOnMachineFunction(llvm::MachineFunction&) /home/seanmk/workspace/llvm/lib/Target/AVR/AVRISelDAGToDAG.cpp:70:14
#19 0x170d2ee llvm::MachineFunctionPass::runOnFunction(llvm::Function&) /home/seanmk/workspace/llvm/lib/CodeGen/MachineFunctionPass.cpp:40:10
#20 0x1f31100 llvm::FPPassManager::runOnFunction(llvm::Function&) /home/seanmk/workspace/llvm/lib/IR/LegacyPassManager.cpp:1538:23
#21 0x1f31408 llvm::FPPassManager::runOnModule(llvm::Module&) /home/seanmk/workspace/llvm/lib/IR/LegacyPassManager.cpp:1558:16
#22 0x1f31adc (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) /home/seanmk/workspace/llvm/lib/IR/LegacyPassManager.cpp:1616:23
#23 0x1f316be llvm::legacy::PassManagerImpl::run(llvm::Module&) /home/seanmk/workspace/llvm/lib/IR/LegacyPassManager.cpp:1723:16
#24 0x1f320a1 llvm::legacy::PassManager::run(llvm::Module&) /home/seanmk/workspace/llvm/lib/IR/LegacyPassManager.cpp:1756:10
#25 0x751cb0 compileModule(char**, llvm::LLVMContext&) /home/seanmk/workspace/llvm/tools/llc/llc.cpp:351:3
#26 0x750bf2 main /home/seanmk/workspace/llvm/tools/llc/llc.cpp:200:22
#27 0x7fb67f7f9db5 __libc_start_main (/lib64/libc.so.6+0x24db5)
#28 0x74e3c9 _start (/home/seanmk/workspace/avr-work-newer/bin/llc+0x74e3c9)
Stack dump:
0.	Program arguments: /home/seanmk/workspace/avr-work-newer/bin/llc -mtriple=avr-none fourierf.ll -o fourierf.s 
1.	Running pass 'Function Pass Manager' on module 'fourierf.ll'.
2.	Running pass 'AVR DAG->DAG Instruction Selection' on function '@fft_float'
```

fourierf.ll:
```
; ModuleID = 'fourierf.c'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr-none-none"

%struct.__file = type { i8*, i8, i8, i16, i16, i16 (i8, %struct.__file*)*, i16 (%struct.__file*)*, i8* }

@.str = private unnamed_addr constant [7 x i8] c"RealIn\00", align 1
@.str1 = private unnamed_addr constant [8 x i8] c"RealOut\00", align 1
@.str2 = private unnamed_addr constant [8 x i8] c"ImagOut\00", align 1
@__iob = external global [0 x %struct.__file*]
@.str3 = private unnamed_addr constant [35 x i8] c"Error in fft_float():  %s == NULL\0A\00", align 1

; Function Attrs: nounwind
define void @fft_float(i16 %NumSamples, i16 %InverseTransform, float* readonly %RealIn, float* readonly %ImagIn, float* %RealOut, float* %ImagOut) #0 {
entry:
  %tobool = icmp ne i16 %InverseTransform, 0
  %. = select i1 %tobool, double 0xC01921FB54442D18, double 0x401921FB54442D18
  %0 = bitcast float* %RealIn to i8*
  tail call fastcc void @CheckPointer(i8* %0, i8* getelementptr inbounds ([7 x i8], [7 x i8]* @.str, i16 0, i16 0))
  %1 = bitcast float* %RealOut to i8*
  tail call fastcc void @CheckPointer(i8* %1, i8* getelementptr inbounds ([8 x i8], [8 x i8]* @.str1, i16 0, i16 0))
  %2 = bitcast float* %ImagOut to i8*
  tail call fastcc void @CheckPointer(i8* %2, i8* getelementptr inbounds ([8 x i8], [8 x i8]* @.str2, i16 0, i16 0))
  %call = tail call i16 @NumberOfBitsNeeded(i16 %NumSamples) #4
  %cmp195 = icmp eq i16 %NumSamples, 0
  br i1 %cmp195, label %for.end94, label %for.body.lr.ph

for.body.lr.ph:                                   ; preds = %entry
  %cmp3 = icmp eq float* %ImagIn, null
[... 165 lines cut ...]

; Function Attrs: nounwind
declare i16 @fprintf(%struct.__file* nocapture, i8* nocapture readonly, ...) #0

; Function Attrs: noreturn
declare void @exit(i16) #3

attributes #0 = { nounwind "less-precise-fpmad"="false" "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "stack-protector-buffer-size"="8" "unsafe-fp-math"="false" "use-soft-float"="false" }
attributes #1 = { "less-precise-fpmad"="false" "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "stack-protector-buffer-size"="8" "unsafe-fp-math"="false" "use-soft-float"="false" }
attributes #2 = { nounwind readnone "less-precise-fpmad"="false" "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "stack-protector-buffer-size"="8" "unsafe-fp-math"="false" "use-soft-float"="false" }
attributes #3 = { noreturn "less-precise-fpmad"="false" "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "stack-protector-buffer-size"="8" "unsafe-fp-math"="false" "use-soft-float"="false" }
attributes #4 = { nounwind }
attributes #5 = { nounwind readnone }
attributes #6 = { noreturn nounwind }

!llvm.ident = !{!0}

!0 = !{!"clang version 3.7.0 (https://github.com/avr-llvm/clang.git 00a0b3863ac817790d9b344da4fc499b71140a7a) (llvm/llvm.git 70f966eb9eb5e2cae5d4f62ae213ae035677efee)"}
!1 = !{!2, !2, i64 0}
!2 = !{!"float", !3, i64 0}
!3 = !{!"omnipotent char", !4, i64 0}
!4 = !{!"Simple C/C++ TBAA"}
!5 = !{!6, !6, i64 0}
!6 = !{!"any pointer", !3, i64 0}
```

dylanmckay, 2015-04-23:
This was caused because the `AVRRegisterInfo::getCallPreservedMask` function was not actually overriding the virtual function it should have been, because the signature had been changed and `AVRRegisterInfo::getCallPreservedMask` was not qualified with the `override` keyword, so the problem happened silently.

I have audited all classes in the backend, and added missing `override` keywords where they were needed.

closed 2015-04-23 by dylanmckay

dylanmckay, 2015-04-23:
Note that when compiling your test case, compilation fails with another error: "instruction selection failed".

This is presumably because the MCU targeted (by default, `avr2`) does not support the operations used (such as multiplication/division, I'm not sure which).

If you select one of the higher end MCUs (such as `avrxmega7`, by using `llc fourierf.ll -march=avr -mcpu=avrxmega7`), instruction selection completes, but another error is hit - #37.

dylanmckay, 2015-04-23:
I have opened up an issue for the instruction selection error - #76.

seanmkauffman, 2015-04-23:
Thanks for the help.  FYI, avr-llvm will not compile right now due to an error in the clang branch.  I filed an issue. https://github.com/avr-llvm/clang/issues/3

dylanmckay, 2015-04-24:
Now that #37 is fixed, I can run `llc` on `fourierf.ll` and it succeeds.

commit 3539d69c (2015-04-23): [AVR] Added `override` qualifiers to methods

This uncovered a few bugs wherein methods which were supposed to be
overrided were not being overriden.

I also cleaned up a few comments in the process.

Closes #75.