avr-llvm/llvm #88: Implement jump tables

dylanmckay, 2015-05-07:
Implement [jump table](https://en.wikipedia.org/wiki/Branch_table) support.

dylanmckay, 2015-07-08:
This blocks Arduino because `analogWrite` in `wiring_analoc.c` uses jump tables.

dylanmckay, 2015-07-09:
I have disabled jump table emission in 177a73b6b81a3f631c17e4c3af36853b9873dada, so this no longer blocks Arduino.

4ntoine, 2015-08-17:
Will serial interruptions work? F.e. serial interruptions https://www.arduino.cc/en/Tutorial/SerialEvent

dylanmckay, 2015-08-17:
Jump table emission is disabled - now switch statements are expanded into a set of `if-else` statements. This means that this issue does not stop any code from working.

w.r.t `SerialEvent` - it probably will not work. As stated in #149, `Serial.begin()` uses 32-bit division and so it will fail to compile (i.e. the bug you opened last).

agnat, 2015-08-17:
What @dylanmckay says.

In general however interrupts _should_ work, if that is what you're asking. The interrupt table is not a jump table. "Should" as in I haven't actually tried it... ;)

4ntoine, 2015-08-17:
Oh, i see

dylanmckay, 2015-08-19:
Heads up - #149 is now resolved!

dylanmckay, 2016-12-09:
Jump tables is an optimisation, we can look into it in the future.

For now I'll close this and we can look into it in the distant future.

closed 2016-12-09 by dylanmckay