avr-llvm/llvm #98: Print the lower register when printing register pairs

dylanmckay, 2015-06-04:
As of PR #96, the `MOVW Rd+1:Rd, Rr+1:Rr` instruction is properly parser.

The problem is that GCC does not use this syntax. GCC prints the lower register to indicate the pair. For example, we print `movw r17:r16, r19:r18` but GCC prints `movw r16, r18`.

Our assembly parser correctly handles both syntaxes. We should however print the latter syntax with the lower register, because otherwise GCC will not be able to compile assembly files generated by AVR-LLVM.

This should be as simple as implementing a custom operand printing function in `AVRInstPrinter.cpp`.

agnat, 2015-06-05:
Done in c095afadcd2898f88616731b5b6450b22b0633aa.

dylanmckay, 2015-06-07:
I believe you are right, you're on a roll!

closed 2015-06-07 by dylanmckay