avr-llvm/llvm #100: Assertion error during instruction selection

dylanmckay, 2015-06-05:
Command line:
```
llc -filetype=asm -march=avr -mcpu=atmega328p fourier.ll -o fourier.s
```

(`fourier.ll`):
``` asm
; ModuleID = 'fourierf.c'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr"

%struct.__file = type { i8*, i8, i8, i16, i16, i16 (i8, %struct.__file*)*, i16 (%struct.__file*)*, i8* }

@__iob = external global [0 x %struct.__file*], align 2
@.str = private unnamed_addr constant [52 x i8] c"Error in fft():  NumSamples=%u is not power of two\0A\00", align 1
@.str.1 = private unnamed_addr constant [7 x i8] c"RealIn\00", align 1
@.str.2 = private unnamed_addr constant [8 x i8] c"RealOut\00", align 1
@.str.3 = private unnamed_addr constant [8 x i8] c"ImagOut\00", align 1
@.str.4 = private unnamed_addr constant [35 x i8] c"Error in fft_float():  %s == NULL\0A\00", align 1

; Function Attrs: nounwind
define void @fft_float(i16 %NumSamples, i16 %InverseTransform, float* %RealIn, float* %ImagIn, float* %RealOut, float* %ImagOut) #0 {
entry:
  %NumSamples.addr = alloca i16, align 2
  %InverseTransform.addr = alloca i16, align 2
  %RealIn.addr = alloca float*, align 2
  %ImagIn.addr = alloca float*, align 2
  %RealOut.addr = alloca float*, align 2
  %ImagOut.addr = alloca float*, align 2
  %NumBits = alloca i16, align 2
  %i = alloca i16, align 2
  %j = alloca i16, align 2
  %k = alloca i16, align 2
  %n = alloca i16, align 2
  %BlockSize = alloca i16, align 2
  %BlockEnd = alloca i16, align 2
  %angle_numerator = alloca double, align 8
  %tr = alloca double, align 8
  %ti = alloca double, align 8
  %delta_angle = alloca double, align 8
  %sm2 = alloca double, align 8
  %sm1 = alloca double, align 8
  %cm2 = alloca double, align 8
  %cm1 = alloca double, align 8
  %w = alloca double, align 8
  %ar = alloca [3 x double], align 8
  %ai = alloca [3 x double], align 8
  %temp = alloca double, align 8
  %denom = alloca double, align 8
  store i16 %NumSamples, i16* %NumSamples.addr, align 2
  store i16 %InverseTransform, i16* %InverseTransform.addr, align 2
  store float* %RealIn, float** %RealIn.addr, align 2
  store float* %ImagIn, float** %ImagIn.addr, align 2
  store float* %RealOut, float** %RealOut.addr, align 2
  store float* %ImagOut, float** %ImagOut.addr, align 2
  store double 0x401921FB54442D18, double* %angle_numerator, align 8
  %0 = load i16, i16* %NumSamples.addr, align 2
  %call = call i16 @IsPowerOfTwo(i16 %0)
  %tobool = icmp ne i16 %call, 0
  br i1 %tobool, label %if.end, label %if.then

if.then:                                          ; preds = %entry
  %1 = load %struct.__file*, %struct.__file** getelementptr inbounds ([0 x %struct.__file*], [0 x %struct.__file*]* @__iob, i32 0, i16 2), align 2
  %2 = load i16, i16* %NumSamples.addr, align 2
  %call1 = call i16 (%struct.__file*, i8*, ...) @fprintf(%struct.__file* %1, i8* getelementptr inbounds ([52 x i8], [52 x i8]* @.str, i32 0, i32 0), i16 %2)
  call void @exit(i16 1) #4
  unreachable

if.end:                                           ; preds = %entry
  %3 = load i16, i16* %InverseTransform.addr, align 2
[... 406 lines cut ...]
#8 0x11ff852 llvm::InstrEmitter::EmitMachineNode(llvm::SDNode*, bool, bool, llvm::DenseMap<llvm::SDValue, unsigned int, llvm::DenseMapInfo<llvm::SDValue>, llvm::detail::DenseMapPair<llvm::SDValue, unsigned int> >&) /home/dylan/projects/llvm/lib/CodeGen/SelectionDAG/InstrEmitter.cpp:782:0
#9 0x10988c5 llvm::InstrEmitter::EmitNode(llvm::SDNode*, bool, bool, llvm::DenseMap<llvm::SDValue, unsigned int, llvm::DenseMapInfo<llvm::SDValue>, llvm::detail::DenseMapPair<llvm::SDValue, unsigned int> >&) /home/dylan/projects/llvm/lib/CodeGen/SelectionDAG/InstrEmitter.h:124:0
#10 0x10af29f llvm::ScheduleDAGSDNodes::EmitSchedule(llvm::MachineBasicBlock::bundle_iterator<llvm::MachineInstr, llvm::ilist_iterator<llvm::MachineInstr> >&) /home/dylan/projects/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGSDNodes.cpp:850:0
#11 0x114a5fe llvm::SelectionDAGISel::CodeGenAndEmitDAG() /home/dylan/projects/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:819:0
#12 0x114912a llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator<llvm::Instruction const>, llvm::ilist_iterator<llvm::Instruction const>, bool&) /home/dylan/projects/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:621:0
#13 0x114c362 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) /home/dylan/projects/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:1314:0
#14 0x11482d6 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) /home/dylan/projects/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:465:0
#15 0x9f5db5 (anonymous namespace)::AVRDAGToDAGISel::runOnMachineFunction(llvm::MachineFunction&) /home/dylan/projects/llvm/lib/Target/AVR/AVRISelDAGToDAG.cpp:70:0
#16 0xbcd1bf llvm::MachineFunctionPass::runOnFunction(llvm::Function&) /home/dylan/projects/llvm/lib/CodeGen/MachineFunctionPass.cpp:41:0
#17 0xf6c30f llvm::FPPassManager::runOnFunction(llvm::Function&) /home/dylan/projects/llvm/lib/IR/LegacyPassManager.cpp:1537:0
#18 0xf6c461 llvm::FPPassManager::runOnModule(llvm::Module&) /home/dylan/projects/llvm/lib/IR/LegacyPassManager.cpp:1557:0
#19 0xf6c776 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) /home/dylan/projects/llvm/lib/IR/LegacyPassManager.cpp:1615:0
#20 0xf6cdce llvm::legacy::PassManagerImpl::run(llvm::Module&) /home/dylan/projects/llvm/lib/IR/LegacyPassManager.cpp:1722:0
#21 0xf6cfbb llvm::legacy::PassManager::run(llvm::Module&) /home/dylan/projects/llvm/lib/IR/LegacyPassManager.cpp:1756:0
#22 0x9c6fb1 compileModule(char**, llvm::LLVMContext&) /home/dylan/projects/llvm/tools/llc/llc.cpp:362:0
#23 0x9c61f3 main /home/dylan/projects/llvm/tools/llc/llc.cpp:204:0
#24 0x7f3a36585790 __libc_start_main (/usr/lib/libc.so.6+0x20790)
#25 0x9c5289 _start (/home/dylan/projects/builds/llvm/bin/llc+0x9c5289)
Stack dump:
0.      Program arguments: llc -filetype=asm -march=avr -mcpu=atmega328p fourier.ll -o fourier.s 
1.      Running pass 'Function Pass Manager' on module 'fourier.ll'.
2.      Running pass 'AVR DAG->DAG Instruction Selection' on function '@fft_float'
../Makefile.base:47: recipe for target 'fourier.s' failed
make: *** [fourier.s] Aborted (core dumped)
```

agnat, 2015-06-05:
Additional info: The (pseudo) opcode is `STDWPtrQRr` and it's an off-by-one... of course. `II.getNumOperands()` is three, which sounds correct. But `NumMIOperands` is four. The rest of the terms is zero.

So, it seems like there is a superfluous operand floating around. @dylanmckay, since you touched that area earlier... does that ring a bell?

dylanmckay, 2015-06-06:
That instruction wasn't touched with my fix of the LDD/STD instructions that I think you're referring to.

The last time this instruction was modified was when I cleaned up the whole file, placing related instructions next to each other and instructions with common fields (isLoad,..) in blocks.

It is possible that the instruction lost a property along the way. Specifically I'm thinking of the "constraints" property, where "$srx=$dst" was not copied exactly and LLVM failed to recognize two always equal operands as one.

Just speculation from memory though, I'll check when I get home.

dylanmckay, 2015-07-09:
Note that this error occurs while compiling two Arduino core files - `WMath.cpp` and `Print.cpp`.

* With `WMath.cpp`, the operand corresponds to `STDPtrQRr`
* With `Print.cpp`, the operand corresponds to `LDDWRdPtrQ`

dylanmckay, 2015-07-09:
And with [`fourier.ll](https://raw.githubusercontent.com/avr-llvm/examples/master/fourier/fourier.ll), the opcode correspond to `STDWPtrQRr`.

dylanmckay, 2015-07-09:
Fixed in af88a32108b0f11ee467cf2af5e282930bc61b3e.

The cause was that a long time ago I removed the `memri` custom operand type, thinking it was redundant and could be expressed in pure TableGen. It could - more or less.

The problem was that `reg+imm` addresses were still using a custom `ComplexPattern` (named `addr`) for instruction selection, which wasn't aware of this.

I decided the backend would be a lot cleaner if the `memri` operand kind was added back.

closed 2015-07-09 by dylanmckay