avr-llvm/llvm #104: Implement missing fixups

dylanmckay, 2015-06-07:
See `MCTargetDesc/AVRMCCodeEmitter.cpp:getExprOpValue()` for more details.

Fixups to implement:
- [ ] VK_AVR_HLO8
- [ ] VK_AVR_HHI8
- [ ] VK_AVR_PM_HI8
- [ ] VK_AVR_PM_LO8
- [ ] VK_AVR_PM_HLO8

These are for the assembler modifiers introduced in PR #103.

dylanmckay, 2015-06-07:
@agnat Are you sure `VK_AVR_HHI8` is a real fixup? There is no relocation type that the fixup is mapped to inside the ELF object file.

Now that I think about it, is this because the address of the upper 8 bits is the address anyway, so `hhi8(..)` should actual be a NOP?

agnat, 2015-06-07:
TBH, I'm not sure at all... I'm just diving in and am still trying to get my bearing. If unsure leave it dangling for now.

dylanmckay, 2015-06-08:
Looking what GCC does:

Operator | Relocation
-------------|-----------------
`lo8()`        |  `R_AVR_LO8_LDI`
`hi8()`        |  `R_AVR_HI8_LDI`
`hlo8()`      |  `R_AVR_HH8_LDI`
`hhi8()`      |  `R_AVR_MS8_LDI`

agnat, 2015-06-13:
Done as of a97fd7021288c1ee947c42904c29cb199f6cd3fa. Closing.

closed 2015-06-13 by agnat