avr-llvm/llvm #112: Lower wide load program memory pseudo instructions

dylanmckay, 2015-06-10:
Currently no logic exists in `AVRExpandPseudoInsts.cpp` to lower the `lpmw Rd, Z+` or `lpmw Rd, Z+`.

They are defined as pseudo instructions, but they are not lowered. Because of this, using them will cause an error in LLVM.

dylanmckay, 2015-07-16:
I believe we shouldn't even have wide program memory load instructions.

The instruction selector will never need it (`lpm` instructions will only be needed in hand-crafted assembly).

I will go ahead and remove the wide variants.

agnat, 2015-07-16:
> The instruction selector will never need it (lpm instructions will only be needed in hand-crafted assembly).

I'm not sure that is true. [This code](https://github.com/avr-llvm/llvm/blob/avr-support/lib/Target/AVR/AVRISelDAGToDAG.cpp#L433) looks like it is trying to select a wide flash load.

That said I'm okay with removing it since I don't believe in wide pseudo instructions anyway... :P

dylanmckay, 2015-07-16:
> I'm not sure that is true. This code looks like it is trying to select a wide flash load.

I stand corrected.

> That said I'm okay with removing it since I don't believe in wide pseudo instructions anyway... :P

I think we should hold off until we get the backend sorted out so we don't actually need them ;)