avr-llvm/llvm #114: Implement FRMIDX pseudo instruction

dylanmckay, 2015-06-10:
The instruction is supposed to be lower into an `movw` + `adiw`.

See `AVRInstrInfo.td` for more info.

agnat, 2015-06-25:
I believe this is already handled [here](https://github.com/avr-llvm/llvm/blob/avr-support/lib/Target/AVR/AVRRegisterInfo.cpp#L125).

Maybe we should close the ... pseudo issues. Not that some helpful nerd starts working on them like I did with the asm-parser tickets...

dylanmckay, 2015-06-25:
I opened these up without properly investigating them. I'll have a proper look now.

> Not that some helpful nerd starts working on them like I did with the asm-parser tickets

That would be convenient ;)

dylanmckay, 2015-06-25:
I closed this one but left the following three open.

* #115 is stopping us from compiling Arduino
* #112 and #113 will cause confusing errors when hit

closed 2015-06-25 by dylanmckay