avr-llvm/llvm #124: Investigate and mitigate register allocation hack.

agnat, 2015-06-16:
Ok, looks like `RegAllocGreedy` does not cope well with the scarce register space on the AVR. Currently there is a hack in place using two global variables `RA_ReserveREG_Y` and `RA_InSpillerCode` (like, [here](https://github.com/avr-llvm/llvm/blob/avr-support/lib/Target/AVR/AVRInstrInfo.cpp#L140-L144) and [there](https://github.com/avr-llvm/llvm/blob/avr-support/lib/CodeGen/RegAllocGreedy.cpp#L1599-L1625)).

I've also seen an old thread concluding that it is not easily mitigated. I'll post the URL as soon as I find it again.

dylanmckay, 2015-06-16:
This is a duplicate of our first and most illusive issue - #1.

agnat, 2015-06-16:
Very good. Taking it there, then.

closed 2015-06-16 by agnat