avr-llvm/llvm #128: Only emit `clr r1` on multiplication if necessary

dylanmckay, 2015-06-23:
I had a quick idea and thought it would be good to document it here.

We intend on putting `clr r1` after multiplication instructions because they clobber `r1` and it should be zero.

For multiplication instructions when we can prove that the product is less than 255, we should be able to skip `clr r1` generation because it would be redundant. This could give a [small] performance increase when calculating lots of products.

I am not sure how easy this optimisation would be to implement.

agnat, 2015-06-25:
Hehe... let's see. There is the trivial (boring) case: Both operands are known at compile-time and we never emit a multiplication. If only one operand is immediate we can safely skip the `clr r1` if it is zero or one. I'd rather optimize the bit shifting first... ;)

dylanmckay, 2015-06-25:
Yes, the bit shifting looks rather nasty.

agnat, 2015-06-25:
Yes, specially shifting by multiples of eight hurts. But it's not high priority...

Sorry, if that was to brief. My point above is: Neither multiplication by one nor by zero make it to the executable. Both will be optimized... at least I hope so! So, I think we are covered...

dylanmckay, 2015-06-25:
I was thinking of the case where

``` asm
ldi r18, 8
; r19 gets set somehow

; multiply r19 by 8 
mul r18, r19
```

When I think about it, we couldn't prove that `r19 * 8` was less than 255 without complicated analysis without much of a benefit. Closing because the optimisation isn't worth it.

closed 2015-06-25 by dylanmckay