avr-llvm/llvm #132: Machine code relative jump regression

dylanmckay, 2015-07-06:
We have developed a regression in the MC backend, causing all (non-trivial, by the strongest meaning of the term) generated code to be incorrect.

In a nutshell, all relative jumps have an offset of `0x02` added to them. This is causing us to skip the first instruction after each label.

Take this code example:

``` asm
; ModuleID = 'main_mod'

define i16 @main() {
entry:
  call void @init()
  br label %while_body

while_body:                                       ; preds = %if_cont, %entry
  %direcao.0 = phi i16 [ 1, %entry ], [ %direcao.2, %if_cont ]
  %total_passo.0 = phi i16 [ 0, %entry ], [ %total_passo.1, %if_cont ]
  %tempo_maximo = alloca i16, align 2
  store i16 1000, i16* %tempo_maximo, align 2
  %cmpi2 = icmp eq i16 %direcao.0, 0
  br i1 %cmpi2, label %if_then, label %if_else

if_then:                                          ; preds = %while_body
  call void @digitalWrite(i8 0, i8 -1)
  br label %if_cont

if_else:                                          ; preds = %while_body
  call void @digitalWrite(i8 1, i8 -1)
  br label %if_cont

if_cont:                                          ; preds = %if_else, %if_then
  %passo = alloca i16, align 2
  store i16 200, i16* %passo, align 2
  call void @delay(i32 200)
  %0 = add i16 200, %total_passo.0
  call void @digitalWrite(i8 0, i8 0)
  call void @digitalWrite(i8 1, i8 0)
  %tempo_maximo7 = load i16, i16* %tempo_maximo, align 2
  %cmpi8 = icmp sgt i16 %0, %tempo_maximo7
  %. = select i1 %cmpi2, i16 1, i16 0
  %direcao.2 = select i1 %cmpi8, i16 %., i16 %direcao.0
  %total_passo.1 = select i1 %cmpi8, i16 0, i16 %0
  br label %while_body
}

declare i16 @analogRead(i8)

declare void @digitalWrite(i8, i8)

declare void @delay(i32)

declare void @init()
```

Here is the dumped machine code vs the generated assembly in a [Gist](https://gist.github.com/dylanmckay/59ae2e28909d204932aa).

agnat, 2015-07-06:
Um, that would be me. Let me look into it.

dylanmckay, 2015-07-06:
I think [this](https://github.com/avr-llvm/llvm/blob/bccaf2a859286a36fde29b3e034221538ea5e143/lib/Target/AVR/MCTargetDesc/AVRAsmBackend.cpp#L230) line is the culprit.

It is still calculating the adjusted target, but it is throwing away the result ;)

dylanmckay, 2015-07-06:
I've fixed it in #133.

However, it looks as if we're having other problems with relative jumps. I'll investigate.

closed 2015-07-06 by dylanmckay

agnat, 2015-07-06:
Alright. Let me know if i can help...

dylanmckay, 2015-07-06:
The problem I was referring to ended up being that `adjustFixupValue()` was being called twice, causing PC-relative branches to be adjusted, and then having that value adjusted again, leading to bad offsets and two's compliment errors.

It is fixed in 482c47d9eeb23eb4c09adb34cde947c068a1f81a, which also adds support for branches with labels placed earlier in the program (which were previously encoded incorrectly).

PR #133 (2015-07-06): Fix bug wherein fixup values are not adjusted
I believe this fixes #132.

EDIT: It does.