avr-llvm/llvm #134: Add labels to MC tests

dylanmckay, 2015-07-06:
All of the MC tests were originally written before label support was properly implemented.

As such, there are only about two machine code tests which use labels.

We should add at least one case with labels to every branch instruction test. Currently a bug relating to labels with the `breq` family of instructions would have been caught if tests had existed.

dylanmckay, 2015-07-06:
One problem with this is that `llvm-mc -show-encoding` does not resolve fixups. It will instead print out statements like [this](https://github.com/avr-llvm/llvm/blob/482c47d9eeb23eb4c09adb34cde947c068a1f81a/test/MC/AVR/inst-family-cond-branch.s#L115-L116).

It would be good if there was some way to get LLVM to resolve the fixups it can (i.e. resolve fixups of labels which are in the same file that is being processed (i.e. not external)), so that we could test the actual fixup application code.

Until then, our fixup code could could silently regress without being detected by the test suite, leading to silently broken code.

agnat, 2015-07-06:
>  It will instead print out statements like this.

lol... I was looking at _exactly_ the same yesterday. Like: "Why the heck didn't we catch that."...

> Until then, our fixup code could could silently regress without being detected by the test suite, leading to silently broken code.

That's exactly the point. I bet there are tools to check the actual object code. Like, using lit/llc/FileCheck with different options. We should take a look at what other targets are doing.

We also should look into writing the `llvm-avr-forge` script. Forge as in "Hitting metal". I believe in the long run automated execution tests on the target are the only way. But that's a task for another day...

dylanmckay, 2015-07-06:
> We also should look into writing the llvm-avr-forge script. Forge as in "Hitting metal". I believe in the long run automated execution tests on the target are the only way. But that's a task for another day...

If I understand what you are saying, I agree. I visualize a set of AVR chips soldered onto a circuit board, performing automated testing would be a really good way to check the validity of the backend. There could be a host controller which connects to a PC via USB, with the ability of passing a program to each of the AVR chips. The outputs of GPIOs could be sent back to the computer, and our tests could validate that a test passed because it say, wrote the bit combination `11001011` on `PORTB`.

Of course, these are just [shitty pipe dreams](https://www.youtube.com/watch?v=djigcpyzBUQ).

agnat, 2015-07-06:
> Of course, these are just shitty pipe dreams.

lol... yeah... the best kind. ;)

agnat, 2015-07-06:
As a first step I was thinking more about the A-board everybody keeps asking about.

We could link a `test_suite.o`, built with a known good compiler (aka gcc), with our potentially wonky `test_subject.o`. Starting out really slow by just testing that the calling convention is working, we progress to a full featured test suite that 

* uses the watchdog to trigger a reset in case we never return
* provides [TAP](https://testanything.org) reporting using a virtual USB UART CDC thingy.
* measures execution time
* checks `PORTB` by just _reading_ it. No need to get all electrical...

Our script on the host keeps looking for the boot loader (crash -> fail) and reports results.

dylanmckay, 2015-07-06:
Sounds like a great idea, however - what is an "A-board"?

agnat, 2015-07-06:
The Arduino.

dylanmckay, 2015-07-06:
I am not sure why my brain didn't make that connection.

dylanmckay, 2016-12-09:
A lot of the MC tests now use labels

closed 2016-12-09 by dylanmckay