avr-llvm/llvm #135: Funky encoding of labels with negative offsets with PC-relative fixups

dylanmckay, 2015-07-06:
Take the `test/MC/AVRinst-rjmp.s` tests for example, compile it using `llvm-mc`. You will see that everything is as it should be, except for the one jump that is to a previous point in the program.

`inst-rjmp.s`:
``` asm
  rjmp  .+2
  rjmp  .-2
  rjmp  foo
  rjmp  .+8
  rjmp  end
  rjmp  .+0
end:
  rjmp .-4
  rjmp .-6
```

`avr-objdump` output:
``` asm
00000000 <foo>:
   0:	01 c0       	rjmp	.+2      	; 0x4 <foo+0x4>
   2:	ff cf       	rjmp	.-2      	; 0x2 <foo+0x2>
   4:	fd c0       	rjmp	.+506    	; 0x200 <end+0x1f4>
   6:	04 c0       	rjmp	.+8      	; 0x10 <end+0x4>
   8:	01 c0       	rjmp	.+2      	; 0xc <end>
   a:	00 c0       	rjmp	.+0      	; 0xc <end>

0000000c <end>:
   c:	fe cf       	rjmp	.-4      	; 0xa <foo+0xa>
   e:	fd cf       	rjmp	.-6      	; 0xa <foo+0xa>
```

All is well, until what should be `rjmp .-6` is actually `rjmp .+506`.

dylanmckay, 2015-07-06:
Fixed in 482c47d9eeb23eb4c09adb34cde947c068a1f81a. Now that was a debugging session that took way longer than it should've :(

closed 2015-07-06 by dylanmckay

agnat, 2015-07-06:
*Boom*

Awesome. I was just about to take a look. 

Maybe we should start to (self-) assign the issues before working on them?

dylanmckay, 2015-07-06:
> Maybe we should start to (self-) assign the issues before working on them?

I was thinking the same, it seems to be a growing problem. Let's start using the GitHub assignee feature?

agnat, 2015-07-06:
>  Let's start using the GitHub assignee feature?

Ack.