avr-llvm/llvm #149: Support 32-bit and bigger division

dylanmckay, 2015-07-21:
We support 8 and 16-bit division through the use of custom lowering.

We cannot custom lower an illegal type (greater than 16-bits).

LLVM will use its existing libcall infrastructure, leading to the rtlib function being called incorrectly, producing incorrect results.

dylanmckay, 2015-08-05:
This blocks Arduino - the following files use 32-bit division:
* `WMath.cpp` (for the [`map()`](https://www.arduino.cc/en/Reference/Map) function)
* `Tone.cpp` (for the tone generator frequency calculations)
* `Print.cpp` (for integer printing routines)
* `HardwareSerial.cpp` (for baud-rate calculations in [`Serial.begin()`](https://www.arduino.cc/en/Serial/Begin))

If these features are not used, Arduino is likely to work.

dylanmckay, 2015-08-19:
Fixed in 9724f1d16577d3dbbc4d4415c041e50401ceec52.

@agnat Could you write an on-target test for 32-bit and 64-bit div/rem? I'm sure the generated is correct, but it never hurts :)

closed 2015-08-19 by dylanmckay

4ntoine, 2015-08-19:
I'm having this build error:
```
[ 37%] Building CXX object lib/Target/AVR/CMakeFiles/LLVMAVRCodeGen.dir/AVRISelLowering.cpp.o
/Users/asmirnov/Documents/dev/src/avr-llvm/llvm/lib/Target/AVR/AVRISelLowering.cpp:691:3: error: 
      multiple default labels in one switch
  default:
  ^
/Users/asmirnov/Documents/dev/src/avr-llvm/llvm/lib/Target/AVR/AVRISelLowering.cpp:707:3: note: 
      previous case defined here
  default: {
  ^
1 error generated.
```

agnat, 2015-08-19:
>  Don't worry, I'm working on it.

Hehe... That's the spirit! :-D

dylanmckay, 2015-08-20:
@4ntoine: Whoops, fixed in 9546f716db059112e461794ed0384a6d755ee35a.

dylanmckay, 2015-10-27:
Reproduction code:

``` llvm
define i64 @do_thing(i64 %a, i64 %b) {
    %1 = udiv i64 %a, %b
    ret i64 %1
}
```