avr-llvm/llvm #151: Compile crash: "Don't know how to custom expand this!"

4ntoine, 2015-08-16:
```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp -o /tmp/arduino_test1/HardwareSerial.cpp.o
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:32:
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial_private.h:119:5: warning: 
      expression result unused; assign into a variable to force a volatile load
      [-Wunused-volatile-lvalue]
    *_udr;
    ^~~~~
Don't know how to custom expand this!
UNREACHABLE executed at /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/lib/Target/AVR/AVRISelLowering.cpp:691!
0  clang-3.8                0x00000001081b93ee llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 46
1  clang-3.8                0x00000001081bacb9 PrintStackTraceSignalHandler(void*) + 25
2  clang-3.8                0x00000001081b79a9 llvm::sys::RunSignalHandlers() + 425
3  clang-3.8                0x00000001081bb03e SignalHandler(int) + 366
4  libsystem_platform.dylib 0x00007fff927d5f1a _sigtramp + 26
5  libsystem_platform.dylib 0x00007fff69f41764 _sigtramp + 3614881892
6  clang-3.8                0x00000001081bacdb raise + 27
7  clang-3.8                0x00000001081bad92 abort + 18
8  clang-3.8                0x00000001080fd326 llvm::llvm_unreachable_internal(char const*, char const*, unsigned int) + 198
9  clang-3.8                0x0000000107159370 llvm::AVRTargetLowering::ReplaceNodeResults(llvm::SDNode*, llvm::SmallVectorImpl<llvm::SDValue>&, llvm::SelectionDAG&) const + 96
10 clang-3.8                0x0000000108cf5b27 llvm::DAGTypeLegalizer::CustomLowerNode(llvm::SDNode*, llvm::EVT, bool) + 279
11 clang-3.8                0x0000000108cd5b49 llvm::DAGTypeLegalizer::ExpandIntegerResult(llvm::SDNode*, unsigned int) + 281
12 clang-3.8                0x0000000108cf2023 llvm::DAGTypeLegalizer::run() + 947
13 clang-3.8                0x0000000108cf83e2 llvm::SelectionDAG::LegalizeTypes() + 66
14 clang-3.8                0x0000000108e56761 llvm::SelectionDAGISel::CodeGenAndEmitDAG() + 4193
15 clang-3.8                0x0000000108e556ef llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator<llvm::Instruction const>, llvm::ilist_iterator<llvm::Instruction const>, bool&) + 255
16 clang-3.8                0x0000000108e553ca llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) + 5194
17 clang-3.8                0x0000000108e52ba6 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 1142
18 clang-3.8                0x0000000107147a1b llvm::AVRDAGToDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 59
19 clang-3.8                0x00000001075b133e llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 110
20 clang-3.8                0x0000000107a3bd8d llvm::FPPassManager::runOnFunction(llvm::Function&) + 413
21 clang-3.8                0x0000000107a3c0a5 llvm::FPPassManager::runOnModule(llvm::Module&) + 117
22 clang-3.8                0x0000000107a3cac0 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) + 1392
23 clang-3.8                0x0000000107a3c366 llvm::legacy::PassManagerImpl::run(llvm::Module&) + 310
24 clang-3.8                0x0000000107a3d241 llvm::legacy::PassManager::run(llvm::Module&) + 33
25 clang-3.8                0x0000000108411a8a (anonymous namespace)::EmitAssemblyHelper::EmitAssembly(clang::BackendAction, llvm::raw_pwrite_stream*) + 1866
26 clang-3.8                0x0000000108411172 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::raw_pwrite_stream*) + 114
27 clang-3.8                0x00000001086848ef clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) + 1375
28 clang-3.8                0x00000001099111cf clang::ParseAST(clang::Sema&, bool, bool) + 1263
29 clang-3.8                0x0000000108a72a9a clang::ASTFrontendAction::ExecuteAction() + 522
30 clang-3.8                0x0000000108682d23 clang::CodeGenAction::ExecuteAction() + 3987
31 clang-3.8                0x0000000108a72000 clang::FrontendAction::Execute() + 112
32 clang-3.8                0x00000001089ecf1f clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) + 1023
33 clang-3.8                0x0000000108aec9a0 clang::ExecuteCompilerInvocation(clang::CompilerInstance*) + 3200
34 clang-3.8                0x000000010711bdb8 cc1_main(llvm::ArrayRef<char const*>, char const*, void*) + 4968
35 clang-3.8                0x000000010710e693 ExecuteCC1Tool(llvm::ArrayRef<char const*>, llvm::StringRef) + 163
36 clang-3.8                0x000000010710d325 main + 1573
37 libdyld.dylib            0x00007fff95ec05c9 start + 1
Stack dump:
0.	Program arguments: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.8 -cc1 -triple avr -emit-obj -disable-free -main-file-name HardwareSerial.cpp -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -target-cpu atmega328p -target-linker-version 242 -g -dwarf-column-info -ffunction-sections -fdata-sections -coverage-file /tmp/arduino_test1/HardwareSerial.cpp.o -resource-dir /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/../lib/clang/3.8.0 -dependency-file /tmp/arduino_test1/HardwareSerial.cpp.d -MT /tmp/arduino_test1/HardwareSerial.cpp.o -D F_CPU=16000000L -D USB_VID=null -D USB_PID=null -D ARDUINO=105 -I /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include -Os -Wall -fdeprecated-macro -fdebug-compilation-dir /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin -ferror-limit 19 -fmessage-length 100 -mstackrealign -fobjc-runtime=gcc -fcxx-exceptions -fexceptions -fdiagnostics-show-option -fcolor-diagnostics -vectorize-loops -vectorize-slp -o /tmp/arduino_test1/HardwareSerial.cpp.o -x c++ /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp 
1.	<eof> parser at end of file
2.	Code generation
3.	Running pass 'Function Pass Manager' on module '/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp'.
4.	Running pass 'AVR DAG->DAG Instruction Selection' on function '@_ZN14HardwareSerial5beginEmh'
clang-3.8: error: unable to execute command: Illegal instruction: 4
clang-3.8: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.8.0 (https://github.com/avr-llvm/clang.git bc1779778787f8cfca43849f8a4412b5ca7432c1) (llvm/llvm.git 693f1d90faf67af7ca0ec14e3eb8643ac4a1972f)
Target: avr
Thread model: posix
clang-3.8: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.8: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.8: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/HardwareSerial-33df4b.cpp
clang-3.8: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/HardwareSerial-33df4b.sh
clang-3.8: note: diagnostic msg: 

********************
```

dylanmckay, 2015-08-16:
Without debugging, I'm pretty certain that this issue is caused by #149. There are four Arduino files which use 32-bit division, and `HardwareSerial0.cpp` is one of them.

I recently added a special error message for this case - see [here](https://github.com/avr-llvm/llvm/blob/52902258b763e3c04cc626388200de82d19bd4ee/lib/Target/AVR/AVRISelLowering.cpp#L709).

What revision of the compiler are you using?

Also, could you please run `clang -S -emit-llvm -o HardwareSerial.ll` and attach the IR as well.

4ntoine, 2015-08-16:
MBA-Anton:bin asmirnov$ ./clang -version
clang-3.8: error: unknown argument: '-version'
clang-3.8: error: no input files
MBA-Anton:bin asmirnov$ ./clang --version
clang version 3.8.0 (https://github.com/avr-llvm/clang.git bc1779778787f8cfca43849f8a4412b5ca7432c1) (llvm/llvm.git 693f1d90faf67af7ca0ec14e3eb8643ac4a1972f)
Target: x86_64-apple-darwin14.3.0

4ntoine, 2015-08-16:
```
./clang -S -emit-llvm -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp -o /tmp/arduino_test1/HardwareSerial.ll
```

HardwareSerial.ll:
```
; ModuleID = '/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr"

%class.HardwareSerial = type { %class.Stream, i8*, i8*, i8*, i8*, i8*, i8*, i8, i8, i8, i8, i8, [64 x i8], [64 x i8], i8 }
%class.Stream = type { %class.Print, i32, i32 }
%class.Print = type { i32 (...)**, i16 }

$_ZTS6Stream = comdat any

$_ZTI6Stream = comdat any

@_ZTV14HardwareSerial = unnamed_addr constant [8 x i8*] [i8* null, i8* bitcast ({ i8*, i8*, i8* }* @_ZTI14HardwareSerial to i8*), i8* bitcast (i16 (%class.HardwareSerial*, i8)* @_ZN14HardwareSerial5writeEh to i8*), i8* bitcast (i16 (%class.Print*, i8*, i16)* @_ZN5Print5writeEPKhj to i8*), i8* bitcast (i16 (%class.HardwareSerial*)* @_ZN14HardwareSerial9availableEv to i8*), i8* bitcast (i16 (%class.HardwareSerial*)* @_ZN14HardwareSerial4readEv to i8*), i8* bitcast (i16 (%class.HardwareSerial*)* @_ZN14HardwareSerial4peekEv to i8*), i8* bitcast (void (%class.HardwareSerial*)* @_ZN14HardwareSerial5flushEv to i8*)], align 2
@_ZTVN10__cxxabiv120__si_class_type_infoE = external global i8*
@_ZTS14HardwareSerial = constant [17 x i8] c"14HardwareSerial\00"
@_ZTS6Stream = linkonce_odr constant [8 x i8] c"6Stream\00", comdat
@_ZTI5Print = external constant i8*
@_ZTI6Stream = linkonce_odr constant { i8*, i8*, i8* } { i8* bitcast (i8** getelementptr inbounds (i8*, i8** @_ZTVN10__cxxabiv120__si_class_type_infoE, i16 2) to i8*), i8* getelementptr inbounds ([8 x i8], [8 x i8]* @_ZTS6Stream, i32 0, i32 0), i8* bitcast (i8** @_ZTI5Print to i8*) }, comdat
@_ZTI14HardwareSerial = constant { i8*, i8*, i8* } { i8* bitcast (i8** getelementptr inbounds (i8*, i8** @_ZTVN10__cxxabiv120__si_class_type_infoE, i16 2) to i8*), i8* getelementptr inbounds ([17 x i8], [17 x i8]* @_ZTS14HardwareSerial, i32 0, i32 0), i8* bitcast ({ i8*, i8*, i8* }* @_ZTI6Stream to i8*) }

define weak void @_Z14serialEventRunv() #0 {
entry:
  br i1 icmp ne (i1 ()* @_Z17Serial0_availablev, i1 ()* null), label %land.lhs.true, label %if.end

land.lhs.true:                                    ; preds = %entry
  br i1 icmp ne (void ()* @_Z11serialEventv, void ()* null), label %land.lhs.true.1, label %if.end

land.lhs.true.1:                                  ; preds = %land.lhs.true
  %call = call zeroext i1 @_Z17Serial0_availablev()
  br i1 %call, label %if.then, label %if.end

if.then:                                          ; preds = %land.lhs.true.1
  call void @_Z11serialEventv()
  br label %if.end

if.end:                                           ; preds = %if.then, %land.lhs.true.1, %land.lhs.true, %entry
  ret void
}

declare extern_weak zeroext i1 @_Z17Serial0_availablev() #0

declare extern_weak void @_Z11serialEventv() #0

; Function Attrs: nounwind
define void @_ZN14HardwareSerial17_tx_udr_empty_irqEv(%class.HardwareSerial* %this) #1 align 2 {
entry:
  %this.addr = alloca %class.HardwareSerial*, align 2
  %c = alloca i8, align 1
  store %class.HardwareSerial* %this, %class.HardwareSerial** %this.addr, align 2
  %this1 = load %class.HardwareSerial*, %class.HardwareSerial** %this.addr
  %_tx_buffer_tail = getelementptr inbounds %class.HardwareSerial, %class.HardwareSerial* %this1, i32 0, i32 11
  %0 = load volatile i8, i8* %_tx_buffer_tail, align 1
  %idxprom = zext i8 %0 to i16
  %_tx_buffer = getelementptr inbounds %class.HardwareSerial, %class.HardwareSerial* %this1, i32 0, i32 13
  %arrayidx = getelementptr inbounds [64 x i8], [64 x i8]* %_tx_buffer, i32 0, i16 %idxprom
  %1 = load i8, i8* %arrayidx, align 1
  store i8 %1, i8* %c, align 1
  %_tx_buffer_tail2 = getelementptr inbounds %class.HardwareSerial, %class.HardwareSerial* %this1, i32 0, i32 11
  %2 = load volatile i8, i8* %_tx_buffer_tail2, align 1
  %conv = zext i8 %2 to i16
  %add = add nsw i16 %conv, 1
  %rem = srem i16 %add, 64
  %conv3 = trunc i16 %rem to i8
  %_tx_buffer_tail4 = getelementptr inbounds %class.HardwareSerial, %class.HardwareSerial* %this1, i32 0, i32 11
[... 550 lines cut ...]
  %25 = inttoptr i16 %24 to i8*
  %26 = load volatile i8, i8* %25, align 1
  %conv27 = zext i8 %26 to i16
  %or28 = or i16 %conv27, 32
  %conv29 = trunc i16 %or28 to i8
  store volatile i8 %conv29, i8* %25, align 1
  %_written = getelementptr inbounds %class.HardwareSerial, %class.HardwareSerial* %this1, i32 0, i32 7
  store i8 1, i8* %_written, align 1
  store i16 1, i16* %retval
  br label %return

return:                                           ; preds = %while.end, %if.then
  %27 = load i16, i16* %retval
  ret i16 %27
}

declare i16 @_ZN5Print5writeEPKhj(%class.Print*, i8*, i16) #0

attributes #0 = { "disable-tail-calls"="false" "less-precise-fpmad"="false" "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "stack-protector-buffer-size"="8" "target-cpu"= [... cut ...]
attributes #1 = { nounwind "disable-tail-calls"="false" "less-precise-fpmad"="false" "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "stack-protector-buffer-size"="8" "target-cpu"="atmega328p" "unsafe-fp-math"="false" "use-soft-float"="false" }

!llvm.ident = !{!0}

!0 = !{!"clang version 3.8.0 (https://github.com/avr-llvm/clang.git bc1779778787f8cfca43849f8a4412b5ca7432c1) (llvm/llvm.git 693f1d90faf67af7ca0ec14e3eb8643ac4a1972f)"}
```

dylanmckay, 2015-08-16:
LLVM commit 693f1d90faf67af7ca0ec14e3eb8643ac4a1972f is two weeks old now :)

If you pull the latest changes and run `clang` again, it will most likely print
```
you have reached a bug - division/modulus greater than 16-bits is broken - see GitHub issue #149
```

If it doesn't, then this is a different problem.

I have a fix in the works which will allow us the get rid of almost all of our pseudo instructions and also fix issue #149. I am intending on getting it upstreamed to LLVM trunk however, so it could take a little while.

4ntoine, 2015-08-16:
ok, great.

dylanmckay, 2015-08-16:
I just checked by running the IR through `llc` - this is a duplicate of #149.

Closing this issue, discussion can be moved to #149.

closed 2015-08-16 by dylanmckay

agnat, 2015-08-16:
> I have a fix in the works which will allow us the get rid of almost all of our pseudo instructions [...]

Awesome. Why isn't there a PR for this? I wanna watch... :-D

4ntoine, 2015-08-16:
i've just checked out the latest revision and i'm still having crash:
```
MBA-Anton:bin asmirnov$ ./clang --version
clang version 3.8.0 (https://github.com/avr-llvm/clang.git bc1779778787f8cfca43849f8a4412b5ca7432c1) (llvm/llvm.git c4c1b5f054a5fb93668b4f1780a8b6b72fc27147)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
```

dylanmckay, 2015-08-16:
> Awesome. Why isn't there a PR for this? I wanna watch... :-D

It's not very progressed, but I've recently got the idea fleshed out :)

A new `LegalizeAction` will be added, named `Split`. This will be the same as `LegalizeAction::Expand` except that it will always expand by splitting (which will remove all assumptions about pointer size and legal types).

I messaged `llvm-dev`. Several backends seem to have hit the same problem we have. Notably, this fix would remove some code duplication in the AMDGPU backend. The post did not get many replies (sadly), so I am not sure if this change would be welcomed into trunk.

The infrastructure is already there - I just need to refactor it.

agnat, 2015-08-16:
Sounds good.

> so I am not sure if this change would be welcomed into trunk.

Yeah, it seems a bit ... difficult to get things landed. Even simple fixes are hard to get in until you have a certain track record. But that's just how it is.

One thing they recommend is to start reviewing patches of others. But I don't feel qualified to comment on any _functional_ issue. All I can do is look for general coding/design issues, which feels a bit thin.

No matter what it looks like now, consider putting it in a PR anyway. It is _not_ embarrassing. It's actually quite fun to have a PR to chat over the code. [Cathedrals vs. Bazaars](https://en.wikipedia.org/wiki/The_Cathedral_and_the_Bazaar) &c... Just think about it ;)