avr-llvm/llvm #155: ld reports that the architecture of object files is incompatible with specified mcu

seanmkauffman, 2015-08-18:
When compile to an object file using `llvm-mc` with `-triple=avr-none -mcpu=atmega2560` and then try to link with ld, I get an error that the architecture is incompatible.

```
$ ~/workspace/avr-work/bin/llvm-mc -triple=avr-none -mcpu=atmega2560 -filetype=obj fourierf.s -o fourierf.o
$ ~/workspace/avr-work/bin/llvm-mc -triple=avr-none -mcpu=atmega2560 -filetype=obj fftmisc.s -o fftmisc.o
$ avr-gcc -mmcu=atmega2560 -I. fftmisc.o fourierf.o test.o -o test.elf -Wl,-Map=main.map,--cref
/usr/libexec/gcc/avr/ld: avr architecture of input file `fftmisc.o' is incompatible with avr:6 output
/usr/libexec/gcc/avr/ld: avr architecture of input file `fourierf.o' is incompatible with avr:6 output
...
```

It looks like the object files are specifying the architecture without the version that ld is expecting.  This appears to occur regardless of the contents of the source file, so long as I'm compiling with these flags.

I tried removing the `-triple` argument, but then it won't assemble properly.  It's possible I'm not specifying the flags properly, but I couldn't figure out how to resolve this from digging around in the code.

dylanmckay, 2015-08-18:
It's probably a bug in the subtarget feature descriptions. I'll investigate.

dylanmckay, 2015-08-19:
The atmega2560 is apart of the `avr6` architecture, and we make sure that we only emit instructions that this family supports.

What we aren't doing, however, is setting the `architecture` field inside the ELF header appropriately - we always set it to `avr`. The reason why AVR-GCC is failing to link the program is because it expects the architecture to be set to `avr6` in the ELF header.

PR #157 (2015-08-23): Place architecture value inside ELF header
This fixes #155.

@agnat Can you look over this? It touches a lot of code.
closed 2015-08-23 by dylanmckay