avr-llvm/llvm #156: Missing floating point symbols from libgcc

seanmkauffman, 2015-08-18:
When attempting to link object files generated with AVR-LLVM I am encountering the following missing symbols which are expected in AVR-GCC gcclib:

```
__floatunsidf
__truncdfsf2
__extendsfdf2
__subdf3
__adddf3
__divdf3
__muldf3
```

These all appear to be related to floating point operations, although I am not completely sure.  [This gist demonstrates all these missing symbols](https://gist.github.com/seanmkauffman/c15001d346dfcb73150c#file-fourierf-ll) and was generated using the command

```
$ ~/workspace/avr-work/bin/clang -O0 --target=avr-none-none -emit-llvm -I/usr/avr/include/ -S fourierf.c -o fourierf.ll
```

dylanmckay, 2015-08-18:
Are you using AVR-GCC as the linker?

seanmkauffman, 2015-08-18:
Yes, I am using AVR-GCC.  Here's an example of the command line I'm using and what I'm seeing:

```
$ avr-gcc -mmcu=atmega2560 -I. fftmisc.o fourierf.o test.o -o test.elf -Wl,-Map=main.map,--cref
/usr/libexec/gcc/avr/ld: avr architecture of input file `fftmisc.o' is incompatible with avr:6 output
/usr/libexec/gcc/avr/ld: avr architecture of input file `fourierf.o' is incompatible with avr:6 output
fftmisc.o: In function `LBB3_2':
fftmisc.ll:(.text+0x1de): undefined reference to `__floatunsidf'
fftmisc.ll:(.text+0x1f0): undefined reference to `__floatunsidf'
fftmisc.o: In function `LBB3_4':
fftmisc.ll:(.text+0x220): undefined reference to `__floatunsidf'
fftmisc.ll:(.text+0x240): undefined reference to `__subdf3'
fftmisc.ll:(.text+0x250): undefined reference to `__floatunsidf'
fftmisc.o: In function `LBB3_5':
fftmisc.ll:(.text+0x26c): undefined reference to `__divdf3'
fourierf.o: In function `fft_float':
fourierf.ll:(.text+0x150): undefined reference to `__subdf3'
fourierf.ll:(.text+0x20e): undefined reference to `__truncdfsf2'
fourierf.ll:(.text+0x366): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0x40e): undefined reference to `__floatunsidf'
fourierf.ll:(.text+0x464): undefined reference to `__divdf3'
fourierf.ll:(.text+0x4a6): undefined reference to `__muldf3'
fourierf.ll:(.text+0x51a): undefined reference to `__subdf3'
fourierf.ll:(.text+0x58a): undefined reference to `__muldf3'
fourierf.ll:(.text+0x5fa): undefined reference to `__subdf3'
fourierf.ll:(.text+0x63a): undefined reference to `__adddf3'
fourierf.ll:(.text+0x87e): undefined reference to `__muldf3'
fourierf.ll:(.text+0x8b2): undefined reference to `__subdf3'
fourierf.ll:(.text+0x976): undefined reference to `__muldf3'
fourierf.ll:(.text+0x98a): undefined reference to `__subdf3'
fourierf.ll:(.text+0xa36): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xa5a): undefined reference to `__muldf3'
fourierf.ll:(.text+0xa86): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xaaa): undefined reference to `__muldf3'
fourierf.ll:(.text+0xaca): undefined reference to `__subdf3'
fourierf.ll:(.text+0xb42): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xb66): undefined reference to `__muldf3'
fourierf.ll:(.text+0xb92): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xbb6): undefined reference to `__muldf3'
fourierf.ll:(.text+0xbd6): undefined reference to `__adddf3'
fourierf.ll:(.text+0xc42): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xc86): undefined reference to `__subdf3'
fourierf.ll:(.text+0xc8a): undefined reference to `__truncdfsf2'
fourierf.ll:(.text+0xce6): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xd1e): undefined reference to `__subdf3'
fourierf.ll:(.text+0xd22): undefined reference to `__truncdfsf2'
fourierf.ll:(.text+0xd84): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xdc8): undefined reference to `__adddf3'
fourierf.ll:(.text+0xdcc): undefined reference to `__truncdfsf2'
fourierf.ll:(.text+0xe16): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xe4e): undefined reference to `__adddf3'
fourierf.ll:(.text+0xe52): undefined reference to `__truncdfsf2'
fourierf.ll:(.text+0xefa): undefined reference to `__floatunsidf'
fourierf.ll:(.text+0xf54): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xf68): undefined reference to `__divdf3'
fourierf.ll:(.text+0xf6c): undefined reference to `__truncdfsf2'
fourierf.ll:(.text+0xfb6): undefined reference to `__extendsfdf2'
fourierf.ll:(.text+0xfca): undefined reference to `__divdf3'
fourierf.ll:(.text+0xfce): undefined reference to `__truncdfsf2'
collect2: error: ld returned 1 exit status
```

dylanmckay, 2015-08-19:
This problem is caused by the fact that AVR-GCC deviates from the C standard -  on AVR, `double` values are internally converted into `float` values, so that `sizeof(double) == sizeof(float) == 4`. Note that all of the functions you listed each have the `d` suffix (as opposed to `f` for floats).

This is because AVR-LLVM's `clang` does not take this into account. I am not sure if it will be easy to add support for this (I haven't worked on clang much). If we made clang demote `double`s to `float`s on AVR, this should solve the problem at hand.

One other point, however, is that would then not be able to compile arbitrary IR using AVR-LLVM - as soon as an IR file mentions `double`, it would lead to linking errors. Thus, we should modify LLVM to ensure that doubles are 32-bits on AVR. We should, however then allow the C-standard behavior to be enabled via the command line, for example `--avr-double-64`.

@agnat What are your thoughts?

dylanmckay, 2015-08-20:
A dirty workaround would be to find+replace all instances of `double` with `float` in the generated IR.

dylanmckay, 2015-08-20:
This is fixed in clang commit [`26488ca5`](https://github.com/avr-llvm/clang/commit/26488ca543c8d1bed22a8e3aefcb6a2f218b9edc).

After I fixed that, I discovered there was another bug - LLVM thought the 32-bit builtin sin function was named `sinf`, but as there are no doubles, it was actually called `sin`. Fixed in 9fe85bfeed3c5b123e7bb83b0f1cb0423e85f086.

You code should now compile/run successfully! :)

closed 2015-08-20 by dylanmckay

seanmkauffman, 2015-08-20:
Thanks!  I can confirm that I'm not seeing these errors any more.

dylanmckay, 2015-08-20:
Thanks for discovering this!