avr-llvm/llvm #158: Attempt to use deleted function

4ntoine, 2015-08-19:
```
[ 68%] Building CXX object tools/clang/lib/AST/CMakeFiles/clangAST.dir/ASTDiagnostic.cpp.o
/Users/asmirnov/Documents/dev/src/avr-llvm/llvm/tools/clang/lib/AST/ASTDiagnostic.cpp:448:6: error: 
      attempt to use a deleted function
  OS.flush();
     ^
/Users/asmirnov/Documents/dev/src/avr-llvm/llvm/include/llvm/Support/raw_ostream.h:512:8: note: 
      'flush' has been explicitly marked deleted here
  void flush() = delete;
       ^
1 error generated.
```

4ntoine, 2015-08-19:
```
MBA-Anton:clang asmirnov$ pwd
/Users/asmirnov/Documents/dev/src/avr-llvm/llvm/tools/clang
MBA-Anton:clang asmirnov$ git log
commit bc1779778787f8cfca43849f8a4412b5ca7432c1
Merge: 31dba4e d50fd8a
Author: Dylan McKay <dylanmckay34@gmail.com>
Date:   Fri Jul 31 10:53:58 2015 +1200

    Merge branch 'master' into avr-support
```

4ntoine, 2015-08-19:
```
MBA-Anton:llvm asmirnov$ pwd
/Users/asmirnov/Documents/dev/src/avr-llvm/llvm
MBA-Anton:llvm asmirnov$ git log
commit fa6adf709c783371cc3cf96749ceec6f19570196
Author: Dylan McKay <dylanmckay34@gmail.com>
Date:   Thu Aug 20 05:29:53 2015 +1200

    [AVR] Add on-target execution tests for 32-bit DIV/REM
```

dylanmckay, 2015-08-20:
I'm 98% certain this is caused by me updating LLVM but not clang. I'm in the process of updating them both again currently.

dylanmckay, 2016-01-19:
Fixed.

closed 2016-01-19 by dylanmckay