avr-llvm/llvm #162: Function pointer miscalculation.

majkrzak, 2015-09-03:
It seems that avr-llvm have problem with calculating proper addresses of functions during casting to function pointer. Following code demonstrates it:

```c
void foo(){}
int jmp __attribute__((section(".vectors"))) = 0x940C;
void (*ptr)()  __attribute__((section(".vectors"))) = (&foo);

// avr-gcc -mmcu=atmega328p -nostdlib ptrs.c &&  avr-objdump -SDh   
/*  
 * a.out:     file format elf32-avr
 * 
 * Sections:
 * Idx Name          Size      VMA       LMA       File off  Algn
 *   0 .text         00000014  00000000  00000000  00000074  2**1
 *                   CONTENTS, ALLOC, LOAD, READONLY, CODE
 *   1 .data         00000000  00800060  00000014  00000088  2**0
 *                   CONTENTS, ALLOC, LOAD, DATA
 *   2 .comment      00000011  00000000  00000000  00000088  2**0
 *                   CONTENTS, READONLY
 * 
 * Disassembly of section .text:
 * 
 * 00000000 <jmp>:
 *    0:   0c 94 02 00     jmp     0x4     ; 0x4 <foo>
 * 
 * 00000002 <ptr>:
 *    2:   02 00           .word   0x0002  ; ????
 * 
 * 00000004 <foo>:
 *    4:   cf 93           push    r28
 *    6:   df 93           push    r29
 *    8:   cd b7           in      r28, 0x3d       ; 61
 *    a:   de b7           in      r29, 0x3e       ; 62
 *    c:   00 00           nop
 *    e:   df 91           pop     r29
 *   10:   cf 91           pop     r28
 *   12:   08 95           ret
 * 
 * Disassembly of section .comment:
 * 
 * 00000000 <.comment>:
 *    0:   47 43           sbci    r20, 0x37       ; 55
 *    2:   43 3a           cpi     r20, 0xA3       ; 163
 *    4:   20 28           or      r2, r0
 *    6:   47 4e           sbci    r20, 0xE7       ; 231
 *    8:   55 29           or      r21, r5
 *    a:   20 35           cpi     r18, 0x50       ; 80
 *    c:   2e 32           cpi     r18, 0x2E       ; 46
 *    e:   2e 30           cpi     r18, 0x0E       ; 14
 *         ...
 */

// clang --target=avr -mmcu=atmega328p -nostdlib ptrs.c &&  avr-objdump -SDh
/*  
 * a.out:     file format elf32-avr
 * 
 * Sections:
 * Idx Name          Size      VMA       LMA       File off  Algn
 *  0 .text         00000006  00000000  00000000  00000054  2**2
 *                  CONTENTS, ALLOC, LOAD, READONLY, CODE
 *  1 .data         00000000  00800060  00800060  0000005a  2**0
 *                  CONTENTS, ALLOC, LOAD, DATA
 * 
 * Disassembly of section .text:
 * 
 * 00000000 <jmp>:
 *   0:   0c 94 04 00     jmp     0x8     ; 0x8 <_etext+0x2>
 * 
 * 00000002 <ptr>:
 *   2:   04 00           .word   0x0004  ; ????
 * 
 * 00000004 <foo>:
 *   4:   08 95           ret
 * 
 */
```

dylanmckay, 2015-09-06:
This is a bit strange. It is interesting that our address of `0x8` is twice as much as GCC's `0x4`. One artifact of AVR (which we currently do) is that addresses passed to instructions are supposed left-shifted one place, doubling them. This is probably related.

dylanmckay, 2015-09-24:
IR:

``` llvm
; ModuleID = '/home/dylan/Desktop/fn_ptr.c'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr"

@ptr = global void (...)* bitcast (void ()* @foo to void (...)*), section ".vectors", align 2

; Function Attrs: nounwind
define void @foo() #0 {
entry:
  ret void
}
```

majkrzak, 2015-10-03:
I'm not quite sure but i think that this is not a problem with interrupts. It is rather caused by that  Data memory is addressed each 8 bits and Program memory 16 bits.

dylanmckay, 2015-12-30:
It is definitely not caused by interrupts. I am not sure what you mean by saying "It is rather caused by that Data memory is addressed each 8 bits and Program memory 16 bits" is the cause.

majkrzak, 2015-12-30:
> ### Atmega 328p datasheet: 
> #### 8.2 In-System Reprogrammable Flash Program Memory
> Since all AVR instructions are 16 or 32 bits wide, the Flash is organized as 2/4/8/16K x 16.

Function pointers should address each 16 bits of FLASH not like regular pointers that points to 8 bit SRAM cells.

dylanmckay, 2016-01-07:
For the given program:

``` llvm
; ModuleID = '/Users/dylanmckay/Desktop/avr.c'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr"

@bar = global i16 2, align 2
@abc = global i16 2, align 2
@jmp = global i16 -27636, section ".vectors", align 2
@ptr = global void (...)* bitcast (void ()* @foo to void (...)*), section ".vectors", align 2
@qwerty = global i16 2, align 2

; Function Attrs: nounwind
define void @milan() #0 {
entry:
  ret void
}

; Function Attrs: nounwind
define void @foo() #0 {
entry:
  ret void
}

; Function Attrs: nounwind
define void @inter() #0 {
entry:
  ret void
}

attributes #0 = { nounwind "disable-tail-calls"="false" "less-precise-fpmad"="false" "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "stack-protector-buffer-size"="8" "target-cpu"="avr2" "unsafe-fp-math"="false" "use-soft-float"="false" }

!llvm.ident = !{!0}

!0 = !{!"clang version 3.8.0 (https://github.com/avr-llvm/clang.git 55d2db19d360a92b3b87444a2ea5e07d1624a2bc) (https://github.com/llvm-mirror/llvm.git ed02a4bc0484efbf1362cf04520976ea52d4351f)"}
```

LLVM and GCC generate similar code, although the relocations differ.

GCC:
```
RELOCATION RECORDS FOR [.vectors]:
OFFSET   TYPE              VALUE 
00000004 R_AVR_16          .text+0x00000004
```

LLVM:
```
RELOCATION RECORDS FOR [.vectors]:
OFFSET   TYPE              VALUE 
00000002 R_AVR_16          foo
```

In the object file for GCC, `.text+0x000000004` points to `foo`, and it is correctly relocated.

For LLVM, we directly try and relocate to the symbol `foo` but it gives us a strange value.

dylanmckay, 2016-01-18:
I emailed the `avr-gcc` mailing list, and Senthil suggested that the problem laid in the fact we were generating `R_AVR_16` relocations, but our addresses are specifically for program memory.

We should be creating `R_AVR_16_PM` relocs.

dylanmckay, 2016-05-31:
Looks like the `_PM` relocs _only_ differ by the fact that they are rightshifted by 1.

dylanmckay, 2016-05-31:
Looks like the problem lies in `AVRELFObjectWriter::getRelocType`.

When I initially implemented the backend, I would've naively written

``` cpp
  case FK_Data_1:
    return ELF::R_AVR_8;
  case FK_Data_2:
    return ELF::R_AVR_16;
  case FK_Data_4:
    return ELF::R_AVR_32;
  case AVR::fixup_32:
    return ELF::R_AVR_32;
```

This only get called when creating relocations for symbols in the object file. There will always be a symbol at this point, therefore the address will always be in program memory. This means that the correct relocation for `FK_Data_2` is `R_AVR_16_PM`.

I will turn the other cases into assertion errors, and make sure the test suite passes. This will stop this from happening in the future.

dylanmckay, 2016-05-31:
Fixed as of ef21f19bf0ceb4151866de8e20b66ba0432e4437 😎

closed 2016-05-31 by dylanmckay