avr-llvm/llvm #165: Archive index issue

4ntoine, 2015-09-22:
1) I've compiled few files into .o like this:

```
./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr  /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/WString.cpp -o /tmp/arduino_test1/WString.cpp.o
```

2)  created archive:
```
./llvm-ar rcs /tmp/arduino_test1/core.a /tmp/arduino_test1/WString.cpp.o
```

3) link like this:
```
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-gcc -w -Os -Wl,--gc-sections,--relax -mmcu=atmega328p -o /tmp/arduino_test1/Blink.cpp.elf /tmp/arduino_test1/Blink.cpp.o /tmp/arduino_test1/core.a -L/tmp/arduino_test1 -lm
```

I've got an error:
```
/tmp/arduino_test1/core.a: could not read symbols: Archive has no index; run ranlib to add one
collect2: error: ld returned 1 exit status
```

Should i need system linker or avr's clang from compiled binaries? i believe i've read linker is not provided and it will use system linker anyway. Is it correct?

4) Add index to archive:
```
./llvm-ranlib /tmp/arduino_test1/core.a
```

5) try to link again and still get the same error as in 3)

4ntoine, 2015-09-22:
if i try to use avr-llvm's 'clang' it does not accept some arguments:
```
MBA-Anton:bin asmirnov$ ./clang -w -Os -Wl,--gc-sections,--relax -mmcu=atmega328p -o /tmp/arduino_test1/Blink.cpp.elf /tmp/arduino_test1/Blink.cpp.o /tmp/arduino_test1/core.a -L/tmp/arduino_test1 -lm
ld: unknown option: --gc-sections
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
```

4ntoine, 2015-09-22:
files available for 14 days: https://dropmefiles.com/MQWZB

4ntoine, 2015-09-22:
```
MBA-Anton:bin asmirnov$ ./clang --version
clang version 3.8.0 (https://github.com/avr-llvm/clang.git 3b2c949e21022c841c80fca5e79d1f1387cd012c) (llvm/llvm.git 47a798c880f7baaf6b80b99e8653a22a28efc025)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
InstalledDir: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/.
```

4ntoine, 2015-09-22:
i was able to create index using avr-ranlib from binutils:
```
avr-ranlib /tmp/arduino_test1/core.a
```

but when i try to link i'm getting ld error:
```
MBA-Anton:bin asmirnov$ /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-gcc -w -Os -Wl,--gc-sections,--relax -mmcu=atmega328p -o /tmp/arduino_test1/Blink.cpp.elf /tmp/arduino_test1/Blink.cpp.o /tmp/arduino_test1/core.a -L/tmp/arduino_test1 -lm
/tmp/arduino_test1/core.a(HardwareSerial.cpp.o): In function `serialEventRun()':
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:(.text._Z14serialEventRunv+0x0): undefined reference to `serialEvent()'
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:(.text._Z14serialEventRunv+0x2): undefined reference to `serialEvent()'
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:(.text._Z14serialEventRunv+0x3a): undefined reference to `serialEvent()'
/tmp/arduino_test1/core.a(HardwareSerial.cpp.o):(.rodata._ZTI6Stream[typeinfo for Stream]+0x0): undefined reference to `vtable for __cxxabiv1::__si_class_type_info'
/tmp/arduino_test1/core.a(HardwareSerial.cpp.o):(.rodata._ZTI14HardwareSerial+0x0): undefined reference to `vtable for __cxxabiv1::__si_class_type_info'
/tmp/arduino_test1/core.a(Print.cpp.o):(.rodata._ZTI5Print+0x0): undefined reference to `vtable for __cxxabiv1::__class_type_info'
collect2: error: ld returned 1 exit status
```

4ntoine, 2015-09-22:
It seems that some special linker behavior is needed.
HardwareSerial.cpp code related to missing `serialEvent()`:
```
// this next line disables the entire HardwareSerial.cpp, 
// this is so I can support Attiny series and any other chip without a uart
#if defined(HAVE_HWSERIAL0) || defined(HAVE_HWSERIAL1) || defined(HAVE_HWSERIAL2) || defined(HAVE_HWSERIAL3)

// SerialEvent functions are weak, so when the user doesn't define them,
// the linker just sets their address to 0 (which is checked below).
// The Serialx_available is just a wrapper around Serialx.available(),
// but we can refer to it weakly so we don't pull in the entire
// HardwareSerial instance if the user doesn't also refer to it.
#if defined(HAVE_HWSERIAL0)
  void serialEvent() __attribute__((weak));
  bool Serial0_available() __attribute__((weak));
#endif

#if defined(HAVE_HWSERIAL1)
  void serialEvent1() __attribute__((weak));
  bool Serial1_available() __attribute__((weak));
#endif

#if defined(HAVE_HWSERIAL2)
  void serialEvent2() __attribute__((weak));
  bool Serial2_available() __attribute__((weak));
#endif

#if defined(HAVE_HWSERIAL3)
  void serialEvent3() __attribute__((weak));
  bool Serial3_available() __attribute__((weak));
#endif

void serialEventRun(void)
{
#if defined(HAVE_HWSERIAL0)
  if (Serial0_available && serialEvent && Serial0_available()) serialEvent();
#endif
#if defined(HAVE_HWSERIAL1)
  if (Serial1_available && serialEvent1 && Serial1_available()) serialEvent1();
#endif
#if defined(HAVE_HWSERIAL2)
  if (Serial2_available && serialEvent2 && Serial2_available()) serialEvent2();
#endif
#if defined(HAVE_HWSERIAL3)
  if (Serial3_available && serialEvent3 && Serial3_available()) serialEvent3();
#endif
}
```

dylanmckay, 2015-09-24:
Almost all LLVM tools in `llvm/bin/` are designed to solely be used for LLVM bitcode files. For example `llvm-link` combines two bitcode files into one, `llvm-as` assembles IR files into bitcode files.

`llvm-ar --help` prints

```
OVERVIEW: LLVM Archiver (llvm-ar)

This program archives bitcode files into single libraries
```

This tool is not the same as binutils `avr-ar`.

I only recently added linking support for AVR to `clang` (which is currently broken). It simply finds `avr-ld` and passes the target mcu, etc. It doesn't currently link in a runtime library (i.e. `libgcc`), so usage will probably trigger undefined symbol errors such as `undefined symbol: __udivmodqi4` when division is used and so on.

Because of this, **invoke the linker yourself, do not rely on AVR `clang` to link files correctly**.

dylanmckay, 2015-09-24:
The error

```
ld: unknown option: --gc-sections
```

is curious. These arguments are being directly passed to `avr-ld`. On my machine, `avr-ld --help` does indeed have `--gc-sections`. Could you do as the message says, and post the output of `clang -v` for verbose output.

Post the message as a new issue on the [`avr-llvm/clang`](https://github.com/avr-llvm/clang/issues/new) repository, as it is better suited there.

dylanmckay, 2015-09-24:
I have opened a new issue [here](https://github.com/avr-llvm/clang/issues/9) for the "weak symbol resolution" problem.

Closing this issue because all points have been addressed or moved into separate issues. Thanks for the bug report Anton :)

closed 2015-09-24 by dylanmckay