avr-llvm/llvm #166: TableGen modification breaks PPC backend

dylanmckay, 2015-09-30:
The `+` and `-` token handling modification made in 0553c45c6026de8328acef219f3643db7119e036 breaks the PowerPC backend.

PowerPC has the `blt- <address>`. PowerPC expects `blt` and `-` to be separate tokens, but our patch causes them to be combined into `blt-`. This is strange because this is the reason we introduced the patch - we wanted `+` and `-` to be considered separate.

This causes PowerPC tests to fail.

bootchk, 2016-02-24:
I see code in AVR's AsmWriter that handles this (comments there  reference the token handling modification of TableGen.)  IOW, the syntax of AVR assembly is quirky (has a trailing '+' to indicate a post-increment, instead of a 'I' appended to the opcode like many other assembly languages.)  Instead of modifying llvm generic TableGen code, you can use more conventional opcodes in early llvm passes, and at the Asm writing stage (custom to AVR) mangle the ASM text to the quirky syntax of the AVR ASM language.  Also, to generate the quirky AVR ASM syntax is optional, especially if llvm tools generate object code (no need to use any other assembler tools.)  I.E. it is a would-be-nice, not a blocker.

DavidEGrayson, 2016-02-24:
Being able to reuse inline assembly from an AVR-GCC program is a worthy goal.

dylanmckay, 2016-05-31:
At some point it looks like I have reverted this, and went pack to representing `reg+imm` as `memri` operands (see `AVRInstrInfo.td`) and adding spaces before `+` tokens in `TableGen`.

This is no longer an issue.

closed 2016-05-31 by dylanmckay