avr-llvm/llvm #168: Failed to link Arduino's HardwareSerial

4ntoine, 2015-10-01:
HardwareSerial.cpp
```
#if defined(HAVE_HWSERIAL0)
  void serialEvent() __attribute__((weak));
  bool Serial0_available() __attribute__((weak));
#endif
```

Due to weak issue () i have to add `serialEvent` in blink source code:
```
#line 1 "Blink.ino"
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.
 
  This example code is in the public domain.
 */
 
// Pin 13 has an LED connected on most Arduino boards.
// give it a name:
#include "Arduino.h"
void setup();
void loop();
#line 10
int led = 13;

// the setup routine runs once when you press reset:
void setup() {                
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);     
}

// TODO : remove
// just for testing until ((weak)) is supported
void serialEvent() {

}

// the loop routine runs over and over again forever:
void loop() {
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);               // wait for a second
}
``` 

compile .cpp into .o:
```
./clang -c -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp -o /tmp/arduino_test1/HardwareSerial.cpp.o
```

create archive
add archive index
link the whole app:

```
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -w -Os -Wl,--gc-sections,--relax -mmcu=atmega328p -o /tmp/arduino_test1/Blink.cpp.elf /tmp/arduino_test1/Blink.cpp.o /tmp/arduino_test1/core.a -L/tmp/arduino_test1 -lm --target=avr
```
Error:
```
/tmp/arduino_test1/core.a(HardwareSerial.cpp.o):(.rodata._ZTI6Stream[typeinfo for Stream]+0x0): undefined reference to `vtable for __cxxabiv1::__si_class_type_info'
/tmp/arduino_test1/core.a(HardwareSerial.cpp.o):(.rodata._ZTI14HardwareSerial+0x0): undefined reference to `vtable for __cxxabiv1::__si_class_type_info'
/tmp/arduino_test1/core.a(Print.cpp.o):(.rodata._ZTI5Print+0x0): undefined reference to `vtable for __cxxabiv1::__class_type_info'
collect2: error: ld returned 1 exit status
```

This seems to be something related to c/c++ mangling and linking..

dylanmckay, 2015-10-02:
Can you try passing `-fno-rtti`?

4ntoine, 2015-10-02:
Thanks, this helped

closed 2015-10-02 by 4ntoine

dylanmckay, 2015-10-02:
Sweet :)