avr-llvm/llvm #169: Q: hex is too large?

4ntoine, 2015-10-02:
I've found that avr-llvm's .elf is smaller but .hex is swice larger than according gcc's files:

```
MBA-Anton:arduino_test1 asmirnov$ ls -l ./Blink.cpp.elf 
-rwxr-xr-x  1 asmirnov  wheel  7455  2 окт 08:29 ./Blink.cpp.elf
MBA-Anton:arduino_test1 asmirnov$ ls -l ./Blink.cpp.hex 
-rw-r--r--  1 asmirnov  wheel  6808  2 окт 08:29 ./Blink.cpp.hex
MBA-Anton:arduino_test1 asmirnov$ 
MBA-Anton:arduino_test1 asmirnov$ 
MBA-Anton:arduino_test1 asmirnov$ 
MBA-Anton:arduino_test1 asmirnov$ cd /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/build2279877250002001901.tmp/
MBA-Anton:build2279877250002001901.tmp asmirnov$ ls -l ./Blink.cpp.elf 
-rwxr-xr-x  1 asmirnov  staff  15579  2 окт 08:22 ./Blink.cpp.elf
MBA-Anton:build2279877250002001901.tmp asmirnov$ ls -l ./Blink.cpp.hex 
-rw-r--r--  1 asmirnov  staff  2918  2 окт 08:22 ./Blink.cpp.hex
```

Compilation is done without debug symbols (no '-g') with '-Os':
```
./clang -c -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR  -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/standard /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/hooks.c -o /tmp/arduino_test1/hooks.c.o -I/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/avr/include --target=avr
```

4ntoine, 2015-10-02:
Using the same tools to link and generate hex from elf:
```
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -w -Os -Wl,--gc-sections,--relax -mmcu=atmega328p -o /tmp/arduino_test1/Blink.cpp.elf /tmp/arduino_test1/Blink.cpp.o /tmp/arduino_test1/core.a -L/tmp/arduino_test1 -lm --target=avr 

/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_test1/Blink.cpp.elf /tmp/arduino_test1/Blink.cpp.eep

/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_test1/Blink.cpp.elf /tmp/arduino_test1/Blink.cpp.hex
```

dylanmckay, 2015-10-02:
I edited your first post to make it more readable.

Is the first `Blink.cpp.elf` and `Blink.cpp.hex` from GCC, or is that LLVM?

dylanmckay, 2015-10-02:
Are you comparing the size of the files generated by LLVM and GCC, or are you comparing the size of the hex and elf files just from LLVM?

4ntoine, 2015-10-02:
the first is from avr-llvm, the second is from gcc (Arduino IDE)

dylanmckay, 2016-09-24:
Closing, we can reopen this later on if it doesn't get fixed naturally.

closed 2016-09-24 by dylanmckay