avr-llvm/llvm #170: Mailing list archive inaccessible/403 forbidden.

lberezy, 2015-10-27:
The mailing list archives for both `dev` and `users` are inaccessible and return a 403 Forbidden code when access is attempted.

http://lists.avr-llvm.org/pipermail/users/ returns 403 Forbidden.

dylanmckay, 2015-10-27:
I'm not sure if it's worth fixing this issue - due to the very small number of contributors, the list was only actually used by me to notify a few people about milestones a few times. We are also in the process of moving into LLVM upstream, so the discussion on development will really have to move to the LLVM mailing lists.

If there still is interest, I will work on it however.

lberezy, 2015-10-28:
That's half the reason I was looking for the mailing list archives - I know
there's some progress with getting things merged upstream
(congratulations), I just wanted to understand the current status.

I've done a quick search on code review and mailing lists upstream for
"AVR", though nothing was overly clear.

dylanmckay, 2015-10-28:
I think there were three emails sent in the mailing lists, all by me.

The upstreaming process is slow.

Two weeks ago in [r250492](http://llvm.org/viewvc/llvm-project?diff_format=h&view=revision&revision=250492) I added the AVR target triple definitions.

Four days ago the public AVR ELF definitions landed in [r251096](http://llvm.org/viewvc/llvm-project?diff_format=h&view=revision&revision=251096).

After [this](http://lists.llvm.org/pipermail/llvm-dev/2015-October/091708.html) discussion on LLVMDev, I became the code owner for the LLVM backend in [r251471](http://llvm.org/viewvc/llvm-project/llvm/trunk/CODE_OWNERS.TXT?r1=251367&r2=251471&pathrev=251471&diff_format=h) this morning.

The first real migration of AVR code is currently in review in [D14039](http://reviews.llvm.org/D14039).

dylanmckay, 2015-10-28:
Development has halted until the migration is completed. After that I will probably work on #149, as it is blocking [avr-rust](https://github.com/avr-rust/rust).

DavidEGrayson, 2015-10-28:
@dylanmckay Thank you for this status update.  I intend to make a PIC18 backend for LLVM at some point, so it will be interesting to follow your progress and learn from it.

@lberezy We can get the latest updates by looking at http://reviews.llvm.org/p/dylanmckay/feed/ and by running `svn log | grep --after-context=3 dylanmckay` in the appropriate SVN repository.

lberezy, 2015-10-28:
Thanks for the update.
I'm super excited for avr-rust and in-general, all that LLVM for AVR will do for the community.

dylanmckay, 2015-10-28:
Me too, the guarantees of Rust would be perfect for AVR!

dylanmckay, 2016-01-19:
Closing as there will be no need for a mailing list once upstreaming is completed.

closed 2016-01-19 by dylanmckay