avr-llvm/llvm #172: Assertion failure: Impossible reg-to-reg copy at AVRInstrInfo.cpp:62

shepmaster, 2016-01-18:
While attempting to compile parts of Rust's libcore:

```
Impossible reg-to-reg copy
UNREACHABLE executed at avr-rust/src/llvm/lib/Target/AVR/AVRInstrInfo.cpp:62!
```

```llvm
; ModuleID = 'bugpoint-reduced-simplified.bc'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr-atmel-none"

; Function Attrs: nounwind readnone
declare { i16, i1 } @llvm.umul.with.overflow.i16(i16, i16) #0

; Function Attrs: inlinehint uwtable
define void @_ZN3ops15usize.MulAssign10mul_assign20h9dc48c05c6ade4e6iRkE() unnamed_addr #1 {
entry-block:
  %0 = call { i16, i1 } @llvm.umul.with.overflow.i16(i16 undef, i16 undef)
  %1 = extractvalue { i16, i1 } %0, 1
  %2 = icmp eq i1 %1, true
  br i1 %2, label %cond, label %next

next:                                             ; preds = %entry-block
  ret void

cond:                                             ; preds = %entry-block
  unreachable
}

attributes #0 = { nounwind readnone }
attributes #1 = { inlinehint uwtable }
```

shepmaster, 2016-01-18:
Maybe of interest, but this only is trigged if I compile the Rust code without optimizations. With optimizations, the code is reordered in such a way that it doesn't trigger this issue.

dylanmckay, 2016-01-19:
Reduced testcase:

```
declare { i16, i1 } @llvm.umul.with.overflow.i16(i16, i16)

define void @foo() {
entry-block:
  %0 = call { i16, i1 } @llvm.umul.with.overflow.i16(i16 undef, i16 undef)
  %1 = extractvalue { i16, i1 } %0, 1
  %2 = icmp eq i1 %1, true
  br i1 %2, label %cond, label %next

next:                                             ; preds = %entry-block
  ret void
cond:                                             ; preds = %entry-block
  unreachable
}
```

Added to `test/CodeGen/AVR/impossible-reg-to-reg-copy.ll` in 5d82a2734281af8484d6c76fd32e2891c7ac3fc6.

shepmaster, 2016-01-19:
While debugging, I see that `DestReg=49` and `SrcReg=29`.

shepmaster, 2016-01-19:
But I'm finding it difficult to even see where `AVR::GPR8RegClass` is even defined... grep finds nothing.

dylanmckay, 2016-01-19:
`AVR::GPR8RegClass` is generated from the TableGen description (`AVRRegisterInfo.td`) file. Note that it works this way for all of the TableGen files.

It will be placed in the build directory, with the name `AVRGenRegisterInfo.inc`. You can match up the `DestReg` and `SrcReg` with the values in the registers `enum`.

shepmaster, 2016-01-20:
> in the build directory, with the name AVRGenRegisterInfo.inc

Ah, it ends in `.inc`. I was assuming `.h` and killed my grep of all files because it was too slow.

So `DestReg=R25R24` and `SrcReg=R24`.

dylanmckay, 2016-01-20:
R25R24 is a 16bit register with R25 as the high byte and R24 as the low
byte.

I'm not sure how we generated code that tried to copy from one register to
another register of different sizes (illegal), and even tried to copy it
into itself.

shepmaster, 2016-05-15:
Fixed in #190

dylanmckay, 2016-05-20:
Closing as #190 has been merged.

closed 2016-05-20 by dylanmckay