avr-llvm/llvm #173: Assertion failure: LowerFormalArguments didn't emit the correct number of values at SelectionDAGBuilder.cpp:7405

shepmaster, 2016-01-18:
```
Assertion failed: (InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of values!"), function LowerArguments, file /Users/shep/Projects/avr-rust/src/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp, line 7405.
```

When running `llc -march=avr -mcpu=atmega328p -filetype=obj bug2.ll` on

```llvm
; ModuleID = 'bugpoint-reduced-simplified.bc'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr-atmel-none"

define internal void @_ZN3ops16bool.BitOrAssign12bitor_assign20hf3e6510977e16c6cX4kE(i8* noalias dereferenceable(1), i1 zeroext) unnamed_addr #2 {
entry-block:
  ret void
}
```

This was an intermediate `bugpoint` result from reducing #172.

dylanmckay, 2016-01-19:
Reduced to:

``` llvm
define void @foo(i1) {
  ret void
}
```

Added test case `test/CodeGen/AVR/lower-formal-arguments-assertion.ll` in 5d82a2734281af8484d6c76fd32e2891c7ac3fc6.

dylanmckay, 2016-01-19:
Changing `i1` to `i8` fixes the problem. This is probably caused by the fact that `i1` is not a valid type on AVR, and we should be promoting it to `i8`.

shepmaster, 2016-01-19:
I think you meant "**i1** values not being promoted to i8 in function arguments" In the title?

shepmaster, 2016-01-19:
> we should be promoting it to i8.

Based on my limited knowledge, it [seems like that's happening](https://github.com/avr-llvm/llvm/blob/3c018439744ab391d32c106fe01e466a77b812e4/lib/Target/AVR/AVRISelLowering.cpp#L53-L57). I'm guessing that stuff needs to be done elsewhere.

PR #180 (2016-01-20): 1-bit types consume one byte of space, not zero
Integer division was rounding off `i1` to take 0 bytes.

Closes #173
commit 29f9ff0a (2016-01-20): 1-bit types consume one byte of space, not zero

Integer division was rounding off `i1` to take 0 bytes.

Closes #173
closed 2016-01-20 by dylanmckay