avr-llvm/llvm #175: Assertion failure: Call result #4 has unhandled type i16 at CallingConvLower.cpp:167

shepmaster, 2016-01-18:
```
Call result #4 has unhandled type i16
UNREACHABLE executed at avr-rust/src/llvm/lib/CodeGen/CallingConvLower.cpp:167!
```

Running `llc -march=avr -mcpu=atmega328p -filetype=obj bugpoint-reduced-simplified.ll` on:

```llvm
; ModuleID = 'bugpoint-reduced-simplified.bc'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr-atmel-none"

; Function Attrs: inlinehint nounwind uwtable
define void @_ZN3ops13i64.RemAssign10rem_assign20h23f1175739be271cXrkE(i64* noalias nocapture dereferenceable(8), i64) unnamed_addr #0 {
entry-block:
  %2 = srem i64 0, %1
  store i64 %2, i64* %0, align 1
  ret void
}

attributes #0 = { inlinehint nounwind uwtable }
```

dylanmckay, 2016-01-19:
Reduced testcase:

``` llvm
define void @foo(i64*, i64) {
  %remainder = srem i64 0, %1
  store i64 %remainder, i64* %0, align 1
  ret void
}
```

Added to `test/CodeGen/AVR/calling-conv-assertion.ll` in 5d82a2734281af8484d6c76fd32e2891c7ac3fc6.

shepmaster, 2016-01-21:
Ok, so the error message means that the result of a function call (specifically index `4`) of type `i16` could not be fit within the calling convention. There are actually *8* values to be returned of type `i16`, and only [4 register pairs available](https://github.com/avr-llvm/llvm/blob/cfc3f431c02364847df4c9b09b4cea54a0669e43/lib/Target/AVR/AVRCallingConv.td#L22): `R25R24`, `R23R22`, `R21R20`, `R19R18`.

shepmaster, 2016-01-21:
Slightly smaller:

```llvm
define i64 @foo(i64) {
  %remainder = srem i64 0, %0
  ret i64 %remainder
}
```

shepmaster, 2016-01-21:
An interesting thing is that the version that *returns a `void`* has this issue. Since it's deliberately storing into a pointer, I wouldn't expect there to be any issue here...

shepmaster, 2016-01-21:
Tracking more debugging, it appears to be happening because of the `sdivrem` call specifically, as it would return `i64, i64`, which corresponds to the 8x`i16` spots.

dylanmckay, 2016-01-22:
Could you check what the calling convention is? `C`, `AVR_BUILTIN`, etc.

shepmaster, 2016-01-22:
The calling convention is `0`, which should be `C`. At [this line](https://github.com/avr-llvm/llvm/blob/cfc3f431c02364847df4c9b09b4cea54a0669e43/lib/Target/AVR/AVRISelLowering.cpp#L1271), it's using the `RetCC_AVR` set.

shepmaster, 2016-05-15:
Fixed in #194

closed 2016-05-27 by shepmaster