avr-llvm/llvm #177: Crashed while compiling "Stream.cpp"

4ntoine, 2016-01-19:
MBA-Anton:bin asmirnov$ ./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp" -o "/tmp/arduino_test1/Stream.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
Assertion failed: (InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of values!"), function LowerArguments, file /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp, line 7405.
0  clang-3.8                0x0000000106cd165e llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 46
1  clang-3.8                0x0000000106cd32f9 PrintStackTraceSignalHandler(void*) + 25
2  clang-3.8                0x0000000106ccf8f9 llvm::sys::RunSignalHandlers() + 425
3  clang-3.8                0x0000000106cd3639 SignalHandler(int) + 345
4  libsystem_platform.dylib 0x00007fff98cb7f1a _sigtramp + 26
5  libsystem_platform.dylib 0x0000000000000002 _sigtramp + 1731494146
6  clang-3.8                0x0000000106cd331b raise + 27
7  clang-3.8                0x0000000106cd33d2 abort + 18
8  clang-3.8                0x0000000106cd33b1 __assert_rtn + 129
9  clang-3.8                0x0000000107b84e42 llvm::SelectionDAGISel::LowerArguments(llvm::Function const&) + 4402
10 clang-3.8                0x0000000107bc7deb llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) + 5083
11 clang-3.8                0x0000000107bc5590 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 1136
12 clang-3.8                0x0000000105b2c3bb llvm::AVRDAGToDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 59
13 clang-3.8                0x000000010601960e llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 110
14 clang-3.8                0x00000001064f9d9d llvm::FPPassManager::runOnFunction(llvm::Function&) + 413
15 clang-3.8                0x00000001064fa0d5 llvm::FPPassManager::runOnModule(llvm::Module&) + 117
16 clang-3.8                0x00000001064fae2a (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) + 2010
17 clang-3.8                0x00000001064fa3bb llvm::legacy::PassManagerImpl::run(llvm::Module&) + 347
18 clang-3.8                0x00000001064fba71 llvm::legacy::PassManager::run(llvm::Module&) + 33
19 clang-3.8                0x0000000106f9f11c (anonymous namespace)::EmitAssemblyHelper::EmitAssembly(clang::BackendAction, llvm::raw_pwrite_stream*) + 1900
20 clang-3.8                0x0000000106f9e1c2 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::raw_pwrite_stream*) + 162
21 clang-3.8                0x000000010725c154 clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) + 1524
22 clang-3.8                0x000000010870797e clang::ParseAST(clang::Sema&, bool, bool) + 1294
23 clang-3.8                0x00000001077b3cef clang::ASTFrontendAction::ExecuteAction() + 511
24 clang-3.8                0x000000010725a514 clang::CodeGenAction::ExecuteAction() + 5988
25 clang-3.8                0x00000001077b3250 clang::FrontendAction::Execute() + 112
26 clang-3.8                0x000000010770cbc8 clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) + 1592
27 clang-3.8                0x000000010784131a clang::ExecuteCompilerInvocation(clang::CompilerInstance*) + 4410
28 clang-3.8                0x0000000105afe1eb cc1_main(llvm::ArrayRef<char const*>, char const*, void*) + 4923
29 clang-3.8                0x0000000105aede4c ExecuteCC1Tool(llvm::ArrayRef<char const*>, llvm::StringRef) + 476
30 clang-3.8                0x0000000105aeb9ac main + 3244
31 libdyld.dylib            0x00007fff966e85c9 start + 1
32 libdyld.dylib            0x0000000000000043 start + 1771141755
Stack dump:
0.	Program arguments: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.8 -cc1 -triple avr -emit-obj -disable-free -main-file-name Stream.cpp -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -target-cpu atmega328p -target-linker-version 253.9 -dwarf-column-info -debug-info-kind=limited -dwarf-version=4 -ffunction-sections -fdata-sections -coverage-file /tmp/arduino_test1/Stream.cpp.o -resource-dir /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/../lib/clang/3.8.0 -dependency-file /tmp/arduino_test1/Stream.cpp.d -MT /tmp/arduino_test1/Stream.cpp.o -D F_CPU=16000000L -D ARDUINO=10607 -D ARDUINO_AVR_UNO -D ARDUINO_ARCH_AVR -I /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard -I /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include -Os -std=gnu++11 -fdeprecated-macro -fdebug-compilation-dir /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin -ferror-limit 19 -fmessage-length 138 -fno-threadsafe-statics -fobjc-runtime=gcc -fdiagnostics-show-option -fcolor-diagnostics -vectorize-loops -vectorize-slp -o /tmp/arduino_test1/Stream.cpp.o -x c++ /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp 
1.	<eof> parser at end of file
2.	Code generation
3.	Running pass 'Function Pass Manager' on module '/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp'.
4.	Running pass 'AVR DAG->DAG Instruction Selection' on function '@_ZN6Stream13peekNextDigitE13LookaheadModeb'
clang-3.8: error: unable to execute command: Illegal instruction: 4
clang-3.8: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.8.0 (https://github.com/avr-llvm/clang.git 55d2db19d360a92b3b87444a2ea5e07d1624a2bc) (llvm/llvm.git ed02a4bc0484efbf1362cf04520976ea52d4351f)
Target: avr
Thread model: posix
InstalledDir: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/.
clang-3.8: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.8: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.8: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Stream-076922.cpp
clang-3.8: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Stream-076922.sh
clang-3.8: note: diagnostic msg: 

********************

4ntoine, 2016-01-19:
https://gist.github.com/4ntoine/df8ce91cf24271edfe6b

4ntoine, 2016-01-19:
Stream-076922.sh:

# Crash reproducer for clang version 3.8.0 (https://github.com/avr-llvm/clang.git 55d2db19d360a92b3b87444a2ea5e07d1624a2bc) (llvm/llvm.git ed02a4bc0484efbf1362cf04520976ea52d4351f)
# Driver args: "-c" "-g" "-Os" "-std=gnu++11" "-fno-exceptions" "-ffunction-sections" "-fdata-sections" "-fno-threadsafe-statics" "-MMD" "-mmcu=atmega328p" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp" "-o" "/tmp/arduino_test1/Stream.cpp.o" "-I" "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include" "--target=avr"
# Original command:  "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.8" "-cc1" "-triple" "avr" "-emit-obj" "-disable-free" "-main-file-name" "Stream.cpp" "-mrelocation-model" "static" "-mthread-model" "posix" "-mdisable-fp-elim" "-fmath-errno" "-masm-verbose" "-mconstructor-aliases" "-target-cpu" "atmega328p" "-target-linker-version" "253.9" "-dwarf-column-info" "-debug-info-kind=limited" "-dwarf-version=4" "-ffunction-sections" "-fdata-sections" "-coverage-file" "/tmp/arduino_test1/Stream.cpp.o" "-resource-dir" "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/../lib/clang/3.8.0" "-dependency-file" "/tmp/arduino_test1/Stream.cpp.d" "-MT" "/tmp/arduino_test1/Stream.cpp.o" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I" "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include" "-Os" "-std=gnu++11" "-fdeprecated-macro" "-fdebug-compilation-dir" "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin" "-ferror-limit" "19" "-fmessage-length" "138" "-fno-threadsafe-statics" "-fobjc-runtime=gcc" "-fdiagnostics-show-option" "-fcolor-diagnostics" "-vectorize-loops" "-vectorize-slp" "-o" "/tmp/arduino_test1/Stream.cpp.o" "-x" "c++" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp"
 "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.8" "-cc1" "-triple" "avr" "-emit-obj" "-disable-free" "-main-file-name" "Stream.cpp" "-mrelocation-model" "static" "-mthread-model" "posix" "-mdisable-fp-elim" "-fmath-errno" "-masm-verbose" "-mconstructor-aliases" "-target-cpu" "atmega328p" "-target-linker-version" "253.9" "-dwarf-column-info" "-debug-info-kind=limited" "-dwarf-version=4" "-ffunction-sections" "-fdata-sections" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-Os" "-std=gnu++11" "-fdeprecated-macro" "-ferror-limit" "19" "-fmessage-length" "138" "-fno-threadsafe-statics" "-fobjc-runtime=gcc" "-fdiagnostics-show-option" "-fcolor-diagnostics" "-vectorize-loops" "-vectorize-slp" "-x" "c++" "Stream-076922.cpp"

dylanmckay, 2016-01-19:
Please wrap command output in code blocks as well.

dylanmckay, 2016-01-19:
Could you generate IR assembly by passing `-S -emit-llvm` options to clang. This is likely a duplicate of #173.

4ntoine, 2016-01-19:
no error with -S -emit-llvm:
```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os -S -emit-llvm  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp" -o "/tmp/arduino_test1/Stream.cpp.ll" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
MBA-Anton:bin asmirnov$ 
```

generated ll: https://gist.github.com/4ntoine/cabf1294838eab09a40e

dylanmckay, 2016-01-19:
The IR also has `i1` types in function signatures. Closing as this is a duplicate of #173.

closed 2016-01-19 by dylanmckay