avr-llvm/llvm #179: Compile blink.cpp generates non-binary object file

4ntoine, 2016-01-19:
```
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp.o"
``` generates binary file, but
```
./clang++ -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard ~/Documents/dev/tmp/avr-clang/Blink.cpp -o /tmp/arduino_test1/Blink.cpp.o -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
```
generates some text file (incorrect).

Later that file is not recognized by linker:
```
MBA-Anton:bin asmirnov$ "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc"  -Os -Wl,--gc-sections -mmcu=atmega328p  -o "/tmp/arduino_test1/Blink.ino.elf" "/tmp/arduino_test1/Blink.cpp.o" "/tmp/arduino_test1/core.a" "-L/tmp/arduino_test1" -lm --target=avr 
/tmp/arduino_test1/Blink.cpp.o: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
```

4ntoine, 2016-01-19:
Blink.cpp:
```
#line 1 "Blink.ino"
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.
 
  This example code is in the public domain.
 */
 
// Pin 13 has an LED connected on most Arduino boards.
// give it a name:
#include "Arduino.h"
void setup();
void loop();
#line 10
int led = 13;

// the setup routine runs once when you press reset:
void setup() {                
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);     
}

// TODO : remove
// just for testing until ((weak)) is supported
/*
void serialEvent() {

}
*/

// the loop routine runs over and over again forever:
void loop() {
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);               // wait for a second
}
```

4ntoine, 2016-01-19:
Blink.cpp.o https://gist.github.com/4ntoine/4309ff1e94d09cc3071f

4ntoine, 2016-01-19:
looks like it was not compiled at all..

dylanmckay, 2016-01-19:
Try to always generate `.ll` files and see if they are reproduced when running against `llc`. That makes it easier to reproduce, and also makes it clear whether this is an LLVM or a clang problem.

It also makes it especially easier for me to test as Arduino and its dependencies and endless `clang` arguments are time-consuming to work with. With `.ll` files, we can also use the `bugpoint` tool to automatically minimize the test case down to a dozen or so offending lines of code.

That said, this is a weird error. I notice `-w` is being passed to `clang`, which silences all warnings, which might be a problem. 

Try double checking that you aren't passing `-S` to `clang`. If you really aren't, then this really is a strange error.

4ntoine, 2016-01-19:
```
MBA-Anton:bin asmirnov$ ./clang -S -emit-llvm -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard ~/Documents/dev/tmp/avr-clang/Blink.cpp -o /tmp/arduino_test1/Blink.cpp.ll -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
```

produces not LLVM byte code too: https://gist.github.com/4ntoine/bcdc796ec8a47652456c

Sh4rK, 2016-01-19:
The command line for gcc and clang differs. You are passing `-E` to clang, which means "preprocess only",  so that's what you get.

4ntoine, 2016-01-19:
Oh, i expected clang to be drop-in replacement for gcc and clang's driver to support gcc syntax. I will try. Thanks for the suggestion

4ntoine, 2016-01-19:
Confirmed! `-E and -CC` should be removed

closed 2016-01-19 by 4ntoine

Sh4rK, 2016-01-19:
It IS a drop in replacement (mostly), but the command line you are passing them is different, check your first comment. (The gcc line doesn't have -E)

4ntoine, 2016-01-19:
You're right again. I was not careful enough while analyzing original Arduino IDE invocations:
```
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/dev/null"

"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/dev/null"

"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"

"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino5/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"

"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp.o"
```