avr-llvm/llvm #181: Crash while compiling HardwareSerial0.cpp

4ntoine, 2016-01-20:
```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp" -o "/tmp/arduino_test1/HardwareSerial0.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
Assertion failed: (Val && "isa<> used on a null pointer"), function doit, file /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/include/llvm/Support/Casting.h, line 95.
0  clang-3.9                0x0000000103c5a20e llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 46
1  clang-3.9                0x0000000103c5a659 PrintStackTraceSignalHandler(void*) + 25
2  clang-3.9                0x0000000103c56df9 llvm::sys::RunSignalHandlers() + 425
3  clang-3.9                0x0000000103c5a9b9 SignalHandler(int) + 345
4  libsystem_platform.dylib 0x00007fff98cb7f1a _sigtramp + 26
5  libsystem_platform.dylib 0x00007fff69b65784 _sigtramp + 3505051780
6  clang-3.9                0x0000000103c5a67b raise + 27
7  clang-3.9                0x0000000103c5a732 abort + 18
8  clang-3.9                0x0000000103c5a711 __assert_rtn + 129
9  clang-3.9                0x0000000106a89a5c llvm::isa_impl_cl<clang::FunctionDecl, clang::Decl const*>::doit(clang::Decl const*) + 92
10 clang-3.9                0x0000000106a899e8 llvm::isa_impl_wrap<clang::FunctionDecl, clang::Decl const*, clang::Decl const*>::doit(clang::Decl const* const&) + 24
11 clang-3.9                0x0000000106a899c2 llvm::isa_impl_wrap<clang::FunctionDecl, clang::Decl const* const, clang::Decl const*>::doit(clang::Decl const* const&) + 34
12 clang-3.9                0x0000000106a5b595 bool llvm::isa<clang::FunctionDecl, clang::Decl const*>(clang::Decl const* const&) + 21
13 clang-3.9                0x0000000106a5a4d8 llvm::cast_retty<clang::FunctionDecl, clang::Decl const*>::ret_type llvm::dyn_cast<clang::FunctionDecl, clang::Decl const>(clang::Decl const*) + 24
14 clang-3.9                0x0000000104458741 (anonymous namespace)::AVRTargetCodeGenInfo::setTargetAttributes(clang::Decl const*, llvm::GlobalValue*, clang::CodeGen::CodeGenModule&) const + 33
15 clang-3.9                0x000000010423b19e clang::CodeGen::CodeGenModule::setNonAliasAttributes(clang::Decl const*, llvm::GlobalObject*) + 190
16 clang-3.9                0x000000010423bd25 clang::CodeGen::CodeGenModule::SetInternalFunctionAttributes(clang::Decl const*, llvm::Function*, clang::CodeGen::CGFunctionInfo const&) + 117
17 clang-3.9                0x000000010406c196 clang::CodeGen::CodeGenModule::CreateGlobalInitOrDestructFunction(llvm::FunctionType*, llvm::Twine const&, clang::CodeGen::CGFunctionInfo const&, clang::SourceLocation, bool) + 246
18 clang-3.9                0x000000010406c8ce clang::CodeGen::CodeGenModule::EmitCXXGlobalVarDeclInitFunc(clang::VarDecl const*, llvm::GlobalVariable*, bool) + 542
19 clang-3.9                0x0000000104245081 clang::CodeGen::CodeGenModule::EmitGlobalVarDefinition(clang::VarDecl const*, bool) + 3521
20 clang-3.9                0x000000010423df9d clang::CodeGen::CodeGenModule::EmitGlobalDefinition(clang::GlobalDecl, llvm::GlobalValue*) + 845
21 clang-3.9                0x0000000104242e48 clang::CodeGen::CodeGenModule::EmitGlobal(clang::GlobalDecl) + 1208
22 clang-3.9                0x00000001042465d4 clang::CodeGen::CodeGenModule::EmitTopLevelDecl(clang::Decl*) + 404
23 clang-3.9                0x00000001043b38fd (anonymous namespace)::CodeGeneratorImpl::HandleTopLevelDecl(clang::DeclGroupRef) + 157
24 clang-3.9                0x00000001042181c0 clang::BackendConsumer::HandleTopLevelDecl(clang::DeclGroupRef) + 192
25 clang-3.9                0x000000010569b769 clang::ParseAST(clang::Sema&, bool, bool) + 969
26 clang-3.9                0x0000000104780bbf clang::ASTFrontendAction::ExecuteAction() + 511
27 clang-3.9                0x0000000104216c07 clang::CodeGenAction::ExecuteAction() + 5991
28 clang-3.9                0x0000000104780120 clang::FrontendAction::Execute() + 112
29 clang-3.9                0x00000001046cd5c1 clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) + 1873
30 clang-3.9                0x000000010480ffaa clang::ExecuteCompilerInvocation(clang::CompilerInstance*) + 4410
31 clang-3.9                0x0000000102985b9b cc1_main(llvm::ArrayRef<char const*>, char const*, void*) + 4923
32 clang-3.9                0x000000010297598c ExecuteCC1Tool(llvm::ArrayRef<char const*>, llvm::StringRef) + 476
33 clang-3.9                0x00000001029734ec main + 3244
34 libdyld.dylib            0x00007fff966e85c9 start + 1
Stack dump:
0.	Program arguments: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.9 -cc1 -triple avr -emit-obj -disable-free -main-file-name HardwareSerial0.cpp -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -target-cpu atmega328p -target-linker-version 253.9 -dwarf-column-info -debug-info-kind=limited -dwarf-version=4 -debugger-tuning=gdb -ffunction-sections -fdata-sections -coverage-file /tmp/arduino_test1/HardwareSerial0.cpp.o -resource-dir /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/../lib/clang/3.9.0 -dependency-file /tmp/arduino_test1/HardwareSerial0.cpp.d -MT /tmp/arduino_test1/HardwareSerial0.cpp.o -D F_CPU=16000000L -D ARDUINO=10607 -D ARDUINO_AVR_UNO -D ARDUINO_ARCH_AVR -I /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard -I /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include -Os -std=gnu++11 -fdeprecated-macro -fdebug-compilation-dir /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin -ferror-limit 19 -fmessage-length 116 -fno-threadsafe-statics -fobjc-runtime=gcc -fdiagnostics-show-option -fcolor-diagnostics -vectorize-loops -vectorize-slp -o /tmp/arduino_test1/HardwareSerial0.cpp.o -x c++ /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp 
1.	/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp:75:1 <Spelling=/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/../lib/clang/3.9.0/include/stdbool.h:37:15>: current parser token 'bool'
2.	/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp:70:18: LLVM IR generation of declaration 'Serial'
3.	/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp:70:18: Generating code for declaration 'Serial'
clang-3.9: error: unable to execute command: Illegal instruction: 4
clang-3.9: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.9.0 (https://github.com/avr-llvm/clang.git 57dc8311e802b5cf9a752b4c76f5d2ef0faa5174) (llvm/llvm.git 63488b8be0ea6b8676043bef06058676fd01bcca)
Target: avr
Thread model: posix
InstalledDir: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/.
clang-3.9: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.9: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.9: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/HardwareSerial0-84069f.cpp
clang-3.9: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/HardwareSerial0-84069f.sh
clang-3.9: note: diagnostic msg: 

********************
```

4ntoine, 2016-01-20:
yesterday(or 2 days ago) it worked, so it's because of some latest changes

4ntoine, 2016-01-20:
HardwareSerial0-84069f.cpp - https://gist.github.com/4ntoine/367c67891c0738e669fa

4ntoine, 2016-01-20:
HardwareSerial0-84069f.sh:
```
# Crash reproducer for clang version 3.9.0 (https://github.com/avr-llvm/clang.git 57dc8311e802b5cf9a752b4c76f5d2ef0faa5174) (llvm/llvm.git 63488b8be0ea6b8676043bef06058676fd01bcca)
# Driver args: "-c" "-g" "-Os" "-std=gnu++11" "-fno-exceptions" "-ffunction-sections" "-fdata-sections" "-fno-threadsafe-statics" "-MMD" "-mmcu=atmega328p" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp" "-o" "/tmp/arduino_test1/HardwareSerial0.cpp.o" "-I" "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include" "--target=avr"
# Original command:  "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.9" "-cc1" "-triple" "avr" "-emit-obj" "-disable-free" "-main-file-name" "HardwareSerial0.cpp" "-mrelocation-model" "static" "-mthread-model" "posix" "-mdisable-fp-elim" "-fmath-errno" "-masm-verbose" "-mconstructor-aliases" "-target-cpu" "atmega328p" "-target-linker-version" "253.9" "-dwarf-column-info" "-debug-info-kind=limited" "-dwarf-version=4" "-debugger-tuning=gdb" "-ffunction-sections" "-fdata-sections" "-coverage-file" "/tmp/arduino_test1/HardwareSerial0.cpp.o" "-resource-dir" "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/../lib/clang/3.9.0" "-dependency-file" "/tmp/arduino_test1/HardwareSerial0.cpp.d" "-MT" "/tmp/arduino_test1/HardwareSerial0.cpp.o" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I" "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include" "-Os" "-std=gnu++11" "-fdeprecated-macro" "-fdebug-compilation-dir" "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin" "-ferror-limit" "19" "-fmessage-length" "116" "-fno-threadsafe-statics" "-fobjc-runtime=gcc" "-fdiagnostics-show-option" "-fcolor-diagnostics" "-vectorize-loops" "-vectorize-slp" "-o" "/tmp/arduino_test1/HardwareSerial0.cpp.o" "-x" "c++" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp"
 "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.9" "-cc1" "-triple" "avr" "-emit-obj" "-disable-free" "-main-file-name" "HardwareSerial0.cpp" "-mrelocation-model" "static" "-mthread-model" "posix" "-mdisable-fp-elim" "-fmath-errno" "-masm-verbose" "-mconstructor-aliases" "-target-cpu" "atmega328p" "-target-linker-version" "253.9" "-dwarf-column-info" "-debug-info-kind=limited" "-dwarf-version=4" "-debugger-tuning=gdb" "-ffunction-sections" "-fdata-sections" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-Os" "-std=gnu++11" "-fdeprecated-macro" "-ferror-limit" "19" "-fmessage-length" "116" "-fno-threadsafe-statics" "-fobjc-runtime=gcc" "-fdiagnostics-show-option" "-fcolor-diagnostics" "-vectorize-loops" "-vectorize-slp" "-x" "c++" "HardwareSerial0-84069f.cpp"
```

4ntoine, 2016-01-20:
new crash (seems to be the same reason):

```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial1.cpp" -o "/tmp/arduino_test1/HardwareSerial1.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
MBA-Anton:bin asmirnov$ ./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial2.cpp" -o "/tmp/arduino_test1/HardwareSerial2.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
MBA-Anton:bin asmirnov$ ./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial3.cpp" -o "/tmp/arduino_test1/HardwareSerial3.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
MBA-Anton:bin asmirnov$ ./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/IPAddress.cpp" -o "/tmp/arduino_test1/IPAddress.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
Assertion failed: (Val && "isa<> used on a null pointer"), function doit, file /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/include/llvm/Support/Casting.h, line 95.
0  clang-3.9                0x000000010c60f20e llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 46
1  clang-3.9                0x000000010c60f659 PrintStackTraceSignalHandler(void*) + 25
2  clang-3.9                0x000000010c60bdf9 llvm::sys::RunSignalHandlers() + 425
3  clang-3.9                0x000000010c60f9b9 SignalHandler(int) + 345
4  libsystem_platform.dylib 0x00007fff98cb7f1a _sigtramp + 26
5  libsystem_platform.dylib 0x0000000000000002 _sigtramp + 1731494146
6  clang-3.9                0x000000010c60f67b raise + 27
7  clang-3.9                0x000000010c60f732 abort + 18
8  clang-3.9                0x000000010c60f711 __assert_rtn + 129
9  clang-3.9                0x000000010f43ea5c llvm::isa_impl_cl<clang::FunctionDecl, clang::Decl const*>::doit(clang::Decl const*) + 92
10 clang-3.9                0x000000010f43e9e8 llvm::isa_impl_wrap<clang::FunctionDecl, clang::Decl const*, clang::Decl const*>::doit(clang::Decl const* const&) + 24
11 clang-3.9                0x000000010f43e9c2 llvm::isa_impl_wrap<clang::FunctionDecl, clang::Decl const* const, clang::Decl const*>::doit(clang::Decl const* const&) + 34
12 clang-3.9                0x000000010f410595 bool llvm::isa<clang::FunctionDecl, clang::Decl const*>(clang::Decl const* const&) + 21
13 clang-3.9                0x000000010f40f4d8 llvm::cast_retty<clang::FunctionDecl, clang::Decl const*>::ret_type llvm::dyn_cast<clang::FunctionDecl, clang::Decl const>(clang::Decl const*) + 24
14 clang-3.9                0x000000010ce0d741 (anonymous namespace)::AVRTargetCodeGenInfo::setTargetAttributes(clang::Decl const*, llvm::GlobalValue*, clang::CodeGen::CodeGenModule&) const + 33
15 clang-3.9                0x000000010cbf019e clang::CodeGen::CodeGenModule::setNonAliasAttributes(clang::Decl const*, llvm::GlobalObject*) + 190
16 clang-3.9                0x000000010cbf0d25 clang::CodeGen::CodeGenModule::SetInternalFunctionAttributes(clang::Decl const*, llvm::Function*, clang::CodeGen::CGFunctionInfo const&) + 117
17 clang-3.9                0x000000010ca21196 clang::CodeGen::CodeGenModule::CreateGlobalInitOrDestructFunction(llvm::FunctionType*, llvm::Twine const&, clang::CodeGen::CGFunctionInfo const&, clang::SourceLocation, bool) + 246
18 clang-3.9                0x000000010ca218ce clang::CodeGen::CodeGenModule::EmitCXXGlobalVarDeclInitFunc(clang::VarDecl const*, llvm::GlobalVariable*, bool) + 542
19 clang-3.9                0x000000010cbfa081 clang::CodeGen::CodeGenModule::EmitGlobalVarDefinition(clang::VarDecl const*, bool) + 3521
20 clang-3.9                0x000000010cbf2f9d clang::CodeGen::CodeGenModule::EmitGlobalDefinition(clang::GlobalDecl, llvm::GlobalValue*) + 845
21 clang-3.9                0x000000010cbf7e48 clang::CodeGen::CodeGenModule::EmitGlobal(clang::GlobalDecl) + 1208
[... cut ...]
clang-3.9: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.9: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.9: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/IPAddress-fa9035.cpp
clang-3.9: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/IPAddress-fa9035.sh
clang-3.9: note: diagnostic msg: 
```

dylanmckay, 2016-01-20:
Can you run `-S -emit-llvm` to get an IR file.

4ntoine, 2016-01-20:
sorry, now i remember you've asked to check it too:
```
MBA-Anton:bin asmirnov$ ./clang -g -Os -S -emit-llvm  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp" -o "/tmp/arduino_test1/HardwareSerial0.cpp.ll" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
Assertion failed: (Val && "isa<> used on a null pointer"), function doit, file /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/include/llvm/Support/Casting.h, line 95.
0  clang-3.9                0x000000010d4c920e llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 46
1  clang-3.9                0x000000010d4c9659 PrintStackTraceSignalHandler(void*) + 25
2  clang-3.9                0x000000010d4c5df9 llvm::sys::RunSignalHandlers() + 425
3  clang-3.9                0x000000010d4c99b9 SignalHandler(int) + 345
4  libsystem_platform.dylib 0x00007fff98cb7f1a _sigtramp + 26
5  libsystem_platform.dylib 0x00007fff6af96784 _sigtramp + 3526224004
6  clang-3.9                0x000000010d4c967b raise + 27
7  clang-3.9                0x000000010d4c9732 abort + 18
8  clang-3.9                0x000000010d4c9711 __assert_rtn + 129
9  clang-3.9                0x00000001102f8a5c llvm::isa_impl_cl<clang::FunctionDecl, clang::Decl const*>::doit(clang::Decl const*) + 92
10 clang-3.9                0x00000001102f89e8 llvm::isa_impl_wrap<clang::FunctionDecl, clang::Decl const*, clang::Decl const*>::doit(clang::Decl const* const&) + 24
11 clang-3.9                0x00000001102f89c2 llvm::isa_impl_wrap<clang::FunctionDecl, clang::Decl const* const, clang::Decl const*>::doit(clang::Decl const* const&) + 34
12 clang-3.9                0x00000001102ca595 bool llvm::isa<clang::FunctionDecl, clang::Decl const*>(clang::Decl const* const&) + 21
13 clang-3.9                0x00000001102c94d8 llvm::cast_retty<clang::FunctionDecl, clang::Decl const*>::ret_type llvm::dyn_cast<clang::FunctionDecl, clang::Decl const>(clang::Decl const*) + 24
14 clang-3.9                0x000000010dcc7741 (anonymous namespace)::AVRTargetCodeGenInfo::setTargetAttributes(clang::Decl const*, llvm::GlobalValue*, clang::CodeGen::CodeGenModule&) const + 33
15 clang-3.9                0x000000010daaa19e clang::CodeGen::CodeGenModule::setNonAliasAttributes(clang::Decl const*, llvm::GlobalObject*) + 190
16 clang-3.9                0x000000010daaad25 clang::CodeGen::CodeGenModule::SetInternalFunctionAttributes(clang::Decl const*, llvm::Function*, clang::CodeGen::CGFunctionInfo const&) + 117
17 clang-3.9                0x000000010d8db196 clang::CodeGen::CodeGenModule::CreateGlobalInitOrDestructFunction(llvm::FunctionType*, llvm::Twine const&, clang::CodeGen::CGFunctionInfo const&, clang::SourceLocation, bool) + 246
18 clang-3.9                0x000000010d8db8ce clang::CodeGen::CodeGenModule::EmitCXXGlobalVarDeclInitFunc(clang::VarDecl const*, llvm::GlobalVariable*, bool) + 542
19 clang-3.9                0x000000010dab4081 clang::CodeGen::CodeGenModule::EmitGlobalVarDefinition(clang::VarDecl const*, bool) + 3521
20 clang-3.9                0x000000010daacf9d clang::CodeGen::CodeGenModule::EmitGlobalDefinition(clang::GlobalDecl, llvm::GlobalValue*) + 845
21 clang-3.9                0x000000010dab1e48 clang::CodeGen::CodeGenModule::EmitGlobal(clang::GlobalDecl) + 1208
22 clang-3.9                0x000000010dab55d4 clang::CodeGen::CodeGenModule::EmitTopLevelDecl(clang::Decl*) + 404
23 clang-3.9                0x000000010dc228fd (anonymous namespace)::CodeGeneratorImpl::HandleTopLevelDecl(clang::DeclGroupRef) + 157
24 clang-3.9                0x000000010da871c0 clang::BackendConsumer::HandleTopLevelDecl(clang::DeclGroupRef) + 192
25 clang-3.9                0x000000010ef0a769 clang::ParseAST(clang::Sema&, bool, bool) + 969
26 clang-3.9                0x000000010dfefbbf clang::ASTFrontendAction::ExecuteAction() + 511
27 clang-3.9                0x000000010da85c07 clang::CodeGenAction::ExecuteAction() + 5991
28 clang-3.9                0x000000010dfef120 clang::FrontendAction::Execute() + 112
29 clang-3.9                0x000000010df3c5c1 clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) + 1873
30 clang-3.9                0x000000010e07efaa clang::ExecuteCompilerInvocation(clang::CompilerInstance*) + 4410
31 clang-3.9                0x000000010c1f4b9b cc1_main(llvm::ArrayRef<char const*>, char const*, void*) + 4923
32 clang-3.9                0x000000010c1e498c ExecuteCC1Tool(llvm::ArrayRef<char const*>, llvm::StringRef) + 476
33 clang-3.9                0x000000010c1e24ec main + 3244
34 libdyld.dylib            0x00007fff966e85c9 start + 1
Stack dump:
0.	Program arguments: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.9 -cc1 -triple avr -emit-llvm -disable-free -main-file-name HardwareSerial0.cpp -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -target-cpu atmega328p -target-linker-version 253.9 -dwarf-column-info -debug-info-kind=limited -dwarf-version=4 -debugger-tuning=gdb -ffunction-sections -fdata-sections -coverage-file /tmp/arduino_test1/HardwareSerial0.cpp.ll -resource-dir /Users/asmirnov/Documents [... cut ...]
clang-3.9: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.9: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.9: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/HardwareSerial0-12e9c2.cpp
clang-3.9: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/HardwareSerial0-12e9c2.sh
clang-3.9: note: diagnostic msg: 
```

dylanmckay, 2016-01-20:
This is an LLVM issue, caused by the recent update to LLVM+clang.

dylanmckay, 2016-01-20:
Can you post the fully-preprocessed C file? I don't remember what the command line argument is called.

4ntoine, 2016-01-20:
is it what you ask for? https://github.com/avr-llvm/llvm/issues/181#issuecomment-173151392

dylanmckay, 2016-01-20:
> is it what you ask for?

Yup.

If you remove the `-mmcu=atmega328p`, it doesn't fail.

4ntoine, 2016-01-20:
This example is compiling Blink sketch (arduino source code is called "sketch") for Arduino Uno board and it has 'atmega328p' mcu (which is wide-spread).

I've tried for atmega2560 and it crashes too:
```
./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp" -o "/tmp/arduino_test1/HardwareSerial0.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
```

dylanmckay, 2016-01-20:
I believe I just fixed the clang issue, pull changes and see if the problem still occurs.

4ntoine, 2016-01-20:
i will update and report back asap

4ntoine, 2016-01-20:
no more crashes! well done

closed 2016-01-20 by 4ntoine