avr-llvm/llvm #183: Assertion failed: this is not register operand

4ntoine, 2016-01-21:
```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp" -o "/tmp/arduino_test1/Stream.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
Assertion failed: (isReg() && "This is not a register operand!"), function getReg, file /Users/asmirnov/Documents/dev/src/avr-llvm/llvm/include/llvm/CodeGen/MachineOperand.h, line 268.
0  clang-3.9                0x000000010d5211ee llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 46
1  clang-3.9                0x000000010d521639 PrintStackTraceSignalHandler(void*) + 25
2  clang-3.9                0x000000010d51ddd9 llvm::sys::RunSignalHandlers() + 425
3  clang-3.9                0x000000010d521999 SignalHandler(int) + 345
4  libsystem_platform.dylib 0x00007fff98cb7f1a _sigtramp + 26
5  clang-3.9                0x0000000110be962d guard variable for shouldAddRequirement(clang::Module*, llvm::StringRef, bool&)::IOKitAVC + 71613
6  clang-3.9                0x000000010d52165b raise + 27
7  clang-3.9                0x000000010d521712 abort + 18
8  clang-3.9                0x000000010d5216f1 __assert_rtn + 129
9  clang-3.9                0x000000010c25d9f8 llvm::MachineOperand::getReg() const + 104
10 clang-3.9                0x000000010c28caea llvm::isCopyMulResult(llvm::MachineBasicBlock::bundle_iterator<llvm::MachineInstr, llvm::ilist_iterator<llvm::MachineInstr> > const&) + 42
11 clang-3.9                0x000000010c287ce1 llvm::AVRTargetLowering::insertMul(llvm::MachineInstr*, llvm::MachineBasicBlock*) const + 145
12 clang-3.9                0x000000010c287ec7 llvm::AVRTargetLowering::EmitInstrWithCustomInserter(llvm::MachineInstr*, llvm::MachineBasicBlock*) const + 231
13 clang-3.9                0x000000010c64be9e (anonymous namespace)::ExpandISelPseudos::runOnMachineFunction(llvm::MachineFunction&) + 286
14 clang-3.9                0x000000010c78254e llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 110
15 clang-3.9                0x000000010ccc153d llvm::FPPassManager::runOnFunction(llvm::Function&) + 413
16 clang-3.9                0x000000010ccc1875 llvm::FPPassManager::runOnModule(llvm::Module&) + 117
17 clang-3.9                0x000000010ccc25ca (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) + 2010
18 clang-3.9                0x000000010ccc1b5b llvm::legacy::PassManagerImpl::run(llvm::Module&) + 347
19 clang-3.9                0x000000010ccc3211 llvm::legacy::PassManager::run(llvm::Module&) + 33
20 clang-3.9                0x000000010d80f470 (anonymous namespace)::EmitAssemblyHelper::EmitAssembly(clang::BackendAction, llvm::raw_pwrite_stream*) + 3968
21 clang-3.9                0x000000010d80dd02 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::raw_pwrite_stream*) + 162
22 clang-3.9                0x000000010dae3005 clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) + 2021
23 clang-3.9                0x000000010ef72273 clang::ParseAST(clang::Sema&, bool, bool) + 1299
24 clang-3.9                0x000000010e04e14f clang::ASTFrontendAction::ExecuteAction() + 511
25 clang-3.9                0x000000010dae1127 clang::CodeGenAction::ExecuteAction() + 5991
26 clang-3.9                0x000000010e04d6b0 clang::FrontendAction::Execute() + 112
27 clang-3.9                0x000000010df9a821 clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) + 1873
28 clang-3.9                0x000000010e0dd69a clang::ExecuteCompilerInvocation(clang::CompilerInstance*) + 4410
29 clang-3.9                0x000000010c2426ae cc1_main(llvm::ArrayRef<char const*>, char const*, void*) + 4926
30 clang-3.9                0x000000010c23244f ExecuteCC1Tool(llvm::ArrayRef<char const*>, llvm::StringRef) + 479
31 clang-3.9                0x000000010c22ffad main + 3245
32 libdyld.dylib            0x00007fff966e85c9 start + 1
33 libdyld.dylib            0x0000000000000044 start + 1771141756
Stack dump:
0.	Program arguments: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.9 -cc1 -triple avr -emit-obj -disable-free -main-file-name Stream.cpp -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -target-cpu atmega328p -target-linker-version 253.9 -dwarf-column-info -debug-info-kind=limited -dwarf-version=4 -debugger-tuning=gdb -ffunction-sections -fdata-sections -coverage-file /tmp/arduino_test1/Stream.cpp.o -resource-dir /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/../lib/clang/3.9.0 -dependency-file /tmp/arduino_test1/Stream.cpp.d -MT /tmp/arduino_test1/Stream.cpp.o -D F_CPU=16000000L -D ARDUINO=10607 -D ARDUINO_AVR_UNO -D ARDUINO_ARCH_AVR -I /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard -I /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include -Os -std=gnu++11 -fdeprecated-macro -fdebug-compilation-dir /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin -ferror-limit 19 -fmessage-length 138 -fno-threadsafe-statics -fobjc-runtime=gcc -fdiagnostics-show-option -fcolor-diagnostics -vectorize-loops -vectorize-slp -o /tmp/arduino_test1/Stream.cpp.o -x c++ /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp 
1.	<eof> parser at end of file
2.	Code generation
3.	Running pass 'Function Pass Manager' on module '/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp'.
4.	Running pass 'Expand ISel Pseudo-instructions' on function '@_ZN6Stream9findMultiEPNS_11MultiTargetEi'
clang-3.9: error: unable to execute command: Illegal instruction: 4
clang-3.9: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.9.0 (https://github.com/avr-llvm/clang.git 82a9a3e4b5a70a76621c9a06926e68baae9765c9) (llvm/llvm.git cfc3f431c02364847df4c9b09b4cea54a0669e43)
Target: avr
Thread model: posix
InstalledDir: /Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/.
clang-3.9: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.9: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.9: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Stream-e25064.cpp
clang-3.9: note: diagnostic msg: /var/folders/64/fwfkm1k51zbd4_c5lwpsbljh0000gn/T/Stream-e25064.sh
clang-3.9: note: diagnostic msg: 
```

4ntoine, 2016-01-21:
invocation with `-S -emit-llvm` succeed:
```
MBA-Anton:bin asmirnov$ ./clang -S -emit-llvm -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp" -o "/tmp/arduino_test1/Stream.cpp.ll" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
```

4ntoine, 2016-01-21:
generated Stream.cpp.ll is here: https://gist.github.com/4ntoine/34a9dca57e775d6309d6

4ntoine, 2016-01-21:
Stream-e25064.cpp is here: https://gist.github.com/4ntoine/043f1924c784ba93919d

4ntoine, 2016-01-21:
Stream-e25064.sh:
```
# Crash reproducer for clang version 3.9.0 (https://github.com/avr-llvm/clang.git 82a9a3e4b5a70a76621c9a06926e68baae9765c9) (llvm/llvm.git cfc3f431c02364847df4c9b09b4cea54a0669e43)
# Driver args: "-c" "-g" "-Os" "-std=gnu++11" "-fno-exceptions" "-ffunction-sections" "-fdata-sections" "-fno-threadsafe-statics" "-MMD" "-mmcu=atmega328p" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp" "-o" "/tmp/arduino_test1/Stream.cpp.o" "-I" "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include" "--target=avr"
# Original command:  "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.9" "-cc1" "-triple" "avr" "-emit-obj" "-disable-free" "-main-file-name" "Stream.cpp" "-mrelocation-model" "static" "-mthread-model" "posix" "-mdisable-fp-elim" "-fmath-errno" "-masm-verbose" "-mconstructor-aliases" "-target-cpu" "atmega328p" "-target-linker-version" "253.9" "-dwarf-column-info" "-debug-info-kind=limited" "-dwarf-version=4" "-debugger-tuning=gdb" "-ffunction-sections" "-fdata-sections" "-coverage-file" "/tmp/arduino_test1/Stream.cpp.o" "-resource-dir" "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/../lib/clang/3.9.0" "-dependency-file" "/tmp/arduino_test1/Stream.cpp.d" "-MT" "/tmp/arduino_test1/Stream.cpp.o" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I" "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include" "-Os" "-std=gnu++11" "-fdeprecated-macro" "-fdebug-compilation-dir" "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin" "-ferror-limit" "19" "-fmessage-length" "138" "-fno-threadsafe-statics" "-fobjc-runtime=gcc" "-fdiagnostics-show-option" "-fcolor-diagnostics" "-vectorize-loops" "-vectorize-slp" "-o" "/tmp/arduino_test1/Stream.cpp.o" "-x" "c++" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Stream.cpp"
 "/Users/asmirnov/Documents/dev/src/avr-llvm/build/bin/clang-3.9" "-cc1" "-triple" "avr" "-emit-obj" "-disable-free" "-main-file-name" "Stream.cpp" "-mrelocation-model" "static" "-mthread-model" "posix" "-mdisable-fp-elim" "-fmath-errno" "-masm-verbose" "-mconstructor-aliases" "-target-cpu" "atmega328p" "-target-linker-version" "253.9" "-dwarf-column-info" "-debug-info-kind=limited" "-dwarf-version=4" "-debugger-tuning=gdb" "-ffunction-sections" "-fdata-sections" "-D" "F_CPU=16000000L" "-D" "ARDUINO=10607" "-D" "ARDUINO_AVR_UNO" "-D" "ARDUINO_ARCH_AVR" "-Os" "-std=gnu++11" "-fdeprecated-macro" "-ferror-limit" "19" "-fmessage-length" "138" "-fno-threadsafe-statics" "-fobjc-runtime=gcc" "-fdiagnostics-show-option" "-fcolor-diagnostics" "-vectorize-loops" "-vectorize-slp" "-x" "c++" "Stream-e25064.cpp"
```

shepmaster, 2016-01-21:
Reduced:

```llvm
%class.Stream.1 = type { %class.Print.0, i32, i32 }
%class.Print.0 = type { i32 (...)**, i16 }
%Stream.MultiTarget.2 = type { i8*, i16, i16 }

define i16 @findMulti(%class.Stream.1* %this, %Stream.MultiTarget.2* %targets) align 2 {
entry:
  %sub.ptr.rhs.cast = ptrtoint %Stream.MultiTarget.2* %targets to i16
  %sub.ptr.sub = sub i16 0, %sub.ptr.rhs.cast
  %sub.ptr.div = sdiv exact i16 %sub.ptr.sub, 6
  ret i16 %sub.ptr.div
}
```

dylanmckay, 2016-05-31:
@shepmaster's test case now is successfully lowered to assembly, and assembled into machine code.

closed 2016-05-31 by dylanmckay