avr-llvm/llvm #185: avr-llvm compilation fails

kbartosik, 2016-01-25:
Hi
First of all many thanks for developing avr-llvm & co. 
That is a brilliant project with sound business need.
I wanted to give it a try but compilation fails with:
```
[ 68%] Building CXX object projects/compiler-rt/lib/tsan/CMakeFiles/clang_rt.tsan-x86_64.dir/rtl/tsan_interceptors.cc.o
In file included from /home/krzysiek/devel/llvm/projects/compiler-rt/lib/tsan/rtl/tsan_interceptors.cc:2427:
/home/krzysiek/devel/llvm/projects/compiler-rt/lib/tsan/../sanitizer_common/sanitizer_common_interceptors.inc:41:10: fatal error: 'stdarg.h' file not found
#include <stdarg.h>
```
*Repo was cloned and build as per Getting Started manual, with
cmake -G "Unix Makefiles" ../   and
cmake -G "Unix Makefiles" -DLLVM_TARGETS_TO_BUILD="AVR" ../
both results in the same error.

*host system:
FreeBSD steel 10.2-RELEASE FreeBSD 10.2-RELEASE #0 r286666: Wed Aug 12 15:26:37 UTC 2015     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/sys/GENERIC  amd64

*(cc   --version):
FreeBSD clang version 3.4.1 (tags/RELEASE_34/dot1-final 208032) 20140512
Target: x86_64-unknown-freebsd10.2
Thread model: posix

*Used commit
cfc3f431c02364847df4c9b09b4cea54a0669e43

Thanks for support
Kris

rjordans, 2016-01-25:
I have had some success when not checking out the compiler-rt libraries.  I'm not sure if they are even used by the compiler yet?

kbartosik, 2016-01-25:
Cheers rjordans,
I will give it a try w/o compiler-rt and check results

DavidEGrayson, 2016-01-25:
Looks like a typo to me.  I think `stdargh` should be `stdarg.h`.

kbartosik, 2016-01-26:
OK, after I had removed compiler-rt repo, the compilation finished successfully.
I ran it on some trivial C++ code for atmega8 and assembly looks fine (have not loaded it on chip though).
To project leaders: you can close the issue or leave it open as a reminder to do something about it (either correct compiler-rt or update manual with that this component is not needed, if it is not indeed) -  to your discretion.

dylanmckay, 2016-01-26:
Compiler is not necessarily required for hello world, but any non-trivial program will use it.

It defines builtin functions such as `__udivmodsi4` which is called whenever we need to do division. Without it, `undefined reference` errors would ensue.

It is strange that `compiler-rt` does not build correctly - the file that causes that error has not been modified for AVR - it is an upstream problem.

I've been able to successfully compile LLVM+compiler_rt+clang on Linux and Mac.

@rjordans: are you also on FreeBSD?

rjordans, 2016-01-27:
I'm on Linux but my code is simple enough that I don't need compiler_rt for now so I didn't try that part

rjordans, 2016-01-27:
I tried to follow the instructions on the getting started manual for adding compiler-rt to my build and now remember why I was skipping it previously.  It seems that the compiler-rt build is having a problem for me in using the right compiler to build.

During the CMake configure I get an empty list of supported compiler-rt platforms (I build the AVR LLVM with only AVR support) and it then skips building compiler-rt...  My guess is that it tries to use the compiler installed on the host system to build the library in-stead of the newly built AVR-LLVM.

I was planning to try building compiler-rt as a standalone project with the freshly built AVR-LLVM compiler but I didn't get to that yet and my current project doesn't need the compiler-rt functionality yet.

kbartosik, 2016-01-29:
I have repeated test on vanilla system in VM just to be 100% sure there are no artifacts from my environment that would influence compilation and the result is the same.

```
Scanning dependencies of target clang_rt.tsan-x86_64
[ 57%] Building CXX object projects/compiler-rt/lib/tsan/CMakeFiles/clang_rt.tsan-x86_64.dir/rtl/tsan_clock.cc.o
[ 57%] Building CXX object projects/compiler-rt/lib/tsan/CMakeFiles/clang_rt.tsan-x86_64.dir/rtl/tsan_flags.cc.o
[ 57%] Building CXX object projects/compiler-rt/lib/tsan/CMakeFiles/clang_rt.tsan-x86_64.dir/rtl/tsan_fd.cc.o
[ 57%] Building CXX object projects/compiler-rt/lib/tsan/CMakeFiles/clang_rt.tsan-x86_64.dir/rtl/tsan_ignoreset.cc.o
[ 57%] Building CXX object projects/compiler-rt/lib/tsan/CMakeFiles/clang_rt.tsan-x86_64.dir/rtl/tsan_interceptors.cc.o
In file included from /home/krzysiek/devel/llvm/projects/compiler-rt/lib/tsan/rtl/tsan_interceptors.cc:2427:
/home/krzysiek/devel/llvm/projects/compiler-rt/lib/tsan/../sanitizer_common/sanitizer_common_interceptors.inc:41:10: fatal error: 'stdarg.h' file not found
#include <stdarg.h>
         ^
1 error generated.
*** Error code 1

Stop.
make[2]: stopped in /usr/home/krzysiek/devel/llvm/build
*** Error code 1

free11% cc --version
FreeBSD clang version 3.7.1 (tags/RELEASE_371/final 255217) 20151225
Target: x86_64-unknown-freebsd11.0
Thread model: posix
free11% uname -a
FreeBSD free11 11.0-CURRENT FreeBSD 11.0-CURRENT #0 r294499: Thu Jan 21 15:46:19 UTC 2016     root@releng2.nyi.freebsd.org:/usr/obj/usr/src/sys/GENERIC  amd64
```
It is reproducible - I used this image from FreeBSD.org (it's a raw image - the only piece of sw to be installed in virtual machine is cmake)
[ftp://ftp.freebsd.org/pub/FreeBSD/snapshots/VM-IMAGES/11.0-CURRENT/amd64/Latest/FreeBSD-11.0-CURRENT-amd64.raw.xz](url)

dylanmckay, 2016-01-29:
Just found [this](http://lists.llvm.org/pipermail/llvm-dev/2015-December/093133.html) mailing list post.

Seems like this is an LLVM problem in general, although I'm not sure why it hasn't been fixed.

kbartosik, 2016-01-29:
Thanks Dylan. Indeed compilation issues following
```
"/usr/bin/CC   -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/krzysiek/devel/llvm/build/projects/compiler-rt/lib/tsan -I/home/krzysiek/devel/llvm/projects/compiler-rt/lib/tsan -I/home/krzysiek/devel/llvm/build/include -I/home/krzysiek/devel/llvm/include -I/usr/local/include -I/home/krzysiek/devel/llvm/projects/compiler-rt/lib/tsan/..   -fPIC -fvisibility-inlines-hidden -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wcovered-switch-default -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -std=c++11 -ffunction-sections -fdata-sections -Wall -std=c++11    -m64 -fPIC -fno-builtin -fno-exceptions -fomit-frame-pointer -funwind-tables -fno-stack-protector -fno-sanitize=safe-stack -fvisibility=hidden -fvisibility-inlines-hidden -fno-function-sections -fno-lto -O3 -gline-tables-only -Wno-gnu -Wno-variadic-macros -Wno-c99-extensions -Wno-non-virtual-dtor -fPIE -fno-rtti -msse3 -Wframe-larger-than=512 -Wglobal-constructors --sysroot=. -o CMakeFiles/clang_rt.tsan-x86_64.dir/rtl/tsan_interceptors.cc.o -c /home/krzysiek/devel/llvm/projects/compiler-rt/lib/tsan/rtl/tsan_interceptors.cc",
  "file": "/home/krzysiek/devel/llvm/projects/compiler-rt/lib/tsan/rtl/tsan_interceptors.cc"
```

but the header is actually in _tools/clang/lib/Headers_
I do not see this directory among includes nor is it copied into any _*/build/*_ dir as far as I can see. 

The post you mentioned seems showing the same problem, however here I would like to avoid including system inc (the _stdarg.h_ file can be found in system _/usr/include_ but that is not the same as the one in _clang/lib/Headers_)... I am not sure what consequences it may have on compiler-rt if system header file is being used instead...

Do you have any idea on how to convince builder to include _clang/lib/Headers_ directory?

dylanmckay, 2016-02-02:
I have no idea sorry. I have not worked with `clang` much, and I have never used BSD :P

closed 2016-05-20 by dylanmckay