avr-llvm/llvm #186: Consider combining or renaming avr_signalcc and avr_intrcc?

shepmaster, 2016-01-25:
Through a bit of source diving, I think the only difference between `avr_signalcc` and `avr_intrcc` is that `avr_intrcc` re-enables the global interrupt flag immediately upon entry.  At best, this difference is non-obvious based on the names.

The version of AVR's `interrupt.h` I have defines these constants:

```c
#  define ISR_BLOCK
#  define ISR_NOBLOCK    __attribute__((interrupt))
#  define ISR_NAKED      __attribute__((naked))
```

With an intended usage like `ISR(timer, ISR_NOBLOCK) { ... }`. Specifically, the `NOBLOCK` one corresponds to the case of re-enabling the global interrupt flag immediately upon entry. 

I might suggest leveraging the nicer names that are now being promoted.

dylanmckay, 2016-02-02:
I think this is good, although some of this stuff has already been merged into LLVM trunk so it would likely need to wait.

dylanmckay, 2016-02-02:
@shepmaster: Also, Rust updated their version of LLVM, and so I updated AVR-LLVM to the same revision. Current `HEAD` is c1e46504289ccceb030731e902fd77e40f597f79.

dylanmckay, 2016-02-03:
I have now updated LLVM again, make sure to use c1e46504289ccceb030731e902fd77e40f597f79.

dylanmckay, 2016-05-31:
It looks like AVR-GCC treats signal handlers and naked functions differently, see [this](http://www.freertos.org/implementation/a00013.html).

To define a signal handler, both attributes need to be set:

``` c++
__attribute__ ( ( signal, naked ) )
```

If we want to be compatible with GCC, we need to support both the `interrupt` and `signal` attributes. Naked functions are a related, but orthogonal concept.

Feel free to reopen this if you have any doubts :)

closed 2016-05-31 by dylanmckay

shepmaster, 2016-05-31:
FWIW, I chose to expose these in Rust as `avr-interrupt` and `avr-non-blocking-interrupt` (Calling conventions 85 and 84 respectively).